raspberrypi/linux #5914: Pi 5 I2C timing out of spec

pelwell, 2024-01-31:
### Describe the bug

As detailed in this thread (https://forums.raspberrypi.com/viewtopic.php?p=2187613#p2187613), the new Pi 5 I2C timing doesn't adhere to the specification. In particular, the SDA line should not start to change until the SCL line has dropped below the 30% level (i.e. become a clear digital low).

### Steps to reproduce the behaviour

Use I2C, look at scope traces.

### Device (s)

Raspberry Pi 5

### System

Seen all all current Linux branches.

pelwell, 2024-01-31:
There's now a Pull Request that should help with the I2C timing. See https://github.com/raspberrypi/linux/pull/5916.

There should be a test build available in the next 45 minutes (it should definitely be done by 1900 GMT). After that, run "sudo rpi-update pulls/5916" to install it. However:
1. It's beta software, so should be used with caution.
2. It's a 6.6 kernel, which might be too big a change for you.
3. It's only a Device Tree change, so a full kernel install is overkill.

Alternatively, download just the updated Pi 5 dtb file from here: https://drive.google.com/file/d/1NGJsJoeFf_K8mVi0lNjO6FIPEOYj9lvf/view?usp=sharing

Copy it to /boot/firmware/ (or just /boot on an older image), reboot, and see how you get on.

dgedgedge, 2024-02-04:
Note: This patch correctes our issue.
After put a scope on the I2C bus, there is now a 300ns TsuDAT/ThDAT guard on the signal.

closed 2024-02-04 by pelwell