raspberrypi/linux #5920: Genet doesn't send/recieve when config_hz is set lower than 100 Hz

mrwizardwizard, 2024-02-04:
### Describe the bug

Genet doesn't send/receive when config_hz is set lower than 100 Hz (using a custom timer frequency) on raspberry Pi 4 model B

Setting a custom timer value is located in
kernel/Kconfig.hz

it can be set as low as 16 Hz

### Steps to reproduce the behaviour

Change Kconfig.hz located in kernel/Kconfig.hz and add/change

choice
	prompt "Timer frequency"
	default HZ_16
	help

config HZ_16
		bool "16 HZ"
	help

endchoice

config HZ
	int
	default 16 if HZ_16

config SCHED_HRTICK
	def_bool HIGH_RES_TIMERS

recompile the kernel, boot it, and genet won't send/receive properly.

### Device (s)

Raspberry Pi 4 Mod. B

### System

OS: Gentoo Linux
Kernel Version: Linux 6.7.2+ SMP aarch64 GNU/Linux

pelwell, 2024-02-04:
Are you saying that 99Hz doesn't work?

mrwizardwizard, 2024-02-05:
> Are you saying that 99Hz doesn't work?

I tried 16Hz and 50Hz and genet doesn't seem to work correctly, only works on 100Hz and above, didn't try 99Hz lol, but it is some kind of bug because with raspberry pi 3 model B+, i was able to use any Hz lower than 100Hz, like 16Hz.