raspberrypi/linux #5921: HWMON - thermal zone trip point temperature numbering scheme changed

aBUGSworstnightmare-rpi, 2024-02-05:
### Describe the bug

e/o last year I've demonstrated how to change the cooling map of the Pi5 fan - shortly before an official change was implemented
https://forums.raspberrypi.com/viewtopic.php?p=2174791&hilit=fan#p2174791

changing the cooling map i.e. like (added to config.txt)
```
# change the cooling map
dtparam=cpu_tepid=43000,cpu_tepid_hyst=2000
dtparam=cpu_warm=48000,cpu_warm_hyst=3000
dtparam=cpu_hot=52000,cpu_hot_hyst=4000
dtparam=cpu_vhot=58000,cpu_vhot_hyst=5000
```

resulted in below 

> pi@pi55thdez:~ $ grep -s . /sys/class/thermal/thermal_zone*/* /sys/class/thermal/cooling_device*/*
> /sys/class/thermal/thermal_zone0/available_policies:step_wise 
> /sys/class/thermal/thermal_zone0/cdev0_trip_point:1
> /sys/class/thermal/thermal_zone0/cdev0_weight:0
> /sys/class/thermal/thermal_zone0/cdev1_trip_point:2
> /sys/class/thermal/thermal_zone0/cdev1_weight:0
> /sys/class/thermal/thermal_zone0/cdev2_trip_point:3
> /sys/class/thermal/thermal_zone0/cdev2_weight:0
> /sys/class/thermal/thermal_zone0/cdev3_trip_point:4
> /sys/class/thermal/thermal_zone0/cdev3_weight:0
> /sys/class/thermal/thermal_zone0/cdev4_trip_point:0
> /sys/class/thermal/thermal_zone0/cdev4_weight:0
> /sys/class/thermal/thermal_zone0/integral_cutoff:0
> /sys/class/thermal/thermal_zone0/k_d:0
> /sys/class/thermal/thermal_zone0/k_i:0
> /sys/class/thermal/thermal_zone0/k_po:0
> /sys/class/thermal/thermal_zone0/k_pu:0
> /sys/class/thermal/thermal_zone0/mode:enabled
> /sys/class/thermal/thermal_zone0/offset:450000
> /sys/class/thermal/thermal_zone0/policy:step_wise
> /sys/class/thermal/thermal_zone0/slope:-550
> /sys/class/thermal/thermal_zone0/sustainable_power:0
> /sys/class/thermal/thermal_zone0/temp:42450
> /sys/class/thermal/thermal_zone0/trip_point_0_hyst:0
> /sys/class/thermal/thermal_zone0/trip_point_0_temp:110000
> /sys/class/thermal/thermal_zone0/trip_point_0_type:critical
> /sys/class/thermal/thermal_zone0/trip_point_1_hyst:2000
> /sys/class/thermal/thermal_zone0/trip_point_1_temp:43000
> /sys/class/thermal/thermal_zone0/trip_point_1_type:active
> /sys/class/thermal/thermal_zone0/trip_point_2_hyst:3000
> /sys/class/thermal/thermal_zone0/trip_point_2_temp:48000
> /sys/class/thermal/thermal_zone0/trip_point_2_type:active
> /sys/class/thermal/thermal_zone0/trip_point_3_hyst:4000
> /sys/class/thermal/thermal_zone0/trip_point_3_temp:52000
> /sys/class/thermal/thermal_zone0/trip_point_3_type:active
> /sys/class/thermal/thermal_zone0/trip_point_4_hyst:5000
> /sys/class/thermal/thermal_zone0/trip_point_4_temp:58000
> /sys/class/thermal/thermal_zone0/trip_point_4_type:active
> /sys/class/thermal/thermal_zone0/type:cpu-thermal
> /sys/class/thermal/cooling_device0/cur_state:1
> /sys/class/thermal/cooling_device0/max_state:4
> /sys/class/thermal/cooling_device0/type:pwm-fan

trip point temperatures were shown in ascending order (lowest to highest).

I've prepared an overlay which can be used on other Pis https://forums.raspberrypi.com/viewtopic.php?p=2189814#p2182625

Compiled the overlay today (on the DUT 6.1.65 kernel sources were used) will show the values in descending order

> pi@pi4gold:~ $ uname -a
> Linux pi4gold 6.1.65-v8+ #1 SMP PREEMPT Mon Feb  5 09:53:12 CET 2024 aarch64 GNU/Linux
> pi@pi4gold:~ $ grep -s . /sys/class/thermal/thermal_zone*/* /sys/class/thermal/cooling_device*/*
> /sys/class/thermal/thermal_zone0/available_policies:step_wise 
> /sys/class/thermal/thermal_zone0/cdev0_trip_point:0
> /sys/class/thermal/thermal_zone0/cdev0_weight:0
> /sys/class/thermal/thermal_zone0/cdev1_trip_point:1
> /sys/class/thermal/thermal_zone0/cdev1_weight:0
> /sys/class/thermal/thermal_zone0/cdev2_trip_point:2
> /sys/class/thermal/thermal_zone0/cdev2_weight:0
> /sys/class/thermal/thermal_zone0/cdev3_trip_point:3
> /sys/class/thermal/thermal_zone0/cdev3_weight:0
> /sys/class/thermal/thermal_zone0/integral_cutoff:0
> /sys/class/thermal/thermal_zone0/k_d:0
> /sys/class/thermal/thermal_zone0/k_i:0
> /sys/class/thermal/thermal_zone0/k_po:0
> /sys/class/thermal/thermal_zone0/k_pu:0
> /sys/class/thermal/thermal_zone0/mode:enabled
> /sys/class/thermal/thermal_zone0/offset:410040
> /sys/class/thermal/thermal_zone0/policy:step_wise
> /sys/class/thermal/thermal_zone0/slope:-487
> /sys/class/thermal/thermal_zone0/sustainable_power:0
> /sys/class/thermal/thermal_zone0/temp:40894
> /sys/class/thermal/thermal_zone0/trip_point_0_hyst:5000
> /sys/class/thermal/thermal_zone0/trip_point_0_temp:58000
> /sys/class/thermal/thermal_zone0/trip_point_0_type:active
> /sys/class/thermal/thermal_zone0/trip_point_1_hyst:4000
> /sys/class/thermal/thermal_zone0/trip_point_1_temp:52000
> /sys/class/thermal/thermal_zone0/trip_point_1_type:active
> /sys/class/thermal/thermal_zone0/trip_point_2_hyst:3000
> /sys/class/thermal/thermal_zone0/trip_point_2_temp:40000
> /sys/class/thermal/thermal_zone0/trip_point_2_type:active
> /sys/class/thermal/thermal_zone0/trip_point_3_hyst:2000
> /sys/class/thermal/thermal_zone0/trip_point_3_temp:36000
> /sys/class/thermal/thermal_zone0/trip_point_3_type:active
> /sys/class/thermal/thermal_zone0/trip_point_4_hyst:0
> /sys/class/thermal/thermal_zone0/trip_point_4_temp:110000
> /sys/class/thermal/thermal_zone0/trip_point_4_type:critical
> /sys/class/thermal/thermal_zone0/type:cpu-thermal
> /sys/class/thermal/cooling_device0/cur_state:2
> /sys/class/thermal/cooling_device0/max_state:4
> /sys/class/thermal/cooling_device0/type:pwm-fan

Result on 6.6.14 kernel is the same -> values were in descending order

### Steps to reproduce the behaviour

compile the overlay (or use the .dtbo) and copy it into /boot/firmware/overlays

add to config.txt:
```
dtoverlay=pwm-fan
# change the cooling map
dtparam=fan_temp0=36000,fan_temp0_hyst=2000,fan_temp0_speed=90
dtparam=fan_temp1=40000,fan_temp1_hyst=3000,fan_temp1_speed=150
dtparam=fan_temp2=52000,fan_temp2_hyst=4000,fan_temp2_speed=200
dtparam=fan_temp3=58000,fan_temp3_hyst=5000,fan_temp3_speed=255
```

reboot

check thermal_zone details via
`grep -s . /sys/class/thermal/thermal_zone*/* /sys/class/thermal/cooling_device*/*`

### Device (s)

Raspberry Pi 4 Mod. B, Raspberry Pi 5

### System

Bookworm OS
kernel 6.1.65 and 6.6.14 were tested

aBUGSworstnightmare-rpi, 2024-02-05:
Don't know exactly! What I know for sure that I've made a fresh pull of the sources on 31st Dec. That kernel showed them in correct order.

pelwell, 2024-02-05:
Is that the 6.1 or 6.6 kernel?

aBUGSworstnightmare-rpi, 2024-02-05:
I've tested both that day (same as today).
Today both revisions (6.1.65 (!) and 6.6.14) show the same behavior, descending order - where I would expect ascending.

Just to make sure: the overlay is working correctly! It's just confusing what get's shown when checking the settings.

pelwell, 2024-02-05:
> Just to make sure: the overlay is working correctly! 

So this is purely a cosmetic issue? That would have been a good thing to put in the initial report.

aBUGSworstnightmare-rpi, 2024-02-05:
> > Just to make sure: the overlay is working correctly!
> 
> So this is purely a cosmetic issue? That would have been a good thing to put in the initial report.

Sorry, that was stated in the linked posts only!

aBUGSworstnightmare-rpi, 2024-02-06:
It matters which CPU architecture to overlays is build/running on!

On Pi5 the sorting is correct (ascending)

> pi@book27th:~ $ grep -s . /sys/class/thermal/thermal_zone*/* /sys/class/thermal/cooling_device*/*
> /sys/class/thermal/thermal_zone0/available_policies:step_wise 
> /sys/class/thermal/thermal_zone0/cdev0_trip_point:1
> /sys/class/thermal/thermal_zone0/cdev0_weight:0
> /sys/class/thermal/thermal_zone0/cdev1_trip_point:2
> /sys/class/thermal/thermal_zone0/cdev1_weight:0
> /sys/class/thermal/thermal_zone0/cdev2_trip_point:3
> /sys/class/thermal/thermal_zone0/cdev2_weight:0
> /sys/class/thermal/thermal_zone0/cdev3_trip_point:4
> /sys/class/thermal/thermal_zone0/cdev3_weight:0
> /sys/class/thermal/thermal_zone0/cdev4_trip_point:0
> /sys/class/thermal/thermal_zone0/cdev4_weight:0
> /sys/class/thermal/thermal_zone0/integral_cutoff:0
> /sys/class/thermal/thermal_zone0/k_d:0
> /sys/class/thermal/thermal_zone0/k_i:0
> /sys/class/thermal/thermal_zone0/k_po:0
> /sys/class/thermal/thermal_zone0/k_pu:0
> /sys/class/thermal/thermal_zone0/mode:enabled
> /sys/class/thermal/thermal_zone0/offset:450000
> /sys/class/thermal/thermal_zone0/policy:step_wise
> /sys/class/thermal/thermal_zone0/slope:-550
> /sys/class/thermal/thermal_zone0/sustainable_power:0
> /sys/class/thermal/thermal_zone0/temp:47400
> /sys/class/thermal/thermal_zone0/trip_point_0_hyst:0
> /sys/class/thermal/thermal_zone0/trip_point_0_temp:110000
> /sys/class/thermal/thermal_zone0/trip_point_0_type:critical
> /sys/class/thermal/thermal_zone0/trip_point_1_hyst:2000
> /sys/class/thermal/thermal_zone0/trip_point_1_temp:36000
> /sys/class/thermal/thermal_zone0/trip_point_1_type:active
> /sys/class/thermal/thermal_zone0/trip_point_2_hyst:3000
> /sys/class/thermal/thermal_zone0/trip_point_2_temp:40000
> /sys/class/thermal/thermal_zone0/trip_point_2_type:active
> /sys/class/thermal/thermal_zone0/trip_point_3_hyst:4000
> /sys/class/thermal/thermal_zone0/trip_point_3_temp:52000
> /sys/class/thermal/thermal_zone0/trip_point_3_type:active
> /sys/class/thermal/thermal_zone0/trip_point_4_hyst:5000
> /sys/class/thermal/thermal_zone0/trip_point_4_temp:58000
> /sys/class/thermal/thermal_zone0/trip_point_4_type:active
> /sys/class/thermal/thermal_zone0/type:cpu-thermal
> /sys/class/thermal/cooling_device0/cur_state:2
> /sys/class/thermal/cooling_device0/max_state:4
> /sys/class/thermal/cooling_device0/type:pwm-fan
> pi@book27th:~ $ uname -a
> Linux book27th 6.6.14-v8-16k+ #1 SMP PREEMPT Tue Feb  6 09:39:49 CET 2024 aarch64 GNU/Linux
> pi@book27th:~ $ cat /proc/cpuinfo | grep Revision
> Revision	: c04170

On a CM4 the order is shown descending

> pi@cm4-4GB-WC:~ $ grep -s . /sys/class/thermal/thermal_zone*/* /sys/class/thermal/cooling_device*/*
> /sys/class/thermal/thermal_zone0/available_policies:step_wise 
> /sys/class/thermal/thermal_zone0/cdev0_trip_point:0
> /sys/class/thermal/thermal_zone0/cdev0_weight:0
> /sys/class/thermal/thermal_zone0/cdev1_trip_point:1
> /sys/class/thermal/thermal_zone0/cdev1_weight:0
> /sys/class/thermal/thermal_zone0/cdev2_trip_point:2
> /sys/class/thermal/thermal_zone0/cdev2_weight:0
> /sys/class/thermal/thermal_zone0/cdev3_trip_point:3
> /sys/class/thermal/thermal_zone0/cdev3_weight:0
> /sys/class/thermal/thermal_zone0/integral_cutoff:0
> /sys/class/thermal/thermal_zone0/k_d:0
> /sys/class/thermal/thermal_zone0/k_i:0
> /sys/class/thermal/thermal_zone0/k_po:0
> /sys/class/thermal/thermal_zone0/k_pu:0
> /sys/class/thermal/thermal_zone0/mode:enabled
> /sys/class/thermal/thermal_zone0/offset:410040
> /sys/class/thermal/thermal_zone0/policy:step_wise
> /sys/class/thermal/thermal_zone0/slope:-487
> /sys/class/thermal/thermal_zone0/sustainable_power:0
> /sys/class/thermal/thermal_zone0/temp:35537
> /sys/class/thermal/thermal_zone0/trip_point_0_hyst:5000
> /sys/class/thermal/thermal_zone0/trip_point_0_temp:58000
> /sys/class/thermal/thermal_zone0/trip_point_0_type:active
> /sys/class/thermal/thermal_zone0/trip_point_1_hyst:4000
> /sys/class/thermal/thermal_zone0/trip_point_1_temp:52000
> /sys/class/thermal/thermal_zone0/trip_point_1_type:active
> /sys/class/thermal/thermal_zone0/trip_point_2_hyst:3000
> /sys/class/thermal/thermal_zone0/trip_point_2_temp:40000
> /sys/class/thermal/thermal_zone0/trip_point_2_type:active
> /sys/class/thermal/thermal_zone0/trip_point_3_hyst:2000
> /sys/class/thermal/thermal_zone0/trip_point_3_temp:36000
> /sys/class/thermal/thermal_zone0/trip_point_3_type:active
> /sys/class/thermal/thermal_zone0/trip_point_4_hyst:0
> /sys/class/thermal/thermal_zone0/trip_point_4_temp:110000
> /sys/class/thermal/thermal_zone0/trip_point_4_type:critical
> /sys/class/thermal/thermal_zone0/type:cpu-thermal
> /sys/class/thermal/cooling_device0/cur_state:0
> /sys/class/thermal/cooling_device0/max_state:4
> /sys/class/thermal/cooling_device0/type:pwm-fan
> pi@cm4-4GB-WC:~ $ uname -a
> Linux cm4-4GB-WC 6.6.14-v8+ #1 SMP PREEMPT Tue Feb  6 12:51:26 CET 2024 aarch64 GNU/Linux
> pi@cm4-4GB-WC:~ $ cat /proc/cpuinfo | grep Revision
> Revision	: b03141

as can be seen, kernel on both DUT was build from source this morning.