raspberrypi/linux #5927: Wolfson wm8731 audio codec broken between 6.1.73 and 6.1.74

rafael2k, 2024-02-06:
### Describe the bug

6.1.74 broke the wm8731 "snd_soc_audioinjector_pi_soundcard".  
6.1.73 was working fine. I'm running in a Pi 4, latest Raspberry Pi OS, 64 bit, bookworm.
I'm getting same kind of errors:
```
[   84.950454] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[  138.433994] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[  211.219306] bcm2835-i2s fe203000.i2s: ASoC: error at snd_soc_dai_hw_params on fe203000.i2s: -22
[  211.220038]  AudioInjector audio: ASoC: error at __soc_pcm_hw_params on AudioInjector audio: -22
[  215.901118] bcm2835-i2s fe203000.i2s: ASoC: error at snd_soc_dai_hw_params on fe203000.i2s: -22
[  215.902089]  AudioInjector audio: ASoC: error at __soc_pcm_hw_params on AudioInjector audio: -22
```
This is similar to bug https://github.com/raspberrypi/linux/issues/5817 , but got fixed. Now it is happening also in the Pi 4.

### Steps to reproduce the behaviour

Just boot the system.

### Device (s)

Raspberry Pi 4 Mod. B

### System

```
Oct 17 2023 15:39:16 
Copyright (c) 2012 Broadcom
version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start)
```
```
Linux estacao6.hermes.radio 6.1.74-v8+ #1725 SMP PREEMPT Mon Jan 22 13:35:32 GMT 2024 aarch64 GNU/Linux
```

### Logs

Write or read from alsa device does not work.
Sound card is stuck, just like in previous bug.

rafael2k, 2024-02-06:
I have another SD card with this kernel:
Linux estacao9.hermes.radio 6.1.0-rpi8-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux

Which is working.

rafael2k, 2024-02-06:
Thanks @pelwell 
Will update and report back.

rafael2k, 2024-02-06:
Strangely enough I copied over the kernel which was working to the non-working SD card, and it also does not work. Could it be some changes I made to the eeprom config

pelwell, 2024-02-06:
> Could it be some changes I made to the eeprom config

It seems highly unlikely. This looks more like a software/configuration issue than a hardware/communication problem, and there's essentially no software configuration carried over from the EEPROM to Linux (unless it's something to with clock rates...)

rafael2k, 2024-02-06:
```
[  172.228156] snd_audioinjector_pi_soundcard_hw_params(rate 96000)
[  172.228808] bcm2835_i2s_hw_params(width 16)
[  172.229068] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[  172.229073] bcm2835-i2s fe203000.i2s: slots: 2 width: 62 rx mask: 0x03 tx_mask: 0x03
[  172.229079] bcm2835-i2s fe203000.i2s: frame len: 125 sync len: 62 data len: 16
[  172.229083] bcm2835-i2s fe203000.i2s: rx pos: 1,63 tx pos: 1,63
[  172.229087] bcm2835-i2s fe203000.i2s: sampling rate: 96000 bclk rate: 12000000
[  172.229091] bcm2835-i2s fe203000.i2s: CLKM: 1 CLKI: 1 FSM: 1 FSI: 1 frame start: falling edge

```

pelwell, 2024-02-06:
That looks like success (apart from the I2S SYNC error) - can you get one of the hw_params errors?

rafael2k, 2024-02-06:
I'm testing now with the same kernel (5928), but in the other SD which is working.

rafael2k, 2024-02-06:
> That looks like success (apart from the I2S SYNC error) - can you get one of the hw_params errors?

Yes. Tomorrow will post here.

pelwell, 2024-02-06:
> Yes. Tomorrow will post here.

I'm not going anywhere.

rafael2k, 2024-02-06:
Tested it was not this.

pelwell, 2024-02-06:
I don't understand your comment.

pelwell, 2024-02-07:
Thanks for confirming my suspicion.

closed 2024-02-07 by pelwell