raspberrypi/linux #5931: [PI5][6.6] act-led no longer properly working

klslz, 2024-02-07:
### Describe the bug

Recent changes on kernel 6.6.14 break earlier working act_led behaviour on the RPI5.

RPi5 act-led related parameters are still not outlined in overlays/README - another issue I guess.
Therefore below setup is based on earlier foundation advise and partial guess-work.

Below  setup - to turn all RPI5 LEDs off - worked until 6.6.13

```
###disable RPi LEDs
dtparam=pwr_led_trigger=default-on
dtparam=pwr_led_activelow=off
dtparam=act_led_trigger=default-on
dtparam=act_led_activelow=off

###disable ethernet port LEDs
dtparam=eth_led0=4
dtparam=eth_led1=4
```

The README seems to say  to enable the dtoverlay=act-led now. I didn't have that parameter in until 6.6.13.
I just tried it and added.

```
dtoverlay=act-led
```

No success. 

What has changed??? Please advice. 
And please introduce RPi5  and related descriptions into the overlays/README accordingly.

### Steps to reproduce the behaviour

See above.

### Device (s)

Raspberry Pi 5

### System

custom 6.6.14-rpi-2712   CL: 0dc58897a

pelwell, 2024-02-07:
> Recent changes on kernel 6.6.14 break earlier working act_led behaviour on the RPI5.

What isn't working for you? You need to be specific in bug reports - what you did, what you expected to happen, and what actually happened.

> RPi5 act-led related parameters are still not outlined in overlays/README - another issue I guess.

Which ones? Everything should be in there, to some extent at least.

pelwell, 2024-02-07:
There is definitely ambiguity and confusion in this area, driven by the fact that two properties of the system are conflated into one.  The concept of active-low can be applied to the electrical circuit that drives the LED and to the relationship between the LED state and the trigger pattern that drives it. Let's call the former _active-low_ and the latter _inverted_.

In a sense the system as a whole only needs to know whether or not there is an inversion between the logical LED state coming from the pattern (on for activity, or on for the heartbeat) and the drive applied to the SoC pin (on->high/off->low or on->low/off->high). Inverting the sense of the pattern (making it off for activity, or off for the beat) can be achieved by flipping the value of that active-low flag, but doing so requires that the user knows whether or not the LED is active-low. The system becomes easier to follow if there are two separate controls - one for the hardware active-low-ness, and one for the pattern inverted-ness. Only users rerouting a LED to an external GPIO need to care about the active-low-ness, whereas anyone wanting the Pi 5 ACT LED to be off on idle (or forcing all Pis to have a non-inverted heartbeat) should only need to clear the inverted flag. Sadly, making that change now will invalidate lots of documentation and tutorials.

klslz, 2024-02-07:
What did you change recently? It used to work for months on the RPi5. 
And it was you who introduced that behaviour if I am not mistaken. 

On the RPi4 and older, a similar setup,  worked for years.

klslz, 2024-02-07:
There's even been a recent commit in that area: 468ad41
Not sure if it is relevant.

On the RPI4 following settings worked to get them OFF

```
###disable power LED (red)
dtparam=pwr_led_activelow=off

###disable activity LED (green)
dtparam=act_led_trigger=none
dtparam=act_led_activelow=off
```

pelwell, 2024-02-07:
Pi 5 was implemented using the active-low Device Tree property, unlike all the other Pis, and it now uses the 3rd cell of the GPIO declaration.
```
dtparam=act_led_trigger=default-on
dtparam=act_led_activelow=on
```
Green LED is off.

pelwell, 2024-02-07:
```
dtparam=act_led_trigger=none
dtparam=act_led_activelow=off
```
Green LED is also off.

klslz, 2024-02-07:
That's correct.  ON worked. OFF still  doesn't. Don't ask me what was different before 6.6.14. 

If you run an inverted logic:

For the power LED. I'd have to change it accordingly then.

```
dtparam=pwr_led_activelow=on
```
Right?

Anyhow. Seems we're getting there.

In any case the overlays/README needs an RP5update.

klslz, 2024-02-07:
Just tested it. pwr_led_active_low still needs to be OFF to turn off - otherwise its on.

pelwell, 2024-02-07:
The power LED defaults are active-low on and trigger "none".

klslz, 2024-02-07:
Just for confirmation:

Below, and in my case - only below, settings

```
###disable RPi5 PWR (red) LED
dtparam=pwr_led_trigger=default-on
dtparam=pwr_led_activelow=off

###disable RPi5 ACT (green) LED
dtparam=act_led_trigger=default-on
dtparam=act_led_activelow=on
```

currently leads to act and pwr LEDs being turned off at boot on a RPi5. 

That'll do for me. If that is how you guys want it to work. So be it.
You mentioned that a lot of docs would have to be updated otherwise. Perhaps you can refer to the documentation that describes this. It's not in the README. ;)

Thx for your efforts.

pelwell, 2024-02-07:
On Pi 5, the RED power LED defaults to off, so you can drop those lines. And:
```
dtparam=act_led_trigger=none
dtparam=act_led_activelow=off
```
is more logical.

klslz, 2024-02-07:
Logic. ;) OK.

If I set the trigger to "none" instead of "default-on" on pwr and act, I can confirm below 
inverted logic will turn both LEDs off at boot.

```
###disable RPi5 PWR (red) LED
dtparam=pwr_led_trigger=none
dtparam=pwr_led_activelow=on

###disable RPi5 ACT (green) LED
dtparam=act_led_trigger=none
dtparam=act_led_activelow=off
```

I can also confirm that the pwr LED settings are optional (default settings) , thus the related lines don't have to be in there.
For better understanding I prefer to have them in there though. Things might change again.

From a simple user logic perspective we look at two different settings for turning a LED OFF now.  
That's not cool. As far as I understood, there's some logic behind that.

Shall we keep this issue open until the README gets updated?

Many applications and projects out there prefer a "dark" RPi.  Great to have it settled and working for the RPi5 now.

Thx.

pelwell, 2024-02-07:
Read my comments above for why the settings are different for different boards - one setting is capturing user preferences and hardware difference. These differences have been there as long as the parameters, just not between boards that you care about. Improving usability will likely end up breaking a proportion of the existing systems.

klslz, 2024-02-07:
I read all your comments. All I am saying. 1. There's a confusing/inconsistent on/off usage from a user perspective by just looking at the params.
2. As long as this stuff changes, and it did change recently, and as long as it is not properly documented - it isn't - you'll have these discussions and issue reports.

Keep in mind. I am spending my free hours to analyse and report these issues. More unpaid hours to get these issues resolved and implemented on my side.  And in the end all this improves your product quality!  What's in for me ???

Anyhow. The (my) actual issue was resolved. That's what matters to me. Product quality (proper documentation)  is on your books.

Have a good one. Case closed.

closed 2024-02-07 by klslz