raspberrypi/linux #5932: DRM does not with the same color models as FB or GL for playback, resulting in vlc defaulting to incorrect color reproduction in Raspberry Pi OS lite

johanehnberg, 2024-02-07:
### Describe the bug

It would seem that the DRM driver does not support, advertise or reproduce some of the possible color models - at least as probed by `vlc` and `cvlc`.

This affects video playback when using RGB source color model(s) such as photo slideshows over DRM, such as is the default case for `cvlc` in Raspberry Pi OS lite.

This does not affect default desktop nor explicitly selecting frame buffer playback on the command line. This also does not affect the more common video color models, i.e. `cvlc video.mp4` use cases are mostly unaffected.

### Steps to reproduce the behaviour

1) Install Raspberry Pi OS (bookworm) lite
2) Boot up and install vlc package
3) Display a colorful PNG image with 8-bit/color RGB with `cvlc image.png` and observe incorrect colors
4) Display the same image with `cvlc --vout fb image.png` and observe correct colors

### Device (s)

Raspberry Pi 3 Mod. B, Raspberry Pi 3 Mod. B+

### System

```
System Information
------------------

Raspberry Pi 3 Model B Rev 1.2
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"

Raspberry Pi reference 2023-12-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 2acf7afcba7d11500313a7b93bb55a2aae20b2d6, stage2

Linux rpi 6.1.0-rpi8-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux
Revision	: a02082
Serial		: 000000001e6c70ba
Model		: Raspberry Pi 3 Model B Rev 1.2
Throttled flag  : throttled=0x50005
Camera          : supported=0 detected=0, libcamera interfaces=0

Videocore information
---------------------

Oct 17 2023 15:42:39 
Copyright (c) 2012 Broadcom
version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
udev              326524       0    326524   0% /dev
tmpfs              93036    1112     91924   2% /run
/dev/mmcblk0p2  14730784 2718568  11244368  20% /
tmpfs             465168       0    465168   0% /dev/shm
tmpfs               5120      12      5108   1% /run/lock
/dev/mmcblk0p1    522232   62390    459842  12% /boot/firmware
tmpfs              93032       0     93032   0% /run/user/1000

Filename				Type		Size		Used		Priority
/var/swap                               file		102396		0		-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: (none)
[... 1005 lines cut ...]
[   13.167281] Bluetooth: hci0: BCM43430A1
[   13.167316] Bluetooth: hci0: BCM43430A1 (001.002.009) build 0000
[   13.200009] Bluetooth: hci0: BCM43430A1 'brcm/BCM43430A1.raspberrypi,3-model-b.hcd' Patch
[   13.207478] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[   13.291205] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   13.291682] usbcore: registered new interface driver brcmfmac
[   13.575041] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Jul 19 2021 03:24:18 version 7.45.98 (TOB) (56df937 CY) FWID 01-8e14b897
[   13.886253] hwmon hwmon1: Undervoltage detected!
[   14.068677] Bluetooth: hci0: BCM: features 0x2e
[   14.072024] Bluetooth: hci0: BCM43438A1 37.4MHz Raspberry Pi 3-0141
[   14.072056] Bluetooth: hci0: BCM43430A1 (001.002.009) build 0508
[   14.814714] input: Logitech M510 as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3.2/1-1.3.2:1.2/0003:046D:C52B.0005/0003:046D:4051.0006/input/input17
[   14.885020] logitech-hidpp-device 0003:046D:4051.0006: input,hidraw3: USB HID v1.11 Mouse [Logitech M510] on usb-3f980000.usb-1.3.2/input2:1
[   15.454241] input: Logitech K540/K545 as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3.2/1-1.3.2:1.2/0003:046D:C52B.0005/0003:046D:4076.0007/input/input18
[   15.457168] logitech-hidpp-device 0003:046D:4076.0007: input,hidraw4: USB HID v1.11 Keyboard [Logitech K540/K545] on usb-3f980000.usb-1.3.2/input2:2
[   16.012035] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   16.023101] smsc95xx 1-1.1:1.0 eth0: Link is Down
[   16.156952] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   16.505769] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   16.505800] Bluetooth: BNEP filters: protocol multicast
[   16.505837] Bluetooth: BNEP socket layer initialized
[   16.520099] Bluetooth: MGMT ver 1.22
[   16.551982] NET: Registered PF_ALG protocol family
[   18.028227] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
```

popcornmix, 2024-02-07:
If the issue only affects VLC, then https://github.com/RPi-Distro/vlc/issues is a better place to report it.

My initial feeling is the fix for this issue will not lie in the linux kernel code, so this isn't the right place to report it.

johanehnberg, 2024-02-07:
Is there any other app that supports drm output than `vlc` to use to cross-check? I checked that none of `ffmpeg`, `mpv` and `mplayer` do.

johanehnberg, 2024-02-07:
I noticed when testing a few images that the color mode may need to be quite specific (RGB, not RGBA). So I am attaching one PNG that exhibits the issue well both in format and color. The correct color is a bluish tint and the incorrect is brown-red.

[image]

popcornmix, 2024-02-07:
kodi will output to drm. Make sure you run the fullscreen rather than desktop option from desktop launcher.

6by9, 2024-02-07:
Sorry, I'd misread that you were viewing fbdev as being the incorrect rendering.

I suspect that the VLC drm_vout module is getting the mapping wrong to DRM_FORMAT_RGB888 and DRM_FORMAT_BGR888.
My PNG had alpha, hence was mapping to DRM_FORMAT_ABGR8888.

All the formats display consistently when tested with kmstest, so I don't believe the driver has an error there.

johanehnberg, 2024-02-07:
I cannot find a way to set kodi to use drm. It seems to use gl driver by default.

popcornmix, 2024-02-07:
On RPiOS bookworm, you have two launcher icons from desktop:
Kodi (uses egl)
Kodi Fullscreen (uses drm).

jc-kynesim, 2024-02-07:
Yup - that's broken - I've clearly got a broken mapping. Will fix.

6by9, 2024-02-07:
@jc-kynesim Is that plugin your code or mainline VLC? 

Mainline VLC appears to have DRM files in https://github.com/videolan/vlc/tree/master/modules/video_output/drm, but I don't see them in https://github.com/RPi-Distro/vlc/tree/bookworm-rpt/modules/video_output. Is that just because Debian is on 3.0.x, and the new shiny stuff hasn't made a release yet? Is it worth backporting rather than maintaining your module?

jc-kynesim, 2024-02-07:
Fixed - VLC swaps its byte ordering convention in format masks for 24 bit formats... I built my tables expecting it to be the same for all. Trivial fix but I needed to spend some time to convince myself that it was them going crazy not me and my other formats were right.

popcornmix, 2024-02-19:
Closing as this wasn't a kernel issue, and it sounds like VLC has implemented a fix.

closed 2024-02-19 by popcornmix