raspberrypi/linux #5939: Kernel 6.16 aarch64 pi5 fan does not stop at idle

Dark-Sky, 2024-02-09:
### Describe the bug

I have noticed since kernel 6.16 upgrade my pi5 fan does not stop any more when idling @1500 MHz like it used to. It does stop when idling with the 6.8-rc3 kernel. I have tried both `ondemand` and `schedutil` cpu govenors. The latest fan commit today still does not make any difference.

### Steps to reproduce the behaviour

Kernel built with `make bcm2712_defconfig`

### Device (s)

Raspberry Pi 5

### System

[ray@jellyfin ~]$ vcgencmd version
2024/01/05 15:57:40 
Copyright (c) 2012 Broadcom
version 30cc5f37 (release) (embedded)

Linux jellyfin 6.6.16-2-MANJARO-RPI5 #1 SMP PREEMPT Fri Feb  9 17:42:25 UTC 2024 aarch64 GNU/Linux

pelwell, 2024-02-09:
Does the temperature drop below 45C?
```
$ while true; do
echo $(cat /sys/class/thermal/thermal_zone0/temp) $(cat /sys/class/thermal/cooling_device0/cur_state)
sleep 1
done
```
(Those locations are from memory, so be prepared to search a bit)

Dark-Sky, 2024-02-10:
I finally saw the fan stop. Most of the time the temp is 46 to 48 C. When 45C shows up the fan stops and then I see 45 to 48 with the fan stopped but just moving the mouse seems to start the fan back up again. Really do not remember it being that touchy in the past. I guess it could be affected by ambient temperature. The days here are staring to pretty warm the last few days.

Dark-Sky, 2024-02-12:
I am going to chalk this up to ambient temperature. The last couple of days has been cool here and the fan has been acting like it used to.

pelwell, 2024-02-12:
That's good to hear.

closed 2024-02-12 by pelwell