raspberrypi/linux #5944: CFE driver complains in v4l2_subdev_link_validate()

naushir, 2024-02-12:
### Describe the bug

With recent changes to the CFE driver, when running the camera, dmesg throws out the following error:

```
source of link 'rp1-cfe-fe_config':0->'pisp-fe':1 is not a V4L2 sub-device, driver bug!
```

This is expected, as ``rp1-cfe-fe_config`` is a device node:

[image]

I cannot see how to get rid of this log message.

### Steps to reproduce the behaviour

Run any camera on Pi 5.

### Device (s)

Raspberry Pi 5

### System

Using kernel rpi-6.6.y

6by9, 2024-02-12:
Likely one to query with Sakari as he introduced it through 55f1ecb1199000932cf82e357841cc7498ac904f

tomba, 2024-02-12:
In my WIP streams-enabled version I have a pisp_fe_link_validate() which calls the normal v4l2_subdev_link_validate() for FE_STREAM_PAD, but skips the validation for FE_CONFIG_PAD.

tomba, 2024-02-12:
```

static int pisp_fe_link_validate(struct media_link *link)
{
	struct v4l2_subdev *sd = media_entity_to_v4l2_subdev(link->sink->entity);
	struct pisp_fe_device *fe = container_of(sd, struct pisp_fe_device, sd);

	pisp_fe_dbg("%s: link \"%s\":%u -> \"%s\":%u\n", __func__,
		    link->source->entity->name, link->source->index,
		    link->sink->entity->name, link->sink->index);

	if (link->sink->index == FE_STREAM_PAD)
		return v4l2_subdev_link_validate(link);

	if (link->sink->index == FE_CONFIG_PAD)
		return 0;

	return -EINVAL;
}

static const struct media_entity_operations pisp_fe_entity_ops = {
	.link_validate = pisp_fe_link_validate,
};
```

naushir, 2024-02-12:
Great, ill ignore this problem as your changes will fix the log message!