raspberrypi/linux #5946: raspberrypi_axi_monitor shows activity on unused peripheral

cibomahto, 2024-02-12:
### Describe the bug

When attempting to profile a system using raspberrypi_axi_monitor, bus traffic appears to be reported from the wrong bus.

Specifically, traffic that should have originated from the CPU is being shown as originating from the H264 block.

### Steps to reproduce the behaviour

1. Use Rasberry Pi Imager to prepare an SD card with the 'Rasbperry Pi OS Lite (64-bit)' image. Use the custom settings to enable SSH, and set a user/password
2. Place a CM4 (RPI-CM4Lite-R1-3' in the IO board. Connect the ethernet switch as 12V power.
3. Boot the device
4. Once it is on the network, ssh to it and add 'dtparam=axiperf=on' to /boot/config.txt
5. Reboot the device
6. SSH to the device and run 'sudo apt update; sudo apt install stress-ng' to install stress-ng -y
7. Run 'cat /sys/kernel/debug/raspberrypi_axi_monitor/System/data' to read the performance stats: 

     Bus   |    Atrans    Atwait      AMax    Wtrans    Wtwait      WMax    Rtrans    Rtwait      RMax
======================================================================================================
    DMA_L2 |        0K        0K        0K        0K        0K        0K        0K        0K        0K
     TRANS |        0K        0K        0K        0K        0K        0K        0K        0K        0K
      JPEG |        0K        0K        0K        0K        0K        0K        0K        0K        0K
 SYSTEM_UC |        0K        0K        0K        0K        0K        0K        0K        0K        0K
    DMA_UC |        0K        0K        0K        0K        0K        0K        0K        0K        0K
 SYSTEM_L2 |        0K        0K        0K        0K        0K        0K        0K        0K        0K
    CCP2TX |        0K        0K        0K        0K        0K        0K        0K        0K        0K
   MPHI_RX |        0K        0K        0K        0K        0K        0K        0K        0K        0K
   MPHI_TX |        0K        0K        0K        0K        0K        0K        0K        0K        0K
       HVS |       15K        0K        0K        0K        0K        0K       14K        0K        0K
      H264 |        4K        1K        0K        6K        1K        0K        9K        1K        0K
       ISP |        0K        0K        0K        0K        0K        0K        0K        0K        0K
       V3D |        0K        0K        0K        0K        0K        0K        0K        0K        0K
PERIPHERAL |        0K        0K        0K        0K        0K        0K        0K        0K        0K
    CPU_UC |        0K        0K        0K        0K        0K        0K        0K        0K        0K
    CPU_L2 |        0K        0K        0K        0K        0K        0K        0K        0K        0K

8. In a seperate terminal, ssh to the device again, then run 'stress-ng --memcpy 0' to generate memory traffic from the CPU
9. In the first terminal, run 'cat /sys/kernel/debug/raspberrypi_axi_monitor/System/data' to read the performance stats again:

     Bus   |    Atrans    Atwait      AMax    Wtrans    Wtwait      WMax    Rtrans    Rtwait      RMax
======================================================================================================
    DMA_L2 |        0K        0K        0K        0K        0K        0K        0K        0K        0K
     TRANS |        0K        0K        0K        0K        0K        0K        0K        0K        0K
      JPEG |        0K        0K        0K        0K        0K        0K        0K        0K        0K
 SYSTEM_UC |        0K        7K        0K        0K        3K        0K        0K        0K        0K
    DMA_UC |        0K        0K        0K        0K        0K        0K        0K        0K        0K
 SYSTEM_L2 |        0K        0K        0K        0K        0K        0K        0K        0K        0K
    CCP2TX |        0K        0K        0K        0K        0K        0K        0K        0K        0K
   MPHI_RX |        0K        0K        0K        0K        0K        0K        0K        0K        0K
   MPHI_TX |        0K        0K        0K        0K        0K        0K        0K        0K        0K
       HVS |       15K        0K        0K        0K        0K        0K       14K        0K        0K
      H264 |     4803K    19203K        0K    10630K    20777K        0K     8581K     1129K        0K
       ISP |        0K        0K        0K        0K        0K        0K        0K        0K        0K
       V3D |        0K        0K        0K        0K        0K        0K        0K        0K        0K
PERIPHERAL |        0K        0K        0K        0K        0K        0K        0K        0K        0K
    CPU_UC |        0K        0K        0K        0K        0K        0K        0K        0K        0K
    CPU_L2 |        0K        0K        0K        0K        0K        0K        0K        0K        0K

A large amount of traffic is reported from the H264 block, instead of one of the CPU bus sources.

10. Kill stress-ng in the second terminal, and then cat the performance stats again- the overall bus traffic should be close to 0 again.

### Device (s)

Raspberry Pi CM4 Lite

### System

cat /etc/rpi-issue
Raspberry Pi reference 2023-12-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 2acf7afcba7d11500313a7b93bb55a2aae20b2d6, stage2

vcgencmd version
Oct 17 2023 15:39:16
Copyright (c) 2012 Broadcom
version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start)

uname -a
Linux cm4 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux

cibomahto, 2024-02-12:
After reporting this, I was able to repeat the issue on a Raspberry Pi 4b. I wasn't able to enable raspberrypi_axi_monitor on a 5.

On a 2B, it appears to show the correct results:

```
sudo cat /sys/kernel/debug/raspberrypi_axi_monitor/System/data
     Bus   |    Atrans    Atwait      AMax    Wtrans    Wtwait      WMax    Rtrans    Rtwait      RMax
======================================================================================================
    DMA_L2 |        0K        0K        0K        0K        0K        0K        0K        0K        0K
     TRANS |        0K        0K        0K        0K        0K        0K        0K        0K        0K
      JPEG |        0K        0K        0K        0K        0K        0K        0K        0K        0K
 SYSTEM_UC |     3429K       27K        0K     5511K      183K        0K     7276K     4948K        0K
    DMA_UC |        0K        0K        0K        0K        0K        0K        0K        0K        0K
 SYSTEM_L2 |        0K        0K        0K        0K        0K        0K        0K        0K        0K
    CCP2TX |        0K        0K        0K        0K        0K        0K        0K        0K        0K
   MPHI_RX |        0K        0K        0K        0K        0K        0K        0K        0K        0K
   MPHI_TX |        0K        0K        0K        0K        0K        0K        0K        0K        0K
       HVS |        0K        0K        0K        0K        0K        0K        0K        0K        0K
      H264 |        0K        0K        0K        0K        0K        0K        0K        0K        0K
       ISP |        0K        0K        0K        0K        0K        0K        0K        0K        0K
       V3D |        0K        0K        0K        0K        0K        0K        0K        0K        0K
PERIPHERAL |        7K        0K        0K        2K        0K        0K        5K        0K        0K
    CPU_UC |     3446K     1007K        0K     5480K        0K        0K     7148K     6563K        0K
    CPU_L2 |        0K        0K        0K        0K        0K        0K        0K        0K        0K
```

cibomahto, 2024-02-21:
> If you wait about an hour for github CI to complete, you can test a potential fix with:
> 
> ```
> sudo rpi-update pulls 5971
> ```

Thanks! just tried this with a fresh raspberry pi lite 64bit image on an RPI4, with an updated bootloader. I get 'Invalid hash given':

```
test@test:~ $ sudo rpi-update pulls 5971
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
FW_REV:
BOOTLOADER_REV:
 *** Invalid hash given
```

popcornmix, 2024-02-21:
Sorry, missed a `\`:
```
sudo rpi-update pulls/5971
```

cibomahto, 2024-02-21:
That got a little further, but still no luck:

```
sudo rpi-update pulls/5971
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
FW_REV:
BOOTLOADER_REV:36e58db5c2a2656e553441f4f48f32227809105d
WANT_32BIT:0 WANT_64BIT:1 WANT_PI4:1 WANT_PI5:1
 *** Downloading specific artifact revision (this will take a few minutes)
curl  -L https://builds.raspberrypi.com/github/linux/e997061cbc9e5ca3dfdd2444876be2d79b2a26c2/bcmrpi | zcat | tar xf - -C //root/.rpi-firmware --strip-components=2
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:03 --:--:--     0

gzip: stdin: unexpected end of file
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:03 --:--:--     0
Invalid artifact specified. Response: 404.
```

 I'm not familiar with how your build system works, but it looks like the arm64 build might have been cancelled:

[image]

popcornmix, 2024-02-21:
I wonder if github gets upset if there's a push while it's building?

I've rebased which looks like it's triggered a rebuild - hopefully there will be a complete build within the hour.

cibomahto, 2024-02-21:
Looks like that was the trick, the monitor output looks more believable now.

Using 'stress-ng --memcpy 0':
```
test@test:~ $ cat /sys/kernel/debug/raspberrypi_axi_monitor/System/data
     Bus    |    Atrans    Atwait      AMax    Wtrans    Wtwait      WMax    Rtrans    Rtwait      RMax     RPend   RAtrans
===========================================================================================================================
     DMA_L2 |        0         0         0         0         0         0         0         0         0         0      1096M
      TRANS |        0         0         0         0         0         0         0         0         0         0      1096M
       JPEG |        0         0         0         0         0         0         0         0         0         0      1096M
     VPU_UC |      558        13K        1K      282         5K      978       355         0         0         0      1096M
     DMA_UC |        0         0         0         0         0         0         0         0         0         0      1096M
  SYSTEM_L2 |        0         0         0         0         0         0         0         0         0         0      1096M
        HVS |        0         0         0         0         0         0         0         0         0         0      1096M
      ARGON |        0         0         0         0         0         0         0         0         0         0      1096M
       H264 |        0         0         0         0         0         0         0         0         0         0      1096M
 PERIPHERAL |       13K        0         0       841         0         0        12K        0         0         1      1096M
     ARM_UC |     4517K    24813K        1K     9323K    25270K        1K     8745K     1199K        1         1      1096M
     ARM_L2 |        0         0         0         0         0         0         0         0         0         0      1096M
```

popcornmix, 2024-02-21:
Good to hear. It should also be functional on Pi5 now.

cibomahto, 2024-03-11:
@popcornmix I'm happy with the resolution- should I close the issue?

popcornmix, 2024-03-11:
Yes - go ahead.

closed 2024-03-11 by cibomahto