raspberrypi/linux #5948: Pi5 Composite video problem

ArtemKorr, 2024-02-12:
### Describe the bug

The encoder does not start when setting the video output to the composite connector.
```
kmsprint
Connector 0 (44) Composite-1 (connected)
  Encoder 0 (48) NONE
    Crtc 0 (47) 704x432i@59.98 15.429 704/72/72/132/? 432/31/6/56/? 60 (59.98) P|D 
      Plane 0 (45) fb-id: 49 (crtcs: 0) 0,0 704x432 -> 0,0 704x432 (XR24 XB24 RG24 BG24 RG16)
        FB 49 704x432 XR24

```
```
kmsprint 
Connector 0 (32) HDMI-A-1 (connected)
  Encoder 0 (31) TMDS
    Crtc 3 (96) 2560x1440@59.95 241.500 2560/48/32/80/+ 1440/3/5/33/- 60 (59.95) U|D 
      Plane 3 (86) fb-id: 339 (crtcs: 3) 0,0 2560x1440 -> 0,0 2560x1440 (XR24 AR24 AB24 XB24 RG16 BG16 AR15 XR15 RG24 BG24 YU16 YV16 YU12 YV12 NV12 NV21 NV16 NV61 P030 XR30 AR30 AB30 XB30 RGB8 BGR8 XR12 AR12 XB12 AB12 BX12 BA12 RX12 RA12)
        FB 339 2560x1440 RG16
```
		
As I understand it, because of this, it is not possible to utilize DRM preview in picamera2 in console mode works (which is what I am trying to use it for).
I should also note that the display of analog video occurs without any problems both in desktop mode and in console mode.

### Steps to reproduce the behaviour

I've tried on many kernel versions available here. I also attempted to install the kernel version where [fkms-pi5](https://github.com/raspberrypi/linux/pull/5669#issuecomment-1773079340) is compiled, but it did not produce the expected effect. 
On the Raspberry Pi 4 with composite video, everything works perfectly.
As of now, my environment configuration is provided below. I hope that someone will help me in solving and understanding this issue.

### Device (s)

Raspberry Pi 5

### System

```
cat /etc/rpi-issue
Raspberry Pi reference 2023-12-05
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 70cd6f2a1e34d07f5cba7047aea5b92457372e05, stage4
```
```
vcgencmd version
2024/02/08 11:34:47 
Copyright (c) 2012 Broadcom
version 42e0a50b (release) (embedded)
```
`Linux raspberrypi 6.1.58-v8_16k+ #1693 SMP PREEMPT Tue Oct 24 12:17:49 BST 2023 aarch64 GNU/Linux`

### Logs

```
[    5.700582] [drm] Initialized drm-rp1-vec 1.0.0 0 for 1f00144000.vec on minor 1
[    5.704807] drm-rp1-vec 1f00144000.vec: rp1vec_pipe_enable
[    5.704815] drm-rp1-vec 1f00144000.vec: rp1vec_hw_setup: in_fmt='XR24' mode=704x432i [011] tvstd=7 ()
[    5.704877] Console: switching to colour frame buffer device 88x27
[    5.714279] pispbe 1000880000.pisp_be: output0 device node registered as /dev/video33
[    5.737810] pispbe 1000880000.pisp_be: output1 device node registered as /dev/video34
[    5.738728] drm-rp1-vec 1f00144000.vec: [drm] fb0: drm-rp1-vecdrmf frame buffer device
[    5.744995] pispbe 1000880000.pisp_be: tdn_output device node registered as /dev/video35
[    5.758878] pispbe 1000880000.pisp_be: stitch_output device node registered as /dev/video36
[    5.758980] pispbe 1000880000.pisp_be: config device node registered as /dev/video37
[    5.772992] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    5.784792] cfg80211: Loaded X.509 cert 'benh@debian.org: 577e021cb980e0e820821ba7b54b4961b8b4fadf'
[    5.785151] cfg80211: Loaded X.509 cert 'romain.perier@gmail.com: 3abbc6ec146e09d1b6016ab9d6cf71dd233f0328'
[    5.785477] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    5.835077] brcmfmac: F1 signature read @0x18000000=0x15264345
[    5.838864] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.856991] usbcore: registered new interface driver brcmfmac
[    5.980906] vc4-drm axi:gpu: bcm2712_iommu_of_xlate: MMU 1000005200.iommu
[    5.996546] vc4-drm axi:gpu: bound 107c580000.hvs (ops vc4_hvs_ops [vc4])
[    5.997815] vc4-drm axi:gpu: bound 107c500000.mop (ops vc4_txp_ops [vc4])
[    6.013886] vc4-drm axi:gpu: bound 107c501000.moplet (ops vc4_txp_ops [vc4])
[    6.026824] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Nov  1 2021 00:37:25 version 7.45.241 (1a2f2fa CY) FWID 01-703fd60
[    6.034628] [drm] Initialized vc4 0.0.0 20140616 for axi:gpu on minor 2
[    6.034655] vc4-drm axi:gpu: [drm] Cannot find any crtc or sizes
[    6.049702] ieee80211 phy0: brcmf_cfg80211_reg_notifier: Firmware rejected country setting
[    6.519939] Adding 3642352k swap on /var/swap.  Priority:-2 extents:28 across:14516224k SSFS
[    6.723091] imx708 4-001a: failed to read chip id 708, with error -5
[    6.723240] dw9807 4-000c: I2C write STATUS address fail ret = -121
[    6.723246] dw9807 4-000c: dw9807_ramp I2C failure: -121
[    6.725246] dw9807 4-000c: I2C write STATUS address fail ret = -121
[    6.725256] dw9807 4-000c: dw9807_ramp I2C failure: -121
[    6.725380] dw9807 4-000c: I2C write CTL fail ret = -121
[    6.725681] imx708: probe of 4-001a failed with error -5
[    7.270574] macb 1f00100000.ethernet eth0: PHY [1f00100000.ethernet-ffffffff:01] driver [Broadcom BCM54213PE] (irq=POLL)
[    7.270586] macb 1f00100000.ethernet eth0: configuring for phy/rgmii-id link mode
[    7.273668] pps pps0: new PPS source ptp0
[    7.273746] macb 1f00100000.ethernet: gem-ptp-timer ptp clock registered.
[    7.307053] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   10.345376] macb 1f00100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[   10.345398] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   13.229465] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   19.327601] pl011-axi 1f00040000.serial: no DMA platform data
```

### Additional context

cat /boot/config.txt
```
disable_overscan=1
suddtparam=audio=on
camera_auto_detect=1
dtoverlay=imx708
arm_64bit=1
arm_boost=1
dtoverlay=vc4-kms-v3d-pi5,composite=1
max_framebuffers=2
enable_uart=1
dtoverlay=uart3-pi5
dtoverlay=uart4-pi5
dtoverlay=disable-bt
```
cat /boot/cmdline.txt
root=PARTUUID=7d3bf79a-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=UA

6by9, 2024-02-13:
> The encoder does not start when setting the video output to the composite connector.

Just for clarity, which encoder are you referring to here? DRM encoder, or a video codec?
Your kmsprint output implies that the DRM encoder is running correctly, however on my system it does appear that it doesn't initially start up. kmsprint quitting triggers `drm_lastclose` which establishes the framebuffer emulation and wakes everything up. I do remember looking at this one before, but not coming to a conclusion.

On Pi0-4 the analogue video output is driven as part of the vc4 display pipeline. This supports numerous image formats and scaling.
On Pi5 the analogue video output is driven from RP1, and is a far simpler block which only takes XRGB8888, XBGR8888, RGB888, BGR888, or RGB565 images without scaling.

Picamera2 preview tries creating an AB24 (DRM_FORMAT_ABGR8888) buffer in https://github.com/raspberrypi/picamera2/blob/main/picamera2/previews/drm_preview.py#L58, which for the RP1 devices will fail. It appears to only be used for MJPEG display, which we won't do. It does then assume it can use an overlay plane, and there are none on the RP1 devices.

@naushir @davidplowman We had this discussion before. Can we revisit it please? 

```
[    6.723091] imx708 4-001a: failed to read chip id 708, with error -5
[    6.723240] dw9807 4-000c: I2C write STATUS address fail ret = -121
[    6.723246] dw9807 4-000c: dw9807_ramp I2C failure: -121
[    6.725246] dw9807 4-000c: I2C write STATUS address fail ret = -121
[    6.725256] dw9807 4-000c: dw9807_ramp I2C failure: -121
[    6.725380] dw9807 4-000c: I2C write CTL fail ret = -121
[    6.725681] imx708: probe of 4-001a failed with error -5
```
Says it failed to find an imx708 on CAM/DISP1.
```
camera_auto_detect=1
dtoverlay=imx708
```
If your camera is working, then it's probably on CAM/DISP0, and the auto detect is doing the right thing with your dtoverlay being redundant.

ArtemKorr, 2024-02-13:
Regarding the coder, what I meant is that it does not display when the kmsprint command is called, and on pi5, it shows None, which led me to believe that it does not work in kms, or does not start, or something else.
 For instance, on other versions, I saw TBDAC on the composite output, which made me think that it was not starting, and as a result, some formats are not supported as seen with the HDMI output (AR24 AB24). 
 Yes, I understand that in pi2, the code for frame buffer allocation is set specifically to AR24. For the pi5 version with composite output, I tried setting it to what it supported (as shown to me by the kmsprint command) XR24, but the buffer still would not initialize. 
The last thing I tried was thinking that fkms running on pi5 could support more formats. I tried your commit, but that did not help. I'm out of ideas why it's not working, so I'm hoping for your help, advice on how to get the DRM preview up, specifically on the composite output and in console mode. I apologize if I'm writing something silly, I just haven't had such close dealings with the kernel before :) 

P.S. Regarding camera initiation, yes, I simply forgot to leave that command, but there are no problems with the cameras. The camera initiates well. However, it's important to start the preview that uses DRM directly.

6by9, 2024-02-13:
FKMS is not supported on Pi5 - it was a very quick hack for bringing up the platform. It also only controls the equivalent of the vc4 block, so HDMI outputs only (not RP1).

The VEC, DPI, and DSI drivers for RP1 all use the drm_simple_kms_helper, and that always registers the encoder as type NONE - https://github.com/raspberrypi/linux/blob/rpi-6.6.y/drivers/gpu/drm/drm_simple_kms_helper.c#L448

ArtemKorr, 2024-02-13:
Okay, I understand, so the problem is not that it writes None. We need to make it fill the framebuffer with a new data format different from AB24 ?

6by9, 2024-02-13:
On my system from a fresh boot with `dtoverlay=vc4-kms-v3d,composite`, the composite output is disabled initially.

Run `kmsprint`, and as it quits it will trigger the frame buffer emulation to enable the output. (Running `kmsprint` a second time confirms this).

The firmware has left a simpledrm node active on the HDMI display, and that is likely to have claimed /dev/fb0. For me, `cat /dev/urandom > /dev/fb1` fills the composite output with noise.

Getting libcamera to run on those outputs is a different task.

6by9, 2024-02-13:
See https://github.com/raspberrypi/picamera2/issues/922 for the previous discussion over libcamera on SPI (or non-vc4) DRM devices.

ArtemKorr, 2024-02-13:
Okay. I'll try to fill the composite output with noise on my side. 
Thank you, in general, what is needed is a correct DRM implementation for Pi5 that takes into account the new specifics of DRM KMS operation and its display formats.
I saw that issue and thought it wasn't my case because it involves a different display.

closed 2024-02-13 by ArtemKorr

6by9, 2024-02-13:
It's an updated drm_preview module in picamera2 and rpicam-apps that is required that use an appropriate format and plane for the rendering device.

The DRM kernel driver is doing the correct thing for the capabilities of the hardware.