raspberrypi/linux #5950: No HDMI audio in Google Meet via Pipewire/PulseAudio on Bookworm

mghadam, 2024-02-13:
### Describe the bug

Google Meet fails to output audio on Raspberry Pi running Bookworm. Despite successful speaker tests within Google Meet settings, no audio is heard through the TV speakers via HDMI. Interestingly, other platforms like Jitsi Meet provide audio output via HDMI, and VLC functions properly with audio playback.

The issue persists when using Firefox as well. Switching from the default Pipewire to PulseAudio in `raspi-config` does not resolve the problem.

Attempts to resolve the issue by changing the `default args.AES3` parameter as suggested in [this GitHub thread](https://github.com/raspberrypi/linux/issues/5525#issuecomment-1615922687) from `0x02` to `0x01` and rebooting proved unsuccessful.

Analog audio has been disabled via dtparam=audio=off, and the default audio device is set to HDMI by right clicking on the speaker icon on top right side of the desktop.

Interestingly, swapping the SD Card to Debian Bullseye and conducting the same Google Meet session on Debian 11 results in functional HDMI audio without any issue.

### Steps to reproduce the behaviour

1. Connect a Raspberry Pi running Bookworm to a TV via HDMI and attach a USB camera with a microphone.
2. Open Google Meet in Chrome or Firefox on the Raspberry Pi.
3. Add another participant to the meet session.
4. The participant's audio cannot be heard on the Raspberry Pi. 

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry Pi reference 2023-12-05
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 70cd6f2a1e34d07f5cba7047aea5b92457372e05, stage5

version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start)

Linux home.domain.local 6.1.0-rpi8-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux

popcornmix, 2024-02-13:
This doesn't sound like a kernel issue to me, but we'll leave it here until more information is found.
@cillian64 any thoughts?

qrp73, 2024-02-16:
here is the same issue: https://github.com/raspberrypi/bookworm-feedback/issues/205

pelwell, 2024-02-16:
Discussion can continue on the new thread.

closed 2024-02-16 by pelwell

mghadam, 2024-02-16:
> here is the same issue: [raspberrypi/bookworm-feedback#205](https://github.com/raspberrypi/bookworm-feedback/issues/205)

It is not the same issue, here we do not have any output audio in Google meet while other programs or websites have outgoing audio. 

I opened a new feedback at https://github.com/raspberrypi/bookworm-feedback/issues/211 to see if we get more information on it