raspberrypi/linux #5951: Pipewire restarts when hdmi tv is turned on

mghadam, 2024-02-13:
### Describe the bug

When using the microphone on Google Meet or another meeting platform, if an HDMI device like a TV is turned off/on, PulseWire restarts, causing the microphone/speaker in use to become unavailable. This issue does not occur when using PulseAudio.

### Steps to reproduce the behaviour

1. Open a meeting in Google Meet or Jitsi that utilizes the speakers/microphone connected to a Raspberry Pi.
2. Turn off/on the TV that is connected to the Raspberry Pi.
3. After switching the TV back on, the microphone/speaker device becomes unavailable in the meeting platform.

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry Pi reference 2023-12-05
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 70cd6f2a1e34d07f5cba7047aea5b92457372e05, stage5

version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start)

Linux home.domain.local 6.1.0-rpi8-rpi-v8 https://github.com/raspberrypi/linux/issues/1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux

### Logs

Feb 13 08:07:43 home.domain.local sh[963]: Warning: ../src/main.c: 319: Selected output HDMI-A-1 went away
Feb 13 08:07:43 home.domain.local sh[963]: Warning: ../src/main.c: 330: No fallback outputs left. Detaching...
Feb 13 08:07:44 home.domain.local xdg-desktop-portal-wlr[14124]: 2024/02/13 08:07:44 [ERROR] - pipewire: fatal error event from core
Feb 13 08:07:44 home.domain.local xdg-desktop-por[2654]: Caught PipeWire error: connection error
Feb 13 08:07:44 home.domain.local systemd[1304]: Stopping pipewire.service - PipeWire Multimedia Service...
Feb 13 08:07:44 home.domain.local systemd[1304]: Stopped pipewire.service - PipeWire Multimedia Service.
Feb 13 08:07:44 home.domain.local systemd[1304]: xdg-desktop-portal-wlr.service: Main process exited, code=exited, status=1/FAILURE
Feb 13 08:07:44 home.domain.local systemd[1304]: xdg-desktop-portal-wlr.service: Failed with result 'exit-code'.
Feb 13 08:07:44 home.domain.local systemd[1304]: Started pipewire.service - PipeWire Multimedia Service.
Feb 13 08:07:44 home.domain.local systemd[1304]: xdg-desktop-portal-wlr.service: Scheduled restart job, restart counter is at 5.
Feb 13 08:07:44 home.domain.local systemd[1304]: Stopped xdg-desktop-portal-wlr.service - Portal service (wlroots implementation).
Feb 13 08:07:44 home.domain.local systemd[1304]: Starting xdg-desktop-portal-wlr.service - Portal service (wlroots implementation)...
Feb 13 08:07:44 home.domain.local rtkit-daemon[1579]: Successfully made thread 16143 of process 16137 owned by '1000' high priority at nice level -11.
Feb 13 08:07:44 home.domain.local rtkit-daemon[1579]: Supervising 4 threads of 2 processes of 1 users.
Feb 13 08:07:44 home.domain.local xdg-desktop-portal-wlr[16152]: 2024/02/13 08:07:44 [ERROR] - xdpw: Could not find render node
Feb 13 08:07:44 home.domain.local xdg-desktop-portal-wlr[16152]: 2024/02/13 08:07:44 [ERROR] - xdpw: Could not find render node
Feb 13 08:07:45 home.domain.local rtkit-daemon[1579]: Successfully made thread 16144 of process 16137 owned by '1000' RT at priority 20.
Feb 13 08:07:45 home.domain.local rtkit-daemon[1579]: Supervising 5 threads of 2 processes of 1 users.
Feb 13 08:07:45 home.domain.local systemd[1304]: Started xdg-desktop-portal-wlr.service - Portal service (wlroots implementation).
Feb 13 08:07:45 home.domain.local systemd[1304]: Stopping pipewire.service - PipeWire Multimedia Service...
Feb 13 08:07:45 home.domain.local xdg-desktop-por[2654]: Caught PipeWire error: connection error
Feb 13 08:07:45 home.domain.local xdg-desktop-portal-wlr[16152]: 2024/02/13 08:07:45 [ERROR] - pipewire: fatal error event from core
Feb 13 08:07:45 home.domain.local systemd[1304]: Stopped pipewire.service - PipeWire Multimedia Service.
Feb 13 08:07:45 home.domain.local systemd[1304]: xdg-desktop-portal-wlr.service: Main process exited, code=exited, status=1/FAILURE
Feb 13 08:07:45 home.domain.local systemd[1304]: xdg-desktop-portal-wlr.service: Failed with result 'exit-code'.
Feb 13 08:07:46 home.domain.local systemd[1304]: Started pipewire.service - PipeWire Multimedia Service.
Feb 13 08:07:46 home.domain.local systemd[1304]: xdg-desktop-portal-wlr.service: Scheduled restart job, restart counter is at 6.
Feb 13 08:07:46 home.domain.local systemd[1304]: Stopped xdg-desktop-portal-wlr.service - Portal service (wlroots implementation).
Feb 13 08:07:46 home.domain.local systemd[1304]: Starting xdg-desktop-portal-wlr.service - Portal service (wlroots implementation)...
Feb 13 08:07:46 home.domain.local xdg-desktop-portal-wlr[16179]: 2024/02/13 08:07:46 [ERROR] - xdpw: Could not find render node
Feb 13 08:07:46 home.domain.local xdg-desktop-portal-wlr[16179]: 2024/02/13 08:07:46 [ERROR] - xdpw: Could not find render node

apassu, 2024-02-14:
I have a very similar problem with quite the same logs.
I have a RPI 4 with an external HDMI monitor (with Alexa builts in the monitor) and when Alexa activates itself there is a crash of pipewire/wireplumber.

**SYSTEMS INFO:**

Raspberry PI 4
Raspbian 12 (bookworm)
Linux rpi 6.1.0-rpi8-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux
HDMI Monitor Samsung Smart Monitor M7 43" (with Alexa built in)

**Steps to reproduce the behaviour**
The monitor has Alexa built in and when it activates itself there is a crash of pipewire/wireplumber.

**LOGS:**

2024-02-14T14:13:51.581663+01:00 rpi sh[4488]: Warning: ../src/main.c: 319: Selected output HDMI-A-1 went away
2024-02-14T14:13:51.585451+01:00 rpi sh[4488]: Warning: ../src/main.c: 330: No fallback outputs left. Detaching...
2024-02-14T14:13:52.479597+01:00 rpi wireplumber[5848]: stopped by signal: Terminated
2024-02-14T14:13:52.485894+01:00 rpi systemd[1750]: Stopping wireplumber.service - Multimedia Service Session Manager...
2024-02-14T14:13:52.537204+01:00 rpi wireplumber[5848]: disconnected from pipewire
2024-02-14T14:13:52.574138+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSource/ldac
2024-02-14T14:13:52.575006+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSink/aptx_hd
2024-02-14T14:13:52.576500+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSource/aptx_hd
2024-02-14T14:13:52.577591+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSink/aptx
2024-02-14T14:13:52.578691+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSource/aptx
2024-02-14T14:13:52.580101+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSink/sbc
2024-02-14T14:13:52.581530+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSource/sbc
2024-02-14T14:13:52.582848+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSink/sbc_xq
2024-02-14T14:13:52.584948+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSource/sbc_xq
2024-02-14T14:13:52.586777+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSource/aptx_ll_1
2024-02-14T14:13:52.588349+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSource/aptx_ll_0
2024-02-14T14:13:52.589798+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_1
2024-02-14T14:13:52.591290+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_0
2024-02-14T14:13:52.594042+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSource/faststream
2024-02-14T14:13:52.596998+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSource/faststream_duplex
2024-02-14T14:13:52.600100+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSink/opus_05
2024-02-14T14:13:52.601406+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSource/opus_05
2024-02-14T14:13:52.603212+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSink/opus_05_duplex
2024-02-14T14:13:52.603778+01:00 rpi systemd[1750]: Stopped wireplumber.service - Multimedia Service Session Manager.
2024-02-14T14:13:52.604164+01:00 rpi systemd[1750]: wireplumber.service: Consumed 1.551s CPU time.
2024-02-14T14:13:52.605368+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSource/opus_05_duplex
2024-02-14T14:13:52.609849+01:00 rpi systemd[1750]: Stopping pipewire.service - PipeWire Multimedia Service...
2024-02-14T14:13:52.637246+01:00 rpi xdg-desktop-portal-wlr[5866]: 2024/02/14 14:13:52 [ERROR] - pipewire: fatal error event from core
2024-02-14T14:13:52.638465+01:00 rpi xdg-desktop-por[2270]: Caught PipeWire error: connection error
2024-02-14T14:13:52.651400+01:00 rpi systemd[1750]: Stopped pipewire.service - PipeWire Multimedia Service.
2024-02-14T14:13:52.651914+01:00 rpi systemd[1750]: pipewire.service: Consumed 2.980s CPU time.
2024-02-14T14:13:52.685947+01:00 rpi systemd[1750]: Started pipewire.service - PipeWire Multimedia Service.
2024-02-14T14:13:52.692411+01:00 rpi systemd[1750]: Started wireplumber.service - Multimedia Service Session Manager.
2024-02-14T14:13:52.693201+01:00 rpi systemd[1750]: xdg-desktop-portal-wlr.service: Main process exited, code=exited, status=1/FAILURE
2024-02-14T14:13:52.693686+01:00 rpi systemd[1750]: xdg-desktop-portal-wlr.service: Failed with result 'exit-code'.
2024-02-14T14:13:52.880896+01:00 rpi systemd[1750]: xdg-desktop-portal-wlr.service: Scheduled restart job, restart counter is at 22.
2024-02-14T14:13:52.881560+01:00 rpi systemd[1750]: Stopped xdg-desktop-portal-wlr.service - Portal service (wlroots implementation).
2024-02-14T14:13:52.913039+01:00 rpi rtkit-daemon[1336]: Successfully made thread 5946 of process 5938 owned by '1000' high priority at nice level -11.
2024-02-14T14:13:52.914130+01:00 rpi rtkit-daemon[1336]: Supervising 1 threads of 1 processes of 1 users.
2024-02-14T14:13:52.914730+01:00 rpi systemd[1750]: Starting xdg-desktop-portal-wlr.service - Portal service (wlroots implementation)...
[... cut ...]
2024-02-14T14:13:56.734403+01:00 rpi wireplumber[5981]: <WpPortalPermissionStorePlugin:0x55948fa8d0> Failed to call Lookup: GDBus.Error:org.freedesktop.portal.Error.NotFound: No entry for camera
2024-02-14T14:13:56.742703+01:00 rpi wireplumber[5981]: <WpPortalPermissionStorePlugin:0x55948fa8d0> Failed to call Lookup: GDBus.Error:org.freedesktop.portal.Error.NotFound: No entry for camera
2024-02-14T14:14:02.708043+01:00 rpi bluetoothd[911]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down

popcornmix, 2024-02-14:
We're investigating something related to this (USB or bluetooth audio may be killed on HDMI hotplug event).
Does:
```
﻿pkill respawn
pkill wf-panel-pi
```
stop the unwanted behaviour? (note this is a test rather than a fix - it will kill the wf-panel).
wf-panel-pi does a `systemctl --user restart wireplumber.service` on a hotplug event, to get the list of hdmi audio devices updated, but that is obviously having an unwanted side effect. We're looking into a way to get wireplumber/pipewire to reenumerate the hdmi audio devices without affecting other devices.

apassu, 2024-02-15:
I tried your commands, the first one is ok but when I launched the second the RPI freezes and became unresponsive.

mghadam, 2024-02-16:
a similar issue is reported here
https://github.com/raspberrypi/bookworm-feedback/issues/205

pelwell, 2024-02-16:
Yes - as with https://github.com/raspberrypi/linux/issues/5950, this is not a kernel problem. Follow the other issue instead.

closed 2024-02-16 by pelwell

apassu, 2024-02-17:
> Yes - as with #5950, this is not a kernel problem. Follow the other issue instead.

Which one to follow?

pelwell, 2024-02-17:
https://github.com/raspberrypi/bookworm-feedback/issues/205