raspberrypi/linux #5952: Logitech Camera Not Detected on Bookworm Firefox

mghadam, 2024-02-13:
### Describe the bug

Firefox on Raspberry Pi 4B running Bookworm detects the microphone of my Logitech Brio; however, the camera is not detected. Interestingly, the Brio camera/microphone functions properly in Chromium on Bookworm. Additionally, Firefox ESR on Debian Bullseye successfully detects the camera/microphone.

### Steps to reproduce the behaviour

Initiate a Google Meet session in Firefox on Bookworm.
The browser prompts for permission to access the microphone of the Logitech Brio webcam, but it fails to detect any camera and does not request permission for camera access.

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry Pi reference 2023-12-05
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 70cd6f2a1e34d07f5cba7047aea5b92457372e05, stage5

version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start)

Linux home.domain.local 6.1.0-rpi8-rpi-v8 https://github.com/raspberrypi/linux/issues/1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux

bou6, 2024-02-14:
can you check the output of
rpicam-hello –list-cameras

6by9, 2024-02-14:
> can you check the output of rpicam-hello –list-cameras

Thanks for trying to assist, but that is not going to help as rpicam-apps are specifically targetting the Pi CSI cameras, and a Logitech camera will be UVC (USB Video Class).

`v4l2-ctl --list-devices` should list all your cameras. If v4l2-ctl sees your camera but Firefox doesn't, then it's a Firefox issue and not a kernel one.