raspberrypi/linux #5956: NVMe takes over if both SD Card and NVMe are bootable

addohm, 2024-02-13:
### Describe the bug

When both the NVMe drive and the SD card are bootable, regardless of the boot order setting, the Pi still boots from the NVMe drive.

### Steps to reproduce the behaviour

I've gotten my pi5 running extremely well on NVMe so I decided to try and image the drive.  Not a problem.  I then wrote the image to an SD card.  I altered the UUIDs so they weren't conflicting.  I then went into `raspi-config` and set boot order to SD card.  When I did that, it still booted from the NVMe drive.  I then tried `rpi-eeprom-update config` and set boot order to `0xf1` thinking I'd take the NVMe completely out of the boot order.  It still booted from the NVMe drive.  I then removed the NVMe drive from the pi, and it booted from the SD card.  

### Device (s)

Raspberry Pi 5

### System

OS and Version: Raspberry Pi reference 2023-12-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 2acf7afcba7d11500313a7b93bb55a2aae20b2d6, stage2

Version: 2024/01/05 15:57:40 
Copyright (c) 2012 Broadcom
version 30cc5f37 (release) (embedded)

Kernel: Linux pinas 6.1.0-rpi8-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux

peterharperuk, 2024-02-13:
Did you get the partition uuid right in cmdline.txt on the SD card?

addohm, 2024-02-13:
> Did you get the partition uuid right in cmdline.txt on the SD card?

Oof, didn't realize there was another place UUID is kept.  Can cmdline.txt use UUID or does it have to be PARTUUID?  I also just noticed that although I changed the UUID, the PARTUUID didn't change.  I assumed it was just derived from UUID all the time.

closed 2024-02-13 by addohm