raspberrypi/linux #5957: sc16is7xx driver IRQ polarity support

recon-Mike, 2024-02-13:
### Describe the bug

It appears like the driver does not support IRQ polarity configuration from the device tree anymore.  Maybe from [this commit](https://github.com/raspberrypi/linux/commit/37f3965d74d5b5c914f8f8a108ae1d94ab5ffae3) its support was removed?  I have a hardware implementation that passes the devices IRQ line through an inverter.  When using older versions of the driver it worked without issue.  With newer versions, it is broken and the device is stuck in an infinite IRQ loop since the IRQ is never cleared due to the fact the polarity is incorrect.

### Steps to reproduce the behaviour

add change `IRQ_TYPE_EDGE_FALLING` to `IRQ_TYPE_EDGE_RISING` or `IRQ_TYPE_LEVEL_HIGH` to a sc16is7xx device and see no change in behaviour.

### Device (s)

Other

### System

Not using rpi hardware.  Using kernel version 5.15.  Kernel version 5.4 supported inverted polarity.

pelwell, 2024-02-14:
It seems like a thoughtless patch, and I was with you until a read "Not using rpi hardware". What were you expecting to happen?

recon-Mike, 2024-02-14:
I think what I am hoping for is perhaps a few things:
1. A better understanding of this driver and is limitations
2. Helping with creating a patch to improve the hardware compatibility (I don't have much experience with Linux driver development)
3. Having something searchable by others to find out about this current limitation.  It really seems like the raspberrypi github has a majority of the search hits for my embedded Linux work.

pelwell, 2024-02-14:
That patch deliberately removed support for other polarities, for no obvious gain other than saving a few lines of source code and a few bytes of built kernel. Removing or reverting that patch will restore the previous behaviour.

closed 2024-02-15 by recon-Mike