raspberrypi/linux #5967: Debian bookworm failure of awt.robot.createScreenCapture which works on buster and bunsenlabs Boron

TPMoyer, 2024-02-18:
### Describe the bug

On RPi 4b 64 bit bookworm, legacy Processing 4 sketches using java.awt.robot.createScreenCapture() collect only black pixels, not the correct screen image representation.   These sketches continue to work correctly on RPi Buster images (both legacy and 2024/02/14 downloads).   These sketches also work correctly on recent chromebook installs of BunsenLabs Boron.

### Steps to reproduce the behaviour

Use Raspberry Pi Imager to make SD Card images of 64 bit Bookworm and 64 bit Buster for RPi 4b.   
For each of the SD cards, insert in a RPi 4b with appropriate keyboard, mouse, and display.  Get to initial RPi screens.  Install the proffered updates.  Open the browser.  Search on  Processing Downloads.   Go to processing download page but DO NOT click the big frendly blue linux download button.   Scroll down to the Raspberry Pi logo and click on same.   Select the 64 bit download.   Wait for download to finish.   Open file browser, go to /home/pi/Downloads       extract the .tgz       Move the resultant processing-4.3 folder to /home/pi          Open a terminal.    cd into /home/pi/processing-4.3        Execute      sudo sh install.sh             Click on the raspberry icon... programming... processing   and wait for sketch to open.      Download code from https://github.com/TPMoyer/RPiBookwormBug   and run the sketch by double clicking on the RPiBookwormBug.pde from the fileManager   or copy the following into the opened sketch

import java.awt.Dimension;
import java.awt.Toolkit;
import java.awt.image.BufferedImage;
import java.awt.Robot;
import java.awt.Rectangle;
import java.awt.AWTException;
import java.time.*;
import java.text.SimpleDateFormat;
import java.util.Date;
import javax.imageio.ImageIO;

Robot robot;
String fid;

void setup() {  
  size(640, 360);
  SimpleDateFormat mmddyyhhmmssSSS   = new SimpleDateFormat ("yyyy_MM_dd_HH_mm_ss_SSS");

  try{
    robot = new Robot(); /* java Robot class can be used to control mouse from within the program and inject keystrokes. */
  } catch (AWTException e) {
    e.printStackTrace();
  }
  
  Dimension screenSize = Toolkit.getDefaultToolkit().getScreenSize();
  System.out.println("screenSize="+screenSize.getWidth()+","+screenSize.getHeight());  
  System.getProperties().list(System.out);  
  
  BufferedImage bi = robot.createScreenCapture(new Rectangle((int)screenSize.getWidth(),(int)screenSize.getHeight()));
  String nowFormatted =mmddyyhhmmssSSS.format(new Date());
  fid=System.getProperty("user.home")+System.getProperty("file.separator")+nowFormatted+"_shouldBeNonBlank.png";
  File f = new File(fid);
  try {
    ImageIO.write(bi, "PNG", f);
  } catch (IOException e) {
    System.out.println("IOException "+e);
  }
  textSize(20);
}

void draw() {
  background(0);
  text("Fullscreen capture was perfromed...  png was saved as\n"+fid+"\nSystem properties listed in processing black console\nescape key will exit", 20, 120);
}

Run the sketch by clicking on the circle with the little triangle therein, which is located near the upper left corner of the sketch.
If running on Buster, the created .png will be a good screen capture.  If running on Bookworm, the .png will be all black.
GitHub repo contains full listing of System properties for a Buster and Bookworm run of the sketch.

### Device (s)

Raspberry Pi 4 Mod. B

### System

On Bookworm
cat /etc/rpi-issue
Raspberry Pi reference 2023-12-05
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 70cd6f2a1e34d07f5cba7047aea5b92457372e05, stage4
vcgencmd version
Oct 17 2023 15:39:16 
Copyright (c) 2012 Broadcom
version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start)
uname -a
Linux raspberrypi 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux

On Buster
cat /etc/rpi-issue
Raspberry Pi reference 2023-12-05
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, e484aa85818428c3af08de0ba1db5329dfc84143, stage4
vcgencmd version
Mar 17 2023 10:50:39 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)
 uname -a
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

6by9, 2024-02-19:
Bookworm is now using Wayland for the graphical environment. Does your Java class support Wayland backends, or only X11?
You can revert to X11 using raspi-config to test - Advanced Options/Wayland.

(Making your class support Wayland is not a kernel issue)

TPMoyer, 2024-02-19:
Wayland vs X11 is indeed the problem.      

Reversion to X11 via raspi-config did, however, not allow me to move to Bookworm.
WTF is now that WayVNC shows uniform gray, disallowing headless operation.

I'll use Buster for my installs, 
and check in a year or so
to see if they have fixed the java.awt stuff to allow compatibility with whatever Debian/Wayland is then RPi current.  

Thank you for your quick and insightful response.

closed 2024-02-19 by TPMoyer

6by9, 2024-02-19:
I'd recommend Bullseye over Buster. LTS ends on Buster on 30th June 2024, ie 5 months away.

WayVNC isn't going to anything useful against X11 as it is the backend for Wayland compositors using wlroots.

XECDesign, 2024-02-19:
> I'd recommend Bullseye over Buster. LTS ends on Buster on 30th June 2024, ie 5 months away.

That's Debian's LTS. We have stopped updating our Buster packages long ago so running it is a massive security risk.

TPMoyer, 2024-02-19:
Bullseye it is.
Nice catch.