raspberrypi/linux #5975: RPI Zero 2W incorrect memory size

gugolple, 2024-02-20:
**Describe the bug**
After doing an rpi-update on my raspberry pi zero 2W, the available ram reduced from ~484MB to 400MB. I do understand that the memory split will take an amount of ram (check logs section). I do not understand the reason of  the change, but having less resources available in such a constrained board is not beneficial.

**To reproduce**
Executing the command **sudo rpi-update**.

**Expected behaviour**
Available ram for the system to be at around ~484MB.

**Actual behaviour**
Available ram for the system is found to be at around ~400MB.

**System**
```
$ cat /etc/rpi-issue
Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage4
```
```
$ vcgencmd version
Dec  8 2023 12:42:28 
Copyright (c) 2012 Broadcom
version e02d33b3122450accf9dea471a177d3b5623f5ad (clean) (release) (start_cd)
```
```
$ uname -a
Linux raspberrypi 6.6.16-v8+ raspberrypi/firmware#1728 SMP PREEMPT Tue Feb  6 17:21:42 GMT 2024 aarch64 GNU/Linux
```

**Logs**
Reduced to only provide relevant information.
```
$ dmesg | egrep 'Memory|CMA|cma' --color -C3
[    0.000000] Machine model: Raspberry Pi Zero 2 W Rev 1.0
[    0.000000] Memory: 125944K/507904K available (13376K kernel code, 2208K rwdata, 4308K rodata, 4928K init, 1083K bss, 119816K reserved, 262144K cma-reserved)
```
```
$ cat /proc/meminfo
MemTotal:         409760 kB
```
```
$ vcgencmd get_mem gpu
gpu=16M
```

**Additional context**
I do not know if they are relevant, but I have attached the boot configurations just in case I have enabled a device that also is reserving ram at boot.
```
$ cat /boot/firmware/cmdline.txt 
console=serial0,115200 console=tty1 root=PARTUUID=e668d3db-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=BE
```
```
$ cat /boot/firmware/config.txt | grep -v '#' | grep -v -e '^$'
dtparam=spi=on
dtparam=audio=on
camera_auto_detect=0
display_auto_detect=1
auto_initramfs=1
dtoverlay=vc4-kms-v3d,spi1-3cs
max_framebuffers=1
disable_fw_kms_setup=1
arm_64bit=1
disable_overscan=1
arm_boost=1
[cm4]
otg_mode=1
[all]
start_x=0
gpu_mem=16
enable_uart=1
```
Thank you.

popcornmix, 2024-02-20:
start*.elf and fixup*.dat should always be of the same version, so not booting is reasonable.

I've transferred this issue to the linux repo as that appears to be the cause of the increased memory.

As @pelwell says, it is worth testing how much of the increased memory is due to 64-bit kernel (which now also supports pi5, which the 32-bit kernel does not).

Is your OS 32-bit or 64-bit? If it's 32-bit then booting without `arm_64bit=1` would be informative.
If it's 64-bit then it's trickier to test (will need a 32-bit install).

gugolple, 2024-02-20:
> As @pelwell says, it is worth testing how much of the increased memory is due to 64-bit kernel (which now also supports pi5, which the 32-bit kernel does not).
>
> Is your OS 32-bit or 64-bit? If it's [...]

Currently I'm running "Raspberry Pi OS (64-bit) : Raspberry Pi OS Lite".

What tests and versions would be intriguing to execute? I am open for suggestions.

pelwell, 2024-02-21:
Although I've never quite got 460M for total mem, my results are close, seeing a ~60M drop when switching to the 6.6 kernel - whether by rpi-update or building my own.

Looking at the start of the /proc/iomem output with some annotations is instructive.

Before:
```
MemTotal:         440568 kB
00000000-1effffff : System RAM [-0.0M -> 496.0M]
  00000000-00000fff : reserved [-0.0M -> 496.0M]
  00210000-0121ffff : Kernel code [-16.1M -> 479.9M]
  01220000-0165ffff : reserved [-4.2M -> 475.7M]
  01660000-0199ffff : Kernel data [-3.2M -> 472.4M]
  0b200000-0ddfffff : reserved [-44.0M -> 428.4M]
  0de32000-0def2fff : reserved [-0.8M -> 427.7M]
  0def3000-0df66fff : reserved [-0.5M -> 427.2M]
  0df69000-0df69fff : reserved [-0.0M -> 427.2M]
  0df6a000-0df6afff : reserved [-0.0M -> 427.2M]
  0df6b000-1e08efff : reserved [-257.1M -> 170.1M]
  1e08f000-1e096fff : reserved [-0.0M -> 170.0M]
  1e097000-1effffff : reserved [-15.4M -> 154.6M]
```
After:
```
MemTotal:         372896 kB
00000000-1effffff : System RAM [-0.0M -> 496.0M]
  00000000-00000fff : reserved [-0.0M -> 496.0M]
  00210000-0136ffff : Kernel code [-17.4M -> 478.6M]
  01370000-0183ffff : reserved [-4.8M -> 473.8M]
  01840000-01b7ffff : Kernel data [-3.2M -> 470.6M]
  07140000-0ddfffff : reserved [-108.8M -> 361.8M]
  0de2c000-0deecfff : reserved [-0.8M -> 361.1M]
  0deed000-0df64fff : reserved [-0.5M -> 360.6M]
  0df67000-0df67fff : reserved [-0.0M -> 360.6M]
  0df68000-0df68fff : reserved [-0.0M -> 360.6M]
  0df69000-0df79fff : reserved [-0.1M -> 360.5M]
  0df7a000-1e08dfff : reserved [-257.1M -> 103.4M]
  1e08e000-1e08efff : reserved [-0.0M -> 103.4M]
  1e08f000-1e096fff : reserved [-0.0M -> 103.4M]
  1e097000-1effffff : reserved [-15.4M -> 88.0M]
```
Here you can clearly see that the third `reserved` section has gone from 44MB to 108MB, accounting for the ~60MB difference. 
(Ah - I had reserved 5MB for kernel logging; without that I get similar numbers to you.)

The obvious next question is what that area is reserved for. The answer is "software IO TLB" - DMA bounce buffers:
```
[    0.000000] software IO TLB: mapped [mem 0x0000000009600000-0x000000000d600000] (64MB)
```
This memory is used as a staging post for DMA transfers when the target memory is out of range of the DMA controller. There is no such allocation in the 6.1 kernel, which is puzzling. I'll have to dig deeper.

By the way, the `spi1-3cs` in `dtoverlay=vc4-kms-v3d,spi1-3cs` is wrong. It's the name of an overlay, not a parameter, so must be at the start of its own `dtoverlay` line if want it to have any effect.

pelwell, 2024-02-21:
The extra 64MB usage is caused by this commit: https://github.com/raspberrypi/linux/commit/370645f41e6e2fdd2fb6f6982530b04612c9793c

Prior to this, the initialisation of the swiotlb buffers was smart - if all of RAM is addressable by the DMA controller(s) then no bounce buffers are required. This is certainly true on a Z2W with only 512MB, so you save the 64MB carve-out. The above commit, which appeared in Linux 6.5, says that bounce buffers may be required if DMA transfers are not cache-line-aligned, so it is better to have the bounce buffers unconditionally allocated. The DMA_BOUNCE_UNALIGNED_KMALLOC option is turned on by the arm64 Kconfig file. I don't see any way to opt-out without modifying the kernel source - as a config setting without a description it isn't one you can alter from a defconfig file.

Needless to say, the 32-bit (ARCH=arm) Kconfig files don't set DMA_BOUNCE_UNALIGNED_KMALLOC - yet another reason you would be better of with a 32-bit build.

pelwell, 2024-02-21:
Given that 6.1 was quite happy without this option being defined, we opted to not enable it on 6.6 - see https://github.com/raspberrypi/linux/commit/3e0065d1cb5bd58e35e0c7722ffe6303dfe0364f. This change will be in future releases, but if you wait about 40 minutes (until around 15:10GMT) you should be able to install an early build of this beta kernel by running `sudo rpi-update rpi-6.6.y`.

For the record, this is the same RAM usage information taken from 6.6 with the patch applied (and without my oversized kernel log buffer):
```
MemTotal:         476064 kB
00000000-1effffff : System RAM [-0.0M -> 496.0M]
  00000000-00000fff : reserved [-0.0M -> 496.0M]
  00210000-0136ffff : Kernel code [-17.4M -> 478.6M]
  01370000-0183ffff : reserved [-4.8M -> 473.8M]
  01840000-01b7ffff : Kernel data [-3.2M -> 470.6M]
  0d600000-0ddfffff : reserved [-8.0M -> 462.6M]
  0de2c000-0deecfff : reserved [-0.8M -> 461.8M]
  0deed000-0df64fff : reserved [-0.5M -> 461.3M]
  0df67000-0df67fff : reserved [-0.0M -> 461.3M]
  0df68000-0df68fff : reserved [-0.0M -> 461.3M]
  0df69000-0df79fff : reserved [-0.1M -> 461.3M]
  0df7a000-1e08dfff : reserved [-257.1M -> 204.2M]
  1e08e000-1e08efff : reserved [-0.0M -> 204.2M]
  1e08f000-1e096fff : reserved [-0.0M -> 204.1M]
  1e097000-1effffff : reserved [-15.4M -> 188.7M]
```

popcornmix, 2024-02-21:
> This is an interesting and related read

The patch series described there is in our 6.6 kernel (it wasn't in 6.5 and prior ones).
So it isn't solving *this* issue.

gugolple, 2024-02-21:
> Given that 6.1 was quite happy without this option being defined, we opted to not enable it on 6.6 - see [3e0065d](https://github.com/raspberrypi/linux/commit/3e0065d1cb5bd58e35e0c7722ffe6303dfe0364f). This change will be in future releases, but if you wait about 40 minutes (until around 15:10GMT) you should be able to install an early build of this beta kernel by running `sudo rpi-update rpi-6.6.y`.

Seeing that this is not the only project with this issue, maybe the functionality could be kept with a smaller fix just be setting the kernel boot parameter of "swiotlb=0" as they did in the [Yocto project](https://patchwork.yoctoproject.org/project/oe-core/patch/20231006112119.1081596-1-ross.burton@arm.com/) for qemu images.

pelwell, 2024-02-21:
That's not a bad solution, but this works for us for now.

gugolple, 2024-02-21:
Can confirm the fix is working.
```
$ free -h
               total        used        free      shared  buff/cache   available
Mem:           465Mi       161Mi       133Mi       1.4Mi       222Mi       303Mi
```
```
$ uname -a
Linux raspberrypi 6.6.17-v8+ #1 SMP PREEMPT Wed Feb 21 16:23:00 UTC 2024 aarch64 GNU/Linux
```
```
$ vcgencmd version
Oct 17 2023 15:42:53 
Copyright (c) 2012 Broadcom
version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start_cd)
```
Thank you everyone for the support.

pelwell, 2024-02-21:
Thank you - without your observation all Z2W owners would be missing 12.5% of their RAM for no reason.

closed 2024-02-21 by pelwell

ctmarinas, 2024-04-04:
With mainline commit `65033574ade9 ("arm64: swiotlb: Reduce the default size if no ZONE_DMA bouncing needed")` in 6.7, the bounce buffer is resized based on the available RAM when no 32-bit bouncing is needed (I think it will be 1MB on RPi Zero). I consider this a pretty good trade-off given that you can then get the kmalloc-{8,16,32,96} slab caches, saving memory overall and slightly better performance for small allocations.

Only tested on RPi5 but immediately after boot I get from `/proc/slabinfo` (with `DMA_BOUNCE_UNALIGNED_KMALLOC` enabled):

```
kmalloc-96          3753   5610     96  170    1 : tunables    0    0    0 : slabdata     33     33      0
kmalloc-64          4271   4352     64  256    1 : tunables    0    0    0 : slabdata     17     17      0
kmalloc-32          5603   5632     32  512    1 : tunables    0    0    0 : slabdata     11     11      0
kmalloc-16          8192   8192     16 1024    1 : tunables    0    0    0 : slabdata      8      8      0
kmalloc-8          10240  10240      8 2048    1 : tunables    0    0    0 : slabdata      5      5      0
```

If the above were rounded up to kmalloc-64 and kmalloc-128, my calculations show about 1.2MB wasted. I have not included the `kmalloc-rcl-*` and `kmalloc-cg-*` caches, the wastage would be even higher.

pelwell, 2024-04-05:
Running `free` (after giving the device time to go idle) using the two kernels shows an extra MB used after removing the reversion and applying your commit (198332 -> 197324), so what happened to "saving memory overall"?

ctmarinas, 2024-04-05:
I can't tell without more information from `/proc/slabinfo`. It really depends on what the system is doing, how many active slab objects you have around. As I said, I only looked at the statistics of running Debian on a RPi5 while limiting the memory via mem=512M. If it's an idle system, wasting 1MB on the bounce buffer doesn't affect anything. If it's a busy system with lots of small kernel allocations, you recover some memory by having more slab objects in a page. You also get better cache utilisation for those small allocations.

popcornmix, 2024-04-10:
I believe the upstream fix costs 1MB more than the downstream fix here. 

Yes, it's feasible to switch, to avoid a divergence with upstream, but there is a cost.
Is there any other benefit to the upstream version?

popcornmix, 2024-04-10:
If you can provide some evidence that the upstream choice is beneficial is some circumstance, then that will certainly influence any decisions made.

ctmarinas, 2024-04-10:
Up to you really if you want to follow mainline or not (I'm going to revert your patch in my RPi kernel builds ;)). I don't have an RPi Zero to test. In my earlier [comment](https://github.com/raspberrypi/linux/issues/5975#issuecomment-2037563901) on RPi5, I showed that there was a clear gain in reverting your patch. It's also better cache utilisation (and hence performance) that you gain by fitting more slab objects per page. It's not just about memory usage.

BTW, another option would be to play around with the reserved swiotlb buffer, try to set a smaller one either by patching the kernel (as per the upstream fix) or via a command line argument. I admit that my choice of 1MB per 1GB of RAM was pretty arbitrary. Personally I'd leave the choice to the user to disable the swiotlb or go for a smaller size (via the boot config) than patching the kernel to prevent small kmalloc buffers altogether.

ctmarinas, 2024-04-10:
Also, I get your point about not wasting RAM on a RPi Zero but why penalise, with your fix, the other RPi variants with sufficient RAM? You need the swiotlb buffer anyway for 32-bit DMA and you are unnecessarily preventing small kmalloc() allocations by changing the Kconfig directly. If you do want this for RPi Zero, I'd suggest a better patch that takes the amount of RAM into account.

pelwell, 2024-04-17:
Are you a Zero 2 W user?

seamusdemora, 2024-04-17:
I'm a Zero 2 W user... and also a RPi 5 user. This is a very interesting thread; it seems to illustrate the tradeoffs in accommodating the variety of hardware platforms that make up the RPi lineup. Reading through the earlier part of this thread earlier this year led me to consider "downgrading" my Zero 2 W to the 32-bit version. I decided to hold out for a "fix" to the 64-bit OS for the simple reason that all of the work outlined here was based on testing the 64-bit OS; i.e. maybe there are flaws in the 32-bit OS that simply have not been observed - yet. 

I've also noticed that there's "soft block" against the 64-bit OS in the "Raspberry Pi Imager (ver 1.8.5)" for the Zero 2W, that is **not reflected** in the ["OS images" page](https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-32-bit). This creates uncertainty in selection... I'm not criticizing - only wondering. What I'd really like to know is which of the OS this group recommends for Zero 2W.

pelwell, 2024-04-17:
Already answered:

> I note that you've opted into the 64-bit kernel (and perhaps even the 64-bit OS - it's hard to tell). This is not recommended on a Zero 2 W because it will eat into your limited RAM.

seamusdemora, 2024-04-17:
Not completely answered... what about bookworm vs bullseye?

pelwell, 2024-04-18:
> which of the OS this group recommends for Zero 2W

That seems to depend on who you ask. The rpi-imager devs say Bullseye (https://github.com/raspberrypi/rpi-imager/issues/690).
The active devs here say turn off filtering and install Bookworm. You will get an image that doesn't run Wayland, but then neither does the Bullseye image, and Bookworm has an actively maintained kernel.

If I had to guess, the motivation for the Bullseye recommendation is probably to prevent people "upgrading" to something which isn't visibly better, and then wondering why they missed out. The filtering will probably change when Wayland works on more devices.

seamusdemora, 2024-04-19:
@pelwell 

Thanks for the confirmation; I had wondered if this didn't "boil down" to differences of opinion. I feel... *enlightened*  :)

pelwell, 2024-04-19:
> I had wondered if this didn't "boil down" to differences of opinion

Not so much - it's more a difference of audience.

seamusdemora, 2024-04-22:
@pelwell 
> > I note that you've opted into the 64-bit kernel (and perhaps even the 64-bit OS - it's hard to tell). This is not recommended on a Zero 2 W because it will eat into your limited RAM.

Would this (limited RAM) account for the x-sluggish response I get (often) in the terminal when connected via SSH?

pelwell, 2024-04-22:
There is insufficient information to answer that question:
1. Define sluggish.
2. How much free memory is reported?
3. How are you SSHing in?
4. Have you tried overcoming any kind of network power-saving feature with a background ping?

seamusdemora, 2024-04-23:
1. Sluggish: This refers to the "time delay" I see in my SSH terminal window; i.e. the delay between when I press a key on the keyboard until I see that keyboard entry "echoed" to my terminal window. I call "sluggish" a delay of 1-5 seconds. The longest delay is typically for the first key, and then it reverts to "normal" (i.e. no perceptible delay) after a couple of key presses. I also see the delay (for example) when I enter `control-r` to search for a previous command.

2. Free memory: per `free` command: 
```bash
pi@rpi2w:~ $ free -m
               total        used        free      shared  buff/cache   available
Mem:             425         109         185           0         180         316
Swap:             99           0          99
```  

3. How are you SSHing in: 
```bash
# from the `zsh` prompt in iTerm2 on macos, I enter the following command:
% ssh pi@rpi2w
# occasionally (when running the Mac on battery) I use a different command: 
% caffeinate  -s ssh -o ServerAliveInterval=60 -o ServerAliveCountMax=10 pi@rpi2w
# However, I have not noticed a difference in the "sluggishness" between these commands
```

4. power-saving feature?
AFAIK, I do not have any power saving features, but I'm not quite sure how to check/verify this on my 'bookworm' Lite system.

pelwell, 2024-04-23:
Saving the planet, one Zero at a time.

seamusdemora, 2024-04-24:
@pelwell 
I briefly assumed this 'power save = on' default was set in NetworkManager/`nmcli`, but if it is, I've been unable to find it... the default "connection profile" makes no mention of it. Is it set somewhere in "the kernel"? Do you know where?

pelwell, 2024-04-25:
Not off the top of my head, and this is not a general support forum.

seamusdemora, 2024-04-25:
OK - my apologies, just trying to sort out some issues with the `power save` "feature", and it seems it *must* be in the `brcmfmac` module. I'll try the Debian forum.