raspberrypi/linux #5979: overlays: config.txt not creating /dev/i2c* on latest 64-bit version ( 6.1.0-rpi7-rpi-v8 )

RaulTrombin, 2024-02-22:
### Describe the bug

I was testing a shield with sensors on I2C ports, 
unfortunately it did not create the i2c-* devices in the /dev/ folder.

Something changed on the config.txt sintaxe or devices's folder?

It work on previous versions.

### Steps to reproduce the behaviour

on config.txt

```
[all]
enable_uart=1
dtoverlay=uart1
dtoverlay=uart3
dtoverlay=uart4
dtoverlay=uart5
dtparam=i2c_vc=on
dtoverlay=i2c1
dtparam=i2c_arm_baudrate=1000000
dtoverlay=i2c4,pins_6_7,baudrate=1000000
dtoverlay=i2c6,pins_22_23,baudrate=400000
dtparam=spi=on
dtoverlay=spi0-led
dtoverlay=spi1-3cs
gpio=11,24,25=op,pu,dh
gpio=37=op,pd,dl
dtoverlay=dwc2

```

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry Pi reference 2023-12-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 2acf7afcba7d11500313a7b93bb55a2aae20b2d6, stage2

pi@raspberrypi:/dev $ vcgencmd version
Oct 17 2023 15:39:16 
Copyright (c) 2012 Broadcom
version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start)

pi@raspberrypi:/dev $ uname -a
Linux raspberrypi 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux

6by9, 2024-02-22:
Have you loaded i2c_dev?
The kernel will quite happily enable I2C interfaces for use within the kernel, but they are only exposed to userspace via i2c_dev 
(`sudo modprobe i2c_dev`, add it to /etc/modules, or enabling I2C via raspi-config will add it to /etc/modules for you).

RaulTrombin, 2024-02-22:
# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.
# Parameters can be specified after the module name.

bcm2835-v4l2
i2c-bcm2835
i2c-dev

> Have you loaded i2c_dev? The kernel will quite happily enable I2C interfaces for use within the kernel, but they are only exposed to userspace via i2c_dev (`sudo modprobe i2c_dev`, add it to /etc/modules, or enabling I2C via raspi-config will add it to /etc/modules for you).

in fact using the raspi-config, it enables the i2c-1.
now i have i2c-1, i2c-20 and i2c,21. but still missing the i2c-4 and i2c-6.

```
pi@raspberrypi:/dev $ uname -a && ls /dev/i2c-*
Linux raspberrypi 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux
/dev/i2c-1  /dev/i2c-20  /dev/i2c-21

```

vs old version:
```
pi@blueos:/dev $ uname -a && ls /dev/i2c-*
Linux blueos 5.10.33-v7l+ #1415 SMP Fri Apr 30 15:50:57 BST 2021 armv7l GNU/Linux
/dev/i2c-0  /dev/i2c-1  /dev/i2c-10  /dev/i2c-11  /dev/i2c-4  /dev/i2c-6

```

RaulTrombin, 2024-02-22:
@6by9 thks for all infos regards modules,

Spi0-led is a custom neopixel that my script installs.

I was using the webpage lite images [here](https://downloads.raspberrypi.com/raspios_lite_arm64/images/?_gl=1*honw8c*_ga*MTMxMzE4MTc3Ni4xNzA3NzY4ODgy*_ga_22FD70LWDS*MTcwODYyNzk0MS42LjEuMTcwODYyNzk1MS4wLjAuMA..)
That I think is the last available officialy for users,

Even when I do a rpi-update, it still not load and show the i2c devices with the new kernel:
```
pi@raspberrypi:/dev $ uname -a
Linux raspberrypi 6.6.17-v8+ #1735 SMP PREEMPT Wed Feb 21 14:39:52 GMT 2024 aarch64 GNU/Linux

```

I'm going to check if the armv7 have the same behavior.

popcornmix, 2024-02-22:
> I was using the webpage lite images here
That I think is the last available officialy for users,

If you run `sudo apt update / sudo apt full-upgrade` you should get the 6.1.73 kernel.

6by9, 2024-02-22:
I would remove `dtoverlay=spi0-led` to eliminate it from being the problem.

If it is the problem, then you'll have to work out where the GPIO conflict is.

RaulTrombin, 2024-02-22:
but if even without the spi0-led it keep not working yet.
I tryed the current latest 32-bit also, same issue there (after change to boot into correct armv7 kernel which I open an issue).

@6by9 I'm going to use a clean 64 lite image again, and check the "pinctrl get"

seamusdemora, 2024-02-22:
Regard this input as *"unsubstantiated rumor"*: 

I've noticed on a couple of occasions on a Pi4 **bullseye** system that enabling i2c in config.txt followed by a reboot was insufficient to get i2c working. The only way to overcome this was to enable it in `raspi-config`.

RaulTrombin, 2024-02-23:
@6by9 I did it again, on latest official 64bits lite,
with overlays and etc/modules:

```
i2c-bcm2835
i2c-dev
```
It works fine, just missed the two i2c modules.

Can't explain what happened before.

pelwell, 2024-02-23:
Of those, only `i2c-dev` needs to be in /etc/modules - `i2c-bcm2835` (or `i2c-designware-platform`/`i2c-brcmstb` on a Pi 5) will be loaded automatically when needed.

closed 2024-02-23 by pelwell