raspberrypi/linux #5981: 32-bit lite version is loading aarch64 kernel architecture

RaulTrombin, 2024-02-22:
### Describe the bug

The official latest 32-bit lite image is loading aarch64,
It's fixed if place `arm_64bit=0` inside /boot/config.txt

### Steps to reproduce the behaviour

fresh install latest release from website:
then 
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux

### Device (s)

Raspberry Pi 4 Mod. B

### System

pi@raspberrypi:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2023-12-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, dc62cc4644c4810772cceb76502e562eb01b3870, stage2

pi@raspberrypi:~ $ vcgencmd version
Oct 17 2023 15:39:16 
Copyright (c) 2012 Broadcom
version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start)

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 6.1.0-rpi7-rpi-v7l #1 SMP Raspbian 1:6.1.63-1+rpt1 (2023-11-24) armv7l GNU/Linux

pelwell, 2024-02-22:
This behaviour is by design. The Pi 4 performs better with a 64-bit kernel, so it is chosen by default.

RaulTrombin, 2024-02-22:
But it could break any other images that uses the rpi 32-bit as base image.

Could the 32-bit versions just come with `arm_64bit=0`,
like the 64-bit versions already comes with `arm_64bit=1`?

pelwell, 2024-02-22:
Other distros are free to add `arm64bit=0` if they prefer.

qrp73, 2024-03-02:
I tried to set arm64bit=0 on aarch64 version, but it seems that it leads to system hung on the boot. With 32 bit version arm64bit flag works ok with both values. Why?

pelwell, 2024-03-03:
A 64-bit kernel can host a 32-bit OS, but a 32-bit kernel can't host a 64-bit OS - the CPU architecture just doesn't allow it. For that reason, our 64-bit images don't include the 32-bit kernels.