raspberrypi/linux #5986: Seems like waveshare 2.8 inch Panel has wrong touchinput scale in bookworm

TheMarrt, 2024-02-25:
### Describe the bug

This panel:
https://www.waveshare.com/wiki/2.8inch_DSI_LCD#FAQ

can be activated with this dtoverlay:
dtoverlay=vc4-kms-dsi-waveshare-panel,2_8_inch,rotation=90

But the Scale seems to be flipped. Horizontal(480) scale is overexaggerated like it scales to 640px, and vertical(640) to 480).
I suspect this is because the Display is 480x640 rather than 640x480

I suspect this must be fixed, but i am new to github issue reporting:
https://github.com/raspberrypi/linux/blob/afd5f659b0453e4c710ce5cf74c577563ff16239/arch/arm/boot/dts/overlays/vc4-kms-dsi-waveshare-panel-overlay.dts#L80
"touchscreen-size-x:0=640" --> 480
"touchscreen-size-y:0=480" --> 640

### Steps to reproduce the behaviour

try https://www.waveshare.com/wiki/2.8inch_DSI_LCD on raspberry pi 4 b with bookworm installed

### Device (s)

Raspberry Pi 4 Mod. B

### System

cat /etc/rpi-issue
Raspberry Pi reference 2023-12-05
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 70cd6f2a1e34d07f5cba7047aea5b92457372e05, stage4

vcgencmd version
Oct 17 2023 15:39:16
Copyright (c) 2012 Broadcom
version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start)

uname -a
Linux pi8 6.1.0-rpi8-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux

6by9, 2024-02-25:
The wiki you linked to doesn't reference the rotation override at all.

Why do you need to rotate the video from device tree? If using any window manager (ie X11 or Wayfire) then both have easy mechanisms for rotating the displays as part of the composition (`xrandr` and `wlr-randr`), and similar options for mapping touch to match the display (`xinput --map-to-output` and adding an `input-device:` section to wayfire.ini).

I'm not overly bothered about fixing the touchscreen for the basic framebuffer device with touch. My preference would be dropping the rotation override.

TheMarrt, 2024-02-25:
regardless of rotation, the touchinput mismatch is always there as far as i have tried (used almost all combinations of swapxy and invert). This seems to be a scaling problem, when touching and dragging along the larger 640px side from left to right (e.g. to pull up a selection rectangle on the desktop), the rectangle only is 480px wide, vice-versa, when dragging on the shorter dimension, the rectangle exceeds the screen dimensions.

the directions are correct, its just the scaling

6by9, 2024-02-26:
The 2.8" panel is one of the 3 Waveshare panels that I happen to have.
It looks like you're right - width and height on the touch are swapped.

PR #59990 created to fix that.

Yes in /home/<user>/.config/wayfire.ini,
```
[input-device:10-0014 Goodix Capacitive TouchScreen]
output = DSI-1
```
means that Wayfire will automatically transform the input for the touch to match any rotations applied to the image by Wayfire. 

It looks like Wayfire totally ignores the "panel orientation" property from DRM. Just testing with `,rotation=90`, and whilst the connector is advertising (via `kmsprint -m`)
```
   panel orientation (47) = 3 (Right Side Up) [Normal=0|Upside Down=1|Left Side Up=2|Right Side Up=3] (immutable)
```
Wayfire has just displayed the desktop in the normal orientation. The console framebuffer also ignores it, but that I sort of expected.

TheMarrt, 2024-02-26:
I admit i only understand fractions of the code parts in your replies because i bought my first pi a month ago. But thank you for the fast reply and the change you initiated :)

popcornmix, 2024-02-26:
You can test this kernel with:
`sudo rpi-update pulls/5990`

TheMarrt, 2024-02-26:
> Collaborator
Nice, touch is now working perfectly.

**Owners of this Display, beware:**
I noticed another thing but I am not sure if this is related to a dtoverlay in any way: When the Screen turns off by screensaver, it will turn off the backlight but it seems to still render 6 colored Horizontal bars on the TFT. This creates a faintly visible burn-in (left the pi in that state for a few days :/). I am using the screenblanking in RaspberryPiConfiguration>Display>ScreenBlanking with a "dpms_timeout = 420" in the wayfire.ini

Is this Bookworm related or due to the wayfire PCB when the dsi signal turns off, and therefore unfixable? If yes, i will just switch to always on display with backlight reduction and an active screensaver that switches pixels.