raspberrypi/linux #5987: Backport patches to pwm-ir-tx that increase precision

mhyllander, 2024-02-25:
### Describe the bug

I have been trying to use both the gpio-ir-tx and pwm-ir-tx overlays to send IR signals. gpio-ir-tx was a complete failure, while pwm-ir-tx worked some of the time. I eventually found that the low precision of pwm-ir-tx was a known issue, and that patches have just been reviewed and merged into the 6.8 kernel. See https://lore.kernel.org/linux-pwm/cover.1703003288.git.sean@mess.org/.

These patches are available in the rpi-6.8.y branch:

1. c748a6d77c06 - pwm: Rename pwm_apply_state() to pwm_apply_might_sleep() (2023-12-20) <Sean Young>
1. dc518b378dce - pwm: Replace ENOTSUPP with EOPNOTSUPP (2023-12-20) <Sean Young>
1. 752193da3f8b - pwm: renesas: Remove unused include (2023-12-20) <Sean Young>
1. 7170d3beafc2 - pwm: Make it possible to apply PWM changes in atomic context (2023-12-20) <Sean Young>
1. fcc760729359 - pwm: bcm2835: Allow PWM driver to be used in atomic context (2023-12-20) <Sean Young>
1. 363d0e56285e - media: pwm-ir-tx: Trigger edges from hrtimer interrupt context (2023-12-20) <Sean Young>
1. 346c84e281a9 - media: pwm-ir-tx: Depend on CONFIG_HIGH_RES_TIMERS (2024-02-01) <Sean Young>

I have successfully backported them to the rpi-6.6.y branch, and with them the pwm-ir-tx module works _much_ better. I would like to request that they be backported to rpi-6.6.y for the benefit of all who use that module. 

### Steps to reproduce the behaviour

The IR messages that I send are long, consisting of 439 pulses and spaces. The chances that one of these pulses or spaces are not within the error margin is big, but with the backported patches there are no problems anymore.

### Device (s)

Raspberry Pi Zero 2 W

### System

any 6.1 or 6.6 kernel

pelwell, 2024-03-08:
Sorry it took a while to get round to this. When cherry-picking those commits I hit an awkward merge conflict that goes away when https://github.com/raspberrypi/linux/commit/2ce7b7f6704c9a8040fb12eb2b682986d9129e68 is applied first, so that has also been included.

There's now a PR for the backports: https://github.com/raspberrypi/linux/pull/6018

pelwell, 2024-03-10:
That PR is merged now. Please close this issue if that is satisfactory.

closed 2024-03-12 by mhyllander