raspberrypi/linux #5989: TC358762 based Pi 7" DSI panel has a 1 pixel horizontal offset

6by9, 2024-02-26:
### Describe the bug

Shifting from https://forums.raspberrypi.com/viewtopic.php?t=361866 so this doesn't get forgotten.

The TC358762 autodetection of the timing is sometimes wrong, so it needs to be explicitly passed.

Initially proposed fix is #5840, but that needs more testing.

### Steps to reproduce the behaviour

Use kmstest to display the default test screen, and note that you don't see the edge of the boxes in all corners

### Device (s)

Raspberry Pi 4 Mod. B

### System

All 6.1 and 6.6 kernels (probably older too).

Silent-Hunter, 2024-03-07:
I should mention this also affects Raspberry Pi 5.

6by9, 2024-06-23:
Fixed by #6094

closed 2024-06-23 by 6by9