raspberrypi/linux #5992: Composite "TV Mode" via Raspberry Pi5 ignores EVERYTHING I set (NTSC-J,PAL-M,etc)

03stevensmi, 2024-02-27:
### Describe the bug

I'm on the latest RaspbianOS with the latest 6.6 FIRMWIRE kernel!
Whenever I set config.txt to `video=Composite-1:720x480i,tv_mode=NTSC-J` 
or
`video=Composite-1:720x576i,tv_mode=PAL-N`
I GET IGNORED COMPLETELY!

Did rpi-update,
On the latest raspianOS,

This is my cmdline.txt:
`console=serial0,115200 console=tty1 root=PARTUUID=70957c7b-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-console cfg80211.ieee80211_regdom=GB video=Composite-1:720x576i,tv_mode=PAL-N`

This is my config.txt:
`
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

hdmi_ignore_cec_init=1

# Enable audio (loads snd_bcm2835)
dtparam=audio=on
audio_pwm_mode=1

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

hdmi_ignore_hotplug=1

enable_tvout=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d-pi5,composite
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1
`
[... 605 lines cut ...]
[   15.765647] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0382
[   15.765932] Bluetooth: hci0: BCM: Using default device address (43:45:c0:00:1f:ac)
[   16.922596] Adding 102368k swap on /var/swap.  Priority:-2 extents:1 across:102368k SSFS
[   17.428582] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   17.428590] Bluetooth: BNEP filters: protocol multicast
[   17.428597] Bluetooth: BNEP socket layer initialized
[   17.435858] Bluetooth: MGMT ver 1.22
[   17.476017] NET: Registered PF_ALG protocol family
[   24.228569] macb 1f00100000.ethernet eth0: PHY [1f00100000.ethernet-ffffffff:01] driver [Broadcom BCM54213PE] (irq=POLL)
[   24.228578] macb 1f00100000.ethernet eth0: configuring for phy/rgmii-id link mode
[   24.230907] pps pps0: new PPS source ptp0
[   24.230944] macb 1f00100000.ethernet: gem-ptp-timer ptp clock registered.
[   24.239963] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   29.637629] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   37.027426] Bluetooth: RFCOMM TTY layer initialized
[   37.027447] Bluetooth: RFCOMM socket layer initialized
[   37.027457] Bluetooth: RFCOMM ver 1.11
[   42.577564] exFAT-fs (sda1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[   42.577573] exFAT-fs (sda1): Medium has reported failures. Some data may be lost.
[  630.180089] drm-rp1-vec 1f00144000.vec: rp1vec_pipe_disable
pi@raspberrypi:~ $

### Additional context

Please help!

6by9, 2024-02-28:
Duplicate of forum thread https://forums.raspberrypi.com/viewtopic.php?t=365205

njhollinghurst, 2024-03-01:
The log seems to be from 6.1 which used the 'old-style' mode property, and before the very recent change/fix on rpi-6.6.y to inherit the value from the command line.

It should now be correctly implemented on rpi-6.6.y since d1fc9fc881b94a8af8715f1c0995e54e1838541d (Feb 23).

03stevensmi, 2024-03-01:
> The log seems to be from 6.1 which used the 'old-style' mode property, and before the very recent change/fix on rpi-6.6.y to inherit the value from the command line.
> 
> It should now be correctly implemented on rpi-6.6.y since [d1fc9fc](https://github.com/raspberrypi/linux/commit/d1fc9fc881b94a8af8715f1c0995e54e1838541d) (Feb 23).

I was told to do that was to run `rpi-update`
It is mentioned in the form topic posted above.

If thats not true,
Then what would I need to do to update to 6.6 from the command line?

I also mentioned this in the link above,
But ill mention it here too... 
But I was able to get this working by completely random chance by messing about with xandr.

This works for me:
`xrandr -display :0.0 --output Composite-1 --set mode PAL-N`

But It still doesn't work with the cmdline.txt.
Also this methods as well as the cmdline will not work PERIOD!
with Kodi on rasbian or LibreElec.

njhollinghurst, 2024-03-02:
I don't know why it's not updating for you.

A stop-gap workaround for older kernel versions (this was never documented and will disappear soon) is to add e.g.
```
 drm-rp1-vec.tv_norm=PAL-N
```
to the command line.

We're moving to `tv_mode` because it was supposed to be the new standard. Unfortunately nobody else seems to use it.

popcornmix, 2024-03-02:
> What I was told, was to run `rpi-update` (which I did). It is mentioned in the form topic posted above.
> If thats not true, Then what would I need to do to update to 6.6 from the command line?

What does `uname -a` report?
Copy and paste here the output from running `sudo rpi-update`

03stevensmi, 2024-03-02:
> > What I was told, was to run `rpi-update` (which I did). It is mentioned in the form topic posted above.
> > If thats not true, Then what would I need to do to update to 6.6 from the command line?
> 
> What does `uname -a` report? Copy and paste here the output from running `sudo rpi-update`
I think ive found out what the issue was.
I didn't use sudo the first time around when doing `rpi-update`
It did update... But it mustn't have actually done anything.
So I think he's right!
BTW @popcornmix holy sh**, your a huge blast from the past! 
Of all the people that could have helped me,
I remember you back in the day when I first got into kodi!
I had your repo, Class!! Good memories!
;D

njhollinghurst, 2024-04-22:
I assume this is OK now.  Closing.

closed 2024-04-22 by njhollinghurst

03stevensmi, 2024-04-22:
@njhollinghurst sorry, i wasn't that clear. yes, its sorted. the thing i mentioned above was the issue. thanks guys