raspberrypi/linux #5996: OV64A40 kernel driver not working on 6.6.y

ArduCAM, 2024-02-29:
### Describe the bug

Use rpi-update to update to the latest 6.6.18 kernel and test the ov64a40 camera, and find that the camera cannot be recognized normally.

### Steps to reproduce the behaviour

1. sudo rpi-update
2. Modify /boot/firmware/config.txt and add dtoverlay=ov64a40
3. reboot

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry Pi reference 2023-12-05
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 70cd6f2a1e34d07f5cba7047aea5b92457372e05, stage4

Feb 14 2024 14:57:03 
Copyright (c) 2012 Broadcom
version 409a5df2480fbb1bd36b3d48743ec5a714616f44 (clean) (release) (start)

Linux raspberrypi 6.6.18-v8+ #1736 SMP PREEMPT Fri Feb 23 15:37:38 GMT 2024 aarch64 GNU/Linux

### Logs

[    9.136847] vc4-drm gpu: [drm] fb0: vc4drmfb frame buffer device
[    9.235295] ov64a40 10-0036: sub-device fwnode is an endpoint!
[    9.235358] ov64a40 10-0036: failed to register sensor sub-device: -22
[    9.248915] ov64a40: probe of 10-0036 failed with error -22

### Additional context

Hi @jmondi @kbingham

After a simple debugging, I found that `sd.fwnode` was artificially set as an endpoint, 
https://github.com/raspberrypi/linux/blob/f381965c4af82155c1118c9101b2209b093f9618/drivers/media/i2c/ov64a40.c#L3517-L3518

but v4l2-async clearly requires that the fwnode of subdev cannot be an endpoint. 
https://github.com/raspberrypi/linux/blob/f381965c4af82155c1118c9101b2209b093f9618/drivers/media/v4l2-core/v4l2-async.c#L799-L812

I am not sure why it was set as an endpoint in the first place.
Any ideas?

jmondi, 2024-02-29:
That's because in v6.1 (for which we sent the pull request to RPi kernel) the v4l2-async framework accepted endpoints, and at the time the consensus was to go in that direction
https://elixir.bootlin.com/linux/v6.1/source/drivers/media/v4l2-core/v4l2-async.c

Later on endpoint matching has been removed (see `1029939b3782 ("media: v4l: async: Simplify async sub-device fwnode matching")`) and drivers got updated.

In facts, you can see that the version of the driver in upstream, doesn't set the fwnode
https://elixir.bootlin.com/linux/v6.8-rc6/source/drivers/media/i2c/ov64a40.c#L3510

When moving from v6.1 to v6.6 the matching on endpoint should have been dropped. I can send a PR for this.

jmondi, 2024-02-29:
Pull request sent https://github.com/raspberrypi/linux/pull/5998

jmondi, 2024-02-29:
@ArduCAM please test and confirm the PR (now merged) fixes the issue!

ArduCAM, 2024-03-01:
> @ArduCAM please test and confirm the PR (now merged) fixes the issue!

Thank you for your detailed explanation, I just tested and confirmed that the latest patch fixes this problem.

closed 2024-03-01 by ArduCAM