raspberrypi/linux #5997: SELinux disabled

drparsec, 2024-02-29:
Hi,
I very appreciate your work  and definately want to say "Thanks".

I usually use SELinux to gain a bit more security on my Linux installations. Unfortunately, SELinux is disabled (or to be precise not compiled into the kernel). Thus I am used to download a nwe kernel and the correspoding sources, get the config, enable SELinux, and do the kernel and module compile. Since this is quite time consuming, I would rather ask you if you could enable SELinux in your kernel.

Thanks

timg236, 2024-02-29:
Moved issue to Linux.

Right now, I think this is something that's unlikely to be re-opened unless it's virtually zero cost. The default kernel binaries are targetted for desktop-class OS's with board hardware support. So long as the downstream patches don't prevent an SELinux configuration from being built that seems reasonable.
N.B AppArmor can be installed without rebuilding the kernel and seems to be a better fit for the majority of installs where security enhancements are desired.

awillis, 2024-02-29:
I've shifted to using dietpi for my project, as it's a better fit for other reasons. I think it is relatively safe to say that people who request the use of SELinux are interested in using it, and are probably well aware of AppArmor. SELinux has long been enabled and enforcing by default in other desktop focused distributions like Fedora, and can be enabled in Ubuntu and Debian. The intentional omission of SELinux support keeps RPiOS behind other desktop and debian based distributions.

Kernel size impact could be easily obtained by compiling the kernel with the necessary support, it is only a matter of being willing to take the time do to so. If I get some spare time, I will compile the RPi kernel for the RPi devs and have a look. It's no longer a blocker for me though.

awillis, 2024-03-01:
[image]