raspberrypi/linux #6001: Trying to use rpi-update with a hash always results in "invalid hash given"

Xanaeth, 2024-03-02:
### Describe the bug

when i try to update my raspberry pi 5 firmware to a specific hash,  i always get the same result *** Invalid hash given.

> sudo rpi-update  bd1a634692f769aa4b7af653b00fe5098ef9a521
  *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
  *** Performing self-update
  *** Relaunching after update
  *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 FW_REV:
 BOOTLOADER_REV:b745226b41ac202976ee8307fcb179a1193fab3c
  *** Invalid hash given

> sudo rpi-update fbde695f72b9f63ad140f0efb3853c202c5b3d88
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
FW_REV:
BOOTLOADER_REV:b745226b41ac202976ee8307fcb179a1193fab3c
 *** Invalid hash given

### Steps to reproduce the behaviour

all i do is try:
sudo rpi-update #hash (wherein #hash is any hash )

it always results in the above mentioned *** invalid hash given

i also checked my DNS settings, and i have neither a firewall nor a domain or address filter active on my router
ISP is Spectrum,  and i am located in Ohio USA

i tried via Ethernet and Wifi  , same result on either.

### Device (s)

Raspberry Pi 5

### System

> cat /etc/rpi-issue
 Raspberry Pi reference 2023-12-05
 Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 70cd6f2a1e34d07f5cba7047aea5b92457372e05, stage4

> vcgencmd version
2024/02/16 15:28:41 
Copyright (c) 2012 Broadcom
version 4c845bd3 (release) (embedded)

>uname -r
6.1.0-rpi7-rpi-2712

6by9, 2024-03-04:
rpi-update uses commit hashes from the https://github.com/raspberrypi/rpi-firmware repo, not from this kernel repo, as it updates both kernel and firmware.

popcornmix, 2025-06-20:
Almost certainly the reason for the "invalid hash given" was an invalid hash given.
What hash are you trying to use and where did you get it from?