raspberrypi/linux #6002: missing kernel headers for aarch64

qrp73, 2024-03-02:
### Describe the bug

Missing kernel headers directory for aarch64 v8 kernel, /lib/modules/$(uname -r)/build is a symlink for non existing folder.

### Steps to reproduce the behaviour

1) Install raspi os bookworm aarch64
2) rpi-update rpi-6.6.y
3) sudo apt install raspberrypi-kernel-headers
4) ls /lib/modules/$(uname -r)/build

Expected result: kernel headers directory is present

Actual result: missing folder, /lib/modules/$(uname -r)/build is a symlink for non existing folder.

### Device (s)

Raspberry Pi 4 Mod. B

### System

$ cat /etc/rpi-issue
Raspberry Pi reference 2023-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 40f37458ae7cadea1aec913ae10b5e7008ebce0a, stage4

$ vcgencmd version
Dec  8 2023 12:36:15 
Copyright (c) 2012 Broadcom
version e02d33b3122450accf9dea471a177d3b5623f5ad (clean) (release) (start)

$ uname -a
Linux raspi 6.6.18-v8+ #1 SMP PREEMPT Thu Feb 29 13:37:10 UTC 2024 aarch64 GNU/Linux

### Additional context

I need to build kernel driver, for armhf it compiles ok. Any idea on how to get kernel headers for aarch64?

XECDesign, 2024-03-02:
The old raspberrypi-kernel* packages shouldn't be used on Bookworm. We now have packages which are closer to stock Debian.

In this case, you'd want the `linux-headers-rpi-v8` package.

(For people running the armhf image with an am64 kernel, this won't work)

qrp73, 2024-03-02:
yes, I rolled back kernel to a stock 6.1.0-rpi8-rpi-v8 with:
```
sudo apt install --reinstall raspi-firmware
```

and then installed kernel headers with:
```
sudo apt install linux-headers-rpi-v8
```

Now I can compile driver for aarch64.

But I'm still don't see how to get kernel headers if I want to install more fresh kernel with ```rpi-update rpi-6.6.y```?

XECDesign, 2024-03-03:
If the kernel doesn't come from apt, you'd need to install headers manually or use a tool like [rpi-source](https://github.com/RPi-Distro/rpi-source/) (which I can't vouch for).

qrp73, 2024-03-03:
I tried rpi-source, but it failed with some unknown exception on file open, it seems that it is don't works.
Is there any other way to install kernel headers than download kernel sources and rebuild it?

XECDesign, 2024-03-04:
You shouldn't need to rebuild the whole kernel.

I haven't had to do this in quite a while, so I can't give the full steps off the top of my head. You'd need to check out the relevant source tree, apply the defconfig, download and rename the relevant symvers file (for example, https://github.com/raspberrypi/rpi-firmware/raw/master/Module_2712.symvers). Then you should be able to run make modules_prepare (I think) and build whatever you need to against that source tree.

Most out-of-tree makefiles expect a /lib/modules/$(uname -r)/build and source symlinks to your linux source.