raspberrypi/linux #6006: f2fs support missing in Pi 5 kernel

wscheicher, 2024-03-03:
### Describe the bug

When booting the kernel linux-image-6.1.0-rpi8-rpi-2712 , f2fs is not available.
This works fine with kernel linux-image-6.1.0-rpi8-rpi-v8

The f2fs related settings in /boot/config-6.1.0-rpi8-rpi-v8 and /boot/config-6.1.0-rpi8-rpi-2712 are identical.

### Steps to reproduce the behaviour

grep f2fs /proc/filesystems

Or try to mount a f2fs filesystem

### Device (s)

Raspberry Pi 5

### System

OS:
Raspberry Pi reference 2024-03-02

Firmware:
2023/10/30 16:45:10 
Copyright (c) 2012 Broadcom
version 30de0ba5 (release) (embedded)

Kernel Version:
Fail:
Linux pi64 6.1.0-rpi8-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux
Good:
Linux pi64 6.1.0-rpi8-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux

popcornmix, 2024-03-03:
f2fs is not compatible with a 16K page size (at least prior to [6.7 kernel](https://www.phoronix.com/news/Linux-6.7-F2FS)).
You'll need to run with `kernel=kernel8.img` until this is fixed.

Guiorgy, 2024-07-28:
@popcornmix do you mean I should add that anywhere, for example:

`... rootfstype=f2fs kernel=kernel8.img ...`

What about the `initramfs8`?

popcornmix, 2024-07-28:
Yes, it goes in config.txt

pelwell, 2024-07-28:
Note that kernel=kernel.img goes in config.txt.

xornet-sl, 2025-02-06:
Is that still the case with the current 6.6.74 kernel?
Support for 16k pages was added in f2fs in 2023: https://lwn.net/Articles/946318/

xornet-sl, 2025-02-06:
Ah.. found the merge: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=aea6bf908d730b01bd264a8821159db9463c111c
Looks like it was added only to 6.7 :(
Is there a timeline somewhere when raspberry pi os will get a new kernel?

Guiorgy, 2025-02-07:
> Looks like it was added only to 6.7 :(
> Is there a timeline somewhere when raspberry pi os will get a new kernel?

Dince Rasbian is based on Debian, I assume it uses the same kernel version as upstream. If so, according to [package tracker](https://tracker.debian.org/pkg/linux), Debian 12 stable uses:

>  stable: 6.1.123-1

So, the best bet is probably to wait for Trixie (13)

>  testing: 6.12.11-1

pelwell, 2025-02-07:
Although Raspberry Pi OS (we don't use the name Raspbian) is based on Debian, we choose and build our own kernels. We will be switching to 6.12 as our main release kernel very soon - probably in the next few weeks. If you want to get an early preview of what 6.12 looks like, run `sudo rpi-update next`. It's still technically beta software, so you may want to back up any important data before upgrading, but so far the feedback has been good.