raspberrypi/linux #6015: No HDMI audio with Bookworm Lite Pi-5

moodeaudio, 2024-03-05:
### Describe the bug

There is no audio output from HDMI using fresh Bookworm Lite on Pi-5.

Is there a particular ALSA configuration that has to first be created?

### Steps to reproduce the behaviour

Start with a fresh apt updated Bookworm Lite 64-bit on a Pi-5 and then run speaker-test to card 0. Error 524 should occur.

pi@bk-lite:~ $ speaker-test -D plughw:0,0

speaker-test 1.2.8

Playback device is plughw:0,0
Stream parameters are 48000Hz, S16_LE, 1 channels
Using 16 octaves of pink noise
Playback open error: -524,Unknown error 524

### Device (s)

Raspberry Pi 5

### System

pi@bk-lite:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2023-12-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 2acf7afcba7d11500313a7b93bb55a2aae20b2d6, stage2

pi@bk-lite:~ $ vcgencmd version
2024/02/16 15:28:41 
Copyright (c) 2012 Broadcom
version 4c845bd3 (release) (embedded)

pi@bk-lite:~ $ uname -a
Linux bk-lite 6.1.0-rpi8-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux

### Logs

**pi@bk-lite:~ $ aplay -l**
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

**pi@bk-lite:~ $ amixer -c0 info**
Card sysdefault:0 'vc4hdmi0'/'vc4-hdmi-0'
  Mixer name    : ''
  Components    : ''
  Controls      : 4
  Simple ctrls  : 0

**pi@bk-lite:~ $ amixer -c1 info**
Card sysdefault:1 'vc4hdmi1'/'vc4-hdmi-1'
  Mixer name    : ''
  Components    : ''
  Controls      : 4
  Simple ctrls  : 0

**pi@bk-lite:~ $ cat /boot/firmware/config.txt**
```
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]
```

popcornmix, 2024-03-06:
I've just tried this on a lite install on a pi 5. I think you want to use:
```
$ speaker-test -D sysdefault:CARD=vc4hdmi0
speaker-test 1.2.8

Playback device is sysdefault:CARD=vc4hdmi0
Stream parameters are 48000Hz, S16_LE, 1 channels
Using 16 octaves of pink noise
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 4 to 65536
Period size range from 2 to 32768
Using max buffer size 65536
Periods = 4
was set period_size = 16384
was set buffer_size = 65536
 0 - Front Left
Time per period = 1.374103
 0 - Front Left
```

moodeaudio, 2024-03-06:
I'm still getting the 524 error. There must be some difference in our two systems.
```
Authenticated to bk-lite ([192.168.1.167]:22) using "password".
Linux bk-lite 6.1.0-rpi8-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Wed Mar  6 06:21:37 2024 from 192.168.1.125
pi@bk-lite:~ $ 
pi@bk-lite:~ $ speaker-test -D sysdefault:CARD=vc4hdmi0

speaker-test 1.2.8

Playback device is sysdefault:CARD=vc4hdmi0
Stream parameters are 48000Hz, S16_LE, 1 channels
Using 16 octaves of pink noise
Playback open error: -524,Unknown error 524
pi@bk-lite:~ $ 
```

Whats your `amixer info` output look like?

popcornmix, 2024-03-06:
```
pi@raspberrypi:~ $ amixer info
Card default 'vc4hdmi0'/'vc4-hdmi-0'
  Mixer name	: ''
  Components	: ''
  Controls      : 5
  Simple ctrls  : 1
```
Can you post url from `raspinfo | pastebinit` (`sudo apt install pastebinit` if you don't have it).

moodeaudio, 2024-03-06:
Here is the link.
```
pi@bk-lite:~ $ raspinfo | pastebinit
Unit pipewire.socket could not be found.
Unit pipewire.service could not be found.
Unit pulseaudio.service could not be found.
Unit pulseaudio.socket could not be found.
https://paste.debian.net/hidden/507ca57f/
pi@bk-lite:~ $ 
```

moodeaudio, 2024-03-06:
Yes, thats correct there is no display attached.

vc4-kms-v3d is strict in that case as compared to snd_bcm2835 which allows audio on HDMI even when no display is attached.

Ok, so I'll connect to AVR for HDMI audio tests. I've seen some posts referencing "iec958" and ""IEC958_SUBFRAME_LE". Is this a particular ALSA config thats required?

6by9, 2024-03-06:
Audio on HDMI is inserted into the blanking periods of the video, and is also clocked by the HDMI video rate. No video means no blanking periods or clock.

I haven't looked at the firmware code, but I'd suspect you had to enable a particular HDMI mode in order for HDMI audio to work. You may be able to convince vc4 to do the same via a `video=HDMI-A-1....` entry in cmdline.txt, but, as I've said, audio capabilities are filtered based on the EDID. We are discussing producing a set of pre-canned EDIDs to allow for basic overrides, but not done yet.

Yes the hardware requires IEC958 formatted data. ALSA's iec958 plugin should do this for you, and I believe the relevant configuration file ships by default. There's no option for volume control though.
Pulseaudio or Pipewire will also do it, and Pipewire is installed by default on the non-Lite versions of Raspberry Pi OS.

popcornmix, 2024-03-06:
Yes, a connected hdmi is required to output audio.

The kernel driver for hdmi audio only consumes IEC958_SUBFRAME_LE data (that is what the hardware wants).
ALSA (through `/usr/share/alsa/cards/vc4-hdmi.conf`) can convert PCM data to this format when using the default device (`sysdefault:CARD=vc4hdmi0`)

moodeaudio, 2024-03-06:
Got it. I'll post back after testing later today or tomorrow.
-Tim

moodeaudio, 2024-03-07:
Ok, this string `default:vc4hdmi0` works perfectly. Channel sweep is correct at each speaker.

```
pi@moode-test1:~ $ speaker-test -c6 -D default:vc4hdmi0

speaker-test 1.2.8

Playback device is default:vc4hdmi0
Stream parameters are 48000Hz, S16_LE, 6 channels
Using 16 octaves of pink noise
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 2 to 21845
Period size range from 1 to 10922
Using max buffer size 21844
Periods = 4
was set period_size = 5461
was set buffer_size = 21844
 0 - Front Left
 3 - Front Center
 1 - Front Right
 5 - Rear Right
 4 - Rear Left
 2 - LFE
Time per period = 17.302153
```

In hw_params `format: IEC958_SUBFRAME_LE` is the actual bit depth the same as the source bit depth or is it for example always 0-padded to 32 or something like that?

```
pi@moode-test1:~ $ moodeutl --hwparams
access: MMAP_INTERLEAVED
format: IEC958_SUBFRAME_LE
subformat: STD
channels: 2
rate: 44100 (44100/1)
period_size: 4410
buffer_size: 22050
```

popcornmix, 2024-03-08:
The IEC958 subframe looks like:
```
 0           3  4 7  8              27 28 29 30 31
[sync preamble][aux][audio sample word][V][U][C][P]
```
So audio samples are fixed at 20 bits (zero padded at lsb end if using lower bit depth).

But this is handled by alsa-lib - application code doesn't need to know the details.
See [here](https://github.com/alsa-project/alsa-lib/blob/08bf1a7b0c7fb353c7ca453a73a38b2979d27a72/src/pcm/pcm_iec958.c#L93).

moodeaudio, 2024-03-08:
It looks like the standard allows for 24 bit samples by using the aux bits. In that case the LSB would be bit 4 https://cdn.standards.iteh.ai/samples/101994/5f6ff41160314846bcab050fa205b216/IEC-60958-1-2021.pdf

I think what I'll do initially is just assume the output bit depth is the same as source bit depth for the purposes of reporting in our WebUI and then see what bit depth actually shows up in AVR front-panel displays when users play various formats. Note that my older AVR doesn't report the bit depth.

More testing and I can report that a set of multichannel FLAC and WAV files in my test collection play correctly. These files are 6-channel, 24bit 44.1K or 96K formats.

The other thing I noticed is that there appears to be a hardware mixer.
```
pi@moode-test1:~ $ amixer
Simple mixer control 'PCM',0
  Capabilities: pvolume
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 255
  Mono:
  Front Left: Playback 255 [100%] [0.00dB]
  Front Right: Playback 255 [100%] [0.00dB]
```

It's looking super nice for HDMI audio on the Pi-5 :-)

popcornmix, 2024-03-08:
I think the aux bits are used for audio. So I think you will get 24-bit audio over hdmi (when the PCM format is 24-bit or higher). See [here](https://github.com/alsa-project/alsa-lib/blob/08bf1a7b0c7fb353c7ca453a73a38b2979d27a72/src/pcm/pcm_iec958.c#L110-L112).

moodeaudio, 2024-03-08:
Great info, and thanks!

-Tim

closed 2024-03-08 by moodeaudio