raspberrypi/linux #6020: spidev transfer timeout on Pi 5

makermelissa, 2024-03-08:
### Describe the bug

I am writing some code in c and am having spidev timeouts on the Pi 5. The same code runs fine on the Pi 4. The first spidev transfer is fine, but subsequent transfers result in timeout errors (despite it successfully transferring), which produces a much slower transfer.

### Steps to reproduce the behaviour

1. Install Adafruit Snake Eyes Bonnet
2. Install Raspberry Pi OS 64-bit bookworm Desktop on the Pi 5
3. Update the system
4. Follow directions in https://github.com/makermelissa/Pi_Eyes/blob/wayland-update/README.md to install dependencies
5. Build following same instructions

### Device (s)

Raspberry Pi 5

### System

Raspberry Pi reference 2023-12-05
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 70cd6f2a1e34d07f5cba7047aea5b92457372e05, stage4

2024/01/24 12:16:01
Copyright (c) 2012 Broadcom
version 1aeae380 (release) (embedded)

Linux raspberrypi 6.1.0-rpi8-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux

### Logs

These message repeat in dmesg:
[ 1829.452018] spidev spi0.0: DMA transaction timed out
[ 1829.452018] spidev spi1.2: DMA transaction timed out
[ 1829.452022] spidev spi0.0: SPI transfer failed: -110
[ 1829.452024] spidev spi1.2: SPI transfer failed: -110
[ 1829.452035] spi_master spi0: failed to transfer one message from queue
[ 1829.452035] spi_master spi1: failed to transfer one message from queue
[ 1829.452037] spi_master spi1: noqueue transfer failed
[ 1829.452038] spi_master spi0: noqueue transfer failed

### Additional context

Not 100% sure my code is not the issue, but kind of odd it works fine on Pi 4, but not Pi 5.

makermelissa, 2024-03-08:
This may be a duplicate of https://github.com/raspberrypi/linux/issues/5858.

makermelissa, 2024-03-08:
Setting transfers to 64 bytes or less fixes the timeout issue, but still quite slow.

makermelissa, 2024-03-08:
Perhaps https://github.com/raspberrypi/linux/issues/5696 has popped back up or maybe it's not part of the regular release cycle yet.

pelwell, 2024-03-15:
I think this is caused by a bug in the DMA driver that causes the interrupt-enable bit to be cleared. It is caused by a lack of atomicity, and triggered by having transfers submitted on multiple channels simultaneously.

#6044 should be a fix. By the time you read this (a.k.a. any time after 10:40 GMT on the 15th) you'll be able to install a 6.6 kernel with the fix by running `sudo rpi-update pulls/6044`.

pelwell, 2024-03-15:
And yes, I think this is effectively a dup of #5858.

makermelissa, 2024-03-15:
Thanks. Yes, the fix worked.

makermelissa, 2024-03-15:
Ok, after further testing, I think it's not quite working correctly. I am trying to use both spidev0.0 and spidev1.2 in my code. With the update, the clock signal from spidev1.2 doesn't seem to be working anymore. Like there's tons of activity on MOSI, but SCLK is barely pulsing.

makermelissa, 2024-03-18:
80 MHz. I could try slower if you think that would help, but the same code runs fine on a Pi 4 at that clock rate.

pelwell, 2024-03-18:
And in which directions? Try and help me reproduce your setup.

makermelissa, 2024-03-18:
Pretty much all output. I'm writing to 2 ST7789 displays.

pelwell, 2024-03-19:
A few things:
1. The SPI controllers are clocked off clk_sys (200MHz), using an even divisor. This means that your request for 80MHz actually translates to 50MHz. You would have had more flexibility on Pi 4 where the clock is an even divisor of the VPU/core clock, but there the core clock is prone to changing with the ARM clock.
2. With some combinations of transmits and receives I get RX FIFO overflows:
    ```
    $ dmesg -l err
    [12461.687616] dma dma2chan3: dma2chan3 is non-idle!
    [12461.889989] spidev spi1.2: DMA transaction timed out
    [12461.889993] spidev spi1.2: SPI transfer failed: -110
    [12461.890003] spi_master spi1: failed to transfer one message from queue
    [12461.890006] spi_master spi1: noqueue transfer failed
    ```
    What kernel messages do you get?

makermelissa, 2024-03-19:
1. Hmm, ok. I tried 50MHz and it didn't change anything.
2. No error messages at all. It just doesn't work.

pelwell, 2024-03-19:
> Hmm, ok. I tried 50MHz and it didn't change anything.

Yes, because you've been running it at 50MHz all along. In fact, looking at logic analyser trace it's only driving SCLK half the time, so there's an effective rate of 25MHz.

> No error messages at all. It just doesn't work.

At this point it might be best if you put together a cut-down test program that demonstrates the issue.

makermelissa, 2024-03-19:
Yeah, probably.

makermelissa, 2024-03-20:
Ok, here's some pared down code below(from like 1000 lines).

You can save as **screentest.c** and compile with `cc -o screentest screentest.c -lgpiod -Wall -Ofast`

You will need to enable both spi ports in /boot/firmware/config.txt with:
```
dtparam=spi=on
dtparam=spi1=on
dtoverlay=spi1-3cs
```

Here's the code
```
#include <gpiod.h>
#include <stdio.h>
#include <stdint.h>
#include <unistd.h>
#include <math.h>
#include <fcntl.h>
#include <sys/mman.h>
#include <sys/ioctl.h>
#include <linux/spi/spidev.h>
#include <bcm_host.h>

// CONFIGURATION AND GLOBAL STUFF ------------------------------------------

#define DC_PIN    5             // These pins connect
#define RESET_PIN 6             // to BOTH screens
#define DCMASK    (1 << DC_PIN) // GPIO pin bitmasks of reset
#define	CONSUMER  "fbx2"        // For use with gpiod

#define SCREEN_IPS       2 // Other types WILL NOT WORK!

uint8_t screenType = SCREEN_IPS;

static const uint8_t initIPS[] = {
  // IPS initialization
  0x01, 0x80,       150,      // Soft reset, no args, 150 ms delay
  0x11, 0x80,       255,      // Out of sleep, no args, 500 ms delay
  0x3A, 0x81, 0x55,  10,      // COLMOD, 1 arg, 10ms delay
  0x36,    1, 0x00,           // MADCTL, 1 arg (RGB), no delay,
  0x26,    1, 0x02,           // GAMSET, 1 arg (curve 2 (G1.8)), no delay
  0xBA,    1, 0x04,           // DGMEN, 1 arg (enable gamma), no delay,
  0x21, 0x80,        10,      // INVON, no args, 10 ms delay
  0x13, 0x80,        10,      // NORON, no args, 10 ms delay
  0x29, 0x80,       255,      // DISPON, no args, 500 ms delay
  0x00 },
winIPS[] = {
  0x2A, 4, 0, 0, 0, 239,       // CASET (column set) xstart, xend (MSB first)
  0x2B, 4, 0, 0, 0, 239,       // RASET (row set) ystart, yend (MSB first)
  0x2C,                        // RAMWR (RAM write)
  0x00 };                      // EOD

static const struct {
	const int      width;   // Width in pixels
	const int      height;  // Height in pixels
	const int      bitrate; // Default stable SPI bitrate
	const uint8_t *init;    // Pointer to initialization command list
	const uint8_t *win;     // Pointer to window command list
} screen = { 240, 240, 80000000, initIPS , winIPS };

// Per-eye structure
static struct {
	int        fd;                // SPI file descriptor
	uint16_t  *buf[2];            // Double-buffered eye data 16 BPP
	struct spi_ioc_transfer xfer; // ioctl() transfer struct
} eye[2];

static uint8_t           bufIdx = 0;    // Double-buffering index
static int               bufsiz = 4096; // SPI block xfer size (4K default)
[... 160 lines cut ...]
                uint32_t bytesThisPass, bytesToGo, screenBytes =
                screen.width * screen.height * 2;

                eye[i].xfer.tx_buf = (uintptr_t)eye[i].buf[bufIdx];
                bytesToGo = screenBytes;
                do {
                    bytesThisPass = bytesToGo;
                    if(bytesThisPass > bufsiz) bytesThisPass = bufsiz;
                    eye[i].xfer.len = bytesThisPass;
                    ioctl(eye[i].fd, SPI_IOC_MESSAGE(1), &eye[i].xfer);
                    eye[i].xfer.tx_buf += bytesThisPass;
                    bytesToGo          -= bytesThisPass;
                } while(bytesToGo > 0);
            }
            usleep(250 * 1000);
        }
    }

    gpiod_line_release(reset_line);
    gpiod_line_release(dc_line);
    gpiod_chip_close(chip);

	return 0;
}
```

pelwell, 2024-03-21:
Thanks for that, Melissa. Running it on Pi 4 and Pi 5 the only differences I see are in the timing - Pi 4 runs the bus at 62.5MHz, Pi 5 at 50MHz - and that Pi 5 holds MOSI at the previous value between bytes while Pi 4 returns it to low. Other than that, the traces look the same, with SPI0.0 getting about 900K of data first, followed by SPI1.2.

Here's Pi 4:
[image]

And Pi 5:
[image]

So it sounds like my Pi 5 is not showing the same behaviour as yours. Yes there is no hardware attached, but it's all transmit-only so it shouldn't matter unless some external load/pull is interfering.

makermelissa, 2024-03-21:
Ok, I'll take another look. Starting from a fresh SD might fix it. Also, I could try a second Pi 5 I have to see if it's just one of my Pis misbehaving.

pelwell, 2024-03-21:
In case you find that things are still failing, another option is to run the SPI interfaces without DMA. If you replace the standard Pi 5 dtb file (in /boot/firmware) with this one (https://drive.google.com/file/d/1NGJsJoeFf_K8mVi0lNjO6FIPEOYj9lvf/view?usp=drive_link) you'll gain a new dtparam - `nospidma`, which does what you would expect.