raspberrypi/linux #6021: i2c-rtc overlay does not recognise DS3231 RTC on Rpi Zero running RPi OS 12 Bookworm 32bit

Bucking-Horn, 2024-03-10:
### Describe the bug

I am using the following code to activate a DS3231 RTC module to an RPi Zero:
```
dtparam=i2c1=on
dtdebug=1
dtoverlay=i2c-rtc,ds3231
```
But after rebooting, the overlay did not take over, as 60x8 does not show `UU`:
```
$ sudo i2cdetect -y 1 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --
```
Consequently, there is no rtc device:
```
$ ls -lah /dev/rtc*
ls: cannot access '/dev/rtc*': No such file or directory
```
The log shows:
```
$ sudo vclog -m | grep -A3 i2c-rtc
002338.620: dtdebug: Opened overlay file 'overlays/i2c-rtc.dtbo'
002339.849: brfs: File read: /mfs/sd/overlays/i2c-rtc.dtbo
002394.625: Loaded overlay 'i2c-rtc'
002394.640: dtparam: ds3231=true
002395.749: dtdebug: found override ds3231
002396.233: dtparam: audio=on
```
and
```
~ $ dmesg | grep rtc
[   11.886875] rtc-ds1307: probe of 1-0068 failed with error -5

```
So it seems that the overlay was correctly located and loaded, but did not take effect so that probing it failed.

Looking at https://github.com/raspberrypi/linux/blob/rpi-6.6.y/arch/arm/boot/dts/overlays/i2c-rtc-overlay.dts, it seems to no longer contain a specfic entry for ds3231, where previous versions did.
To me, the 6.6.y i2c-rtc-overlay seems fairly generic, so it's probably just missing a reference to i2c1?

Could you please add support DS3231 again in i2c-rtc overlay?

### Steps to reproduce the behaviour

```
dtparam=i2c1=on
dtdebug=1
dtoverlay=i2c-rtc,ds3231
```
reboot with DS3231 attached

### Device (s)

Raspberry Pi Zero

### System

Raspberry Pi reference 2023-12-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, dc62cc4644c4810772cceb76502e562eb01b3870, stage2

Oct 17 2023 15:42:39 
Copyright (c) 2012 Broadcom
version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start)

Linux rpi-zero-1 6.1.0-rpi8-rpi-v6 #1 Raspbian 1:6.1.73-1+rpt1 (2024-01-25) armv6l GNU/Linux

pelwell, 2024-03-10:
The problem isn't the overlay. First, the overlay does mention ds3231, but it does so by including another file containing all the RTCs: https://github.com/raspberrypi/linux/blob/rpi-6.6.y/arch/arm/boot/dts/overlays/i2c-rtc-common.dtsi
Second, the probe function is being called (which it wouldn't be if the overlay wasn't working, but it is returning -5. Error 5 (sent as -5) is EIO, which indicates a communication error.

In other words, your wiring is probably wrong in some way.

Bucking-Horn, 2024-03-12:
You are almost correct, but it wasn't the wiring itself.

I ran that RTC with 10/Buster on the same board before and only popped in a new sd card with 12/Bullseye, so I was quite sure that the wiring was correct.

Following your advice, I pulled off the RTC and reattached it, and it was only then that I noticed a pin was pushed out slightly at the back of the board. Soldering that back to the board brought the RTC back into action again.

In order to swap the sd card, I had to remove the board from its case.
I guess that pin was in a precarious position to begin with, and removing the board from the case has made it lose contact.

So, case closed. ;)

Thank you for your constructive advice.

Are those probing return codes accessible somewhere, so I wouldn't have to bother you again if similar things would happen?

6by9, 2024-03-12:
> Are those probing return codes accessible somewhere, so I wouldn't have to bother you again if similar things would happen?

They are all standard Linux error codes.
https://github.com/torvalds/linux/blob/master/include/uapi/asm-generic/errno-base.h
https://github.com/torvalds/linux/blob/master/include/uapi/asm-generic/errno.h

pelwell, 2024-03-12:
Here's a present for you. Put the following in a file - I call it `err`:
```
#!/bin/sh
grep -P "^#define(\s+)(\w+)(\s+)$1\b" /usr/include/asm-generic/errno*.h
```
Make it executable (`chmod +x err`) and put it in your path somewhere (e.g.`sudo cp err /usr/local/bin`). Now you can look up any error code easily:
```
$ err 5
/usr/include/asm-generic/errno-base.h:#define   EIO              5      /* I/O error */
```

closed 2024-03-27 by pelwell