raspberrypi/linux #6034: Reading from /dev/hwrng on 6.6.20 hangs on Pi3

dividuum, 2024-03-13:
### Describe the bug

In 6.1, /dev/hwrng was (afaik) immediately readable after booting up my Pi3. Now with 6.6.20 reading /dev/hwrng hangs. The required module is active:

```
root@raspberrypi:~# dmesg | grep bcm2835-rng
[    2.118572] bcm2835-rng 3f104000.rng: hwrng registered
```

I also tested on a Pi5 and reading /dev/hwrng works as expected.

### Steps to reproduce the behaviour

```
root@raspberrypi:~# hexdump /dev/hwrng 
<hangs>
```

### Device (s)

Raspberry Pi 3 Mod. B

### System

```
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, fb56ad562991cf3ae5c96ab50983e1deeaefc7b6, stage4

Feb 29 2024 12:26:47 
Copyright (c) 2012 Broadcom
version f4e2138c2adc8f3a92a3a65939e458f11d7298ba (clean) (release) (start)

Linux raspberrypi 6.6.20+rpt-rpi-v7 #1 SMP Raspbian 1:6.6.20-1+rpt1 (2024-03-07) armv7l GNU/Linux
```

popcornmix, 2024-03-14:
I can confirm this. I'll see what I can find.

popcornmix, 2024-03-14:
This is a version of https://github.com/raspberrypi/linux/issues/5390.

Basically an upstream change made hwrng very slow. 
We added a downstream patch to make it fast again on 6.1.
Upstream added a patch to make it *faster*. and we dropped the downstream patch on 6.6.
But the upstream code is still unusably slow, so I've reinstated the downstream patch.

Note: it doesn't actually hang on Pi3 - my `hexdump /dev/hwrng` did output some hex number after several minutes.
With the downstream patch it's pretty much instant.

You can use `sudo rpi-update` (usual caveats apply) to test the fix.

dividuum, 2024-03-15:
Works again. Thanks :+1:

closed 2024-03-15 by dividuum