raspberrypi/linux #6035: DSI Panel on Pi 5 and 6.6 Kernel

macromorgan, 2024-03-13:
### Describe the bug

My DSI panel (an official Raspberry Pi 7 inch panel) will not work anymore after upgrading to the 6.6 kernel. vclog gives the following error when I try to apply the vc4-kms-dsi-7inch overlay:

006863.114: dterror: invalid target (phandle -908620279)

### Steps to reproduce the behaviour

Upgrade to the 6.6 kernel on 64bit Raspbian Bookworm.

### Device (s)

Raspberry Pi 5

### System

`cat /etc/rpi-issue`
Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage4

`uname -a`
Linux SidePi 6.6.20+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux

`vcgencmd version`
2024/02/16 15:28:41
Copyright (c) 2012 Broadcom
version 4c845bd3 (release) (embedded)

### Logs

`sudo raspinfo`
System Information
------------------

Raspberry Pi 5 Model B Rev 1.0
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"

Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage4

Linux SidePi 6.6.20+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux
Revision        : d04170
Serial          : <<redacted>>
Model           : Raspberry Pi 5 Model B Rev 1.0
Throttled flag  : throttled=0x0
Camera          : vc_gencmd_read_response returned -1 error=1 error_msg="Command not registered"

Videocore information
---------------------

2024/02/16 15:28:41
Copyright (c) 2012 Broadcom
version 4c845bd3 (release) (embedded)

vc_gencmd_read_response returned -1
error=1 error_msg="Command not registered"

Filesystem information
----------------------
Filesystem     1K-blocks     Used Available Use% Mounted on
udev             3948128        0   3948128   0% /dev
tmpfs             824576     6208    818368   1% /run
/dev/nvme0n1p2 245610540 21670488 211447164  10% /
tmpfs            4122848       32   4122816   1% /dev/shm
tmpfs               5120       80      5040   2% /run/lock
/dev/nvme0n1p1    523248    75884    447364  15% /boot/firmware
tmpfs             824560      144    824416   1% /run/user/1000
[... 658 lines cut ...]
232: a1    -- | hi // HDMI0_SCL/AON_SGPIO0 = HDMI_TX0_BSC_SCL
233: a1    -- | hi // HDMI0_SDA/AON_SGPIO1 = HDMI_TX0_BSC_SDA
234: a1    -- | hi // HDMI1_SCL/AON_SGPIO2 = HDMI_TX1_BSC_SCL
235: a1    -- | hi // HDMI1_SDA/AON_SGPIO3 = HDMI_TX1_BSC_SDA
236: a2    -- | hi // PMIC_SCL/AON_SGPIO4 = BSC_M2_SCL
237: a2    -- | hi // PMIC_SDA/AON_SGPIO5 = BSC_M2_SDA

vcdbg log messages
------------------

005600.866: Initial voltage 800000 temp 55415
005798.296: avs_2712: AVS pred 8701 870100 temp 55965
005798.307: vpred 870 mV +0
005802.717: FB framebuffer_swap 1
005831.664: Select resolution HDMI0/2 hotplug 0 max_mode 2
005831.679: Select resolution HDMI1/2 hotplug 0 max_mode 2
006493.014: dtb_file 'bcm2712-rpi-5-b.dtb'
006863.114: dterror: invalid target (phandle -908620279)
006863.123: dterror: merge failed
009832.414: Starting OS 9832 ms
009835.050: 00000040: -> 00000480
009835.066: 00000030: -> 00100080
009835.083: 00000034: -> 00100080
009835.098: 00000038: -> 00100080
009835.115: 0000003c: -> 00100080

pelwell, 2024-03-13:
Something in the Device Tree has got corrupted - either on the filesystem or during the firmware's processing (which shouldn't happen!). Can you post the content of your config.txt, and also the output of these commands:
```
$ md5sum /boot/firmware/bcm2712-rpi-5-b.dtb
$ md5sum /boot/firmware/overlays/vc4-kms-dsi-7inch.dtbo
```

macromorgan, 2024-03-13:
7e45c8e547f7cc77838f9738b3e0a6b8  /boot/firmware/bcm2712-rpi-5-b.dtb
0dff4d158f83b8e860267750e2283cc3  /boot/firmware/overlays/vc4-kms-dsi-7inch.dtbo

Those are the md5sums I show.

PinkFreud, 2024-03-13:
I'm seeing the same issue on a Pi 4 w/ the official Raspberry Pi 7" touchscreen.

The following appears in dmesg:
```
[    3.376801] input: vc4-hdmi-1 HDMI Jack as /devices/platform/soc/fef05700.hdmi/sound/card1/input3
[    3.377195] vc4-drm gpu: bound fef05700.hdmi (ops vc4_hdmi_ops [vc4])
[    3.378203] vc4-drm gpu: bound fe700000.dsi (ops vc4_dsi_ops [vc4])
[    3.378644] vc4-drm gpu: bound fe004000.txp (ops vc4_txp_ops [vc4])
[    3.378992] vc4-drm gpu: bound fe206000.pixelvalve (ops vc4_crtc_ops [vc4])
[    3.379337] vc4-drm gpu: bound fe207000.pixelvalve (ops vc4_crtc_ops [vc4])
[    3.381694] vc4-drm gpu: bound fe20a000.pixelvalve (ops vc4_crtc_ops [vc4])
[    3.381962] vc4-drm gpu: bound fe216000.pixelvalve (ops vc4_crtc_ops [vc4])
[    3.382313] vc4-drm gpu: bound fec12000.pixelvalve (ops vc4_crtc_ops [vc4])
[    3.389014] [drm] Initialized vc4 0.0.0 20140616 for gpu on minor 1
[    4.671511] vc4_dsi fe700000.dsi: transfer interrupt wait timeout
[    4.671528] vc4_dsi fe700000.dsi: instat: 0x00000000
[    4.671539] [drm:vc4_dsi_host_transfer [vc4]] *ERROR* DSI transfer failed, resetting: -110
[    4.887504] tc358762 fe700000.dsi.0: error initializing bridge (-110)
[    4.890052] Console: switching to colour frame buffer device 100x30
[    4.903675] vc4-drm gpu: [drm] fb0: vc4drmfb frame buffer device
```

DSI was working fine on here w/ the 6.1.x kernels, but after updating to 6.6, it only works at boot, going dead around the time the desktop starts.

macromorgan, 2024-03-13:
I suspect this is a different issue (or could be related, who knows) because in my case the devicetree overlay doesn't even apply. I confirmed the md5sum from the latest packages and it does appear correct.

PinkFreud, 2024-03-13:
Well, if it helps, I've confirmed that switching back to 6.1.0-rpi8-rpi-v8 fixes DSI for me, even with the newly installed overlays from the 6.6 kernel package.  In my case, the overlays appear to be fine, but one of the kernel drivers in 6.6 is likely not.

You can try copying your old kernel + initrd from /boot to /boot/firmware (filenames in /boot/firmware should be kernel_2712.img and initramfs_2712 for your Pi 5, iirc) and reboot to see if the display is still broken under the old kernel.

macromorgan, 2024-03-14:
I'm booting from NVME. https://www.waveshare.com/pcie-to-m.2-hat-plus.htm

I have the official power supply plugged into a Y adapter that plugs into both the Pi itself and the DSI controller board. Here is what the cable looks like: https://smarticase.com/products/smartipi-touch-2-cable-replacement-set

I have a Rii Wireless keyboard for input (not sure why it shows up as 3 HID devices, should just be the 2 for keyboard and touchpad). I am using an i2s mono amplifier for audio output, and a USB microphone for audio input.

I also have the official RTC battery and official heatsink/fan assembly.

I'm using a non-official camera module that works, but has a green tint (that I need to troubleshoot further but haven't yet).

Again though, the root cause appears to be the vc4-kms-dsi-7inch.dtbo not cleanly applying, still not figuring out why. Is there a known-good configuration you can do where at a minimum the nvme and vc4-kms-dsi-7inch overlays can be applied cleanly on a Pi 5?

Thank you.

pelwell, 2024-03-14:
> Is there a known-good configuration you can do where at a minimum the nvme and vc4-kms-dsi-7inch overlays can be applied cleanly on a Pi 5?

I think I might not be understanding what you are asking, because isn't that what I've done by copying your config.txt? It shouldn't matter that the hardware isn't present because it isn't until the kernel starts that it will be looked for.

Except for the NVME drive - I'm testing on an SD card, so that is one significant difference.

Some ideas for tests:
* If you have a spare SD card, install a fresh Bookworm image on it, copy over your config.txt, and see if you get any error messages about the overlay application.
* It sounds like you are expecting a lot from one power supply. Can you disconnect most of the external hardware and see if anything changes (apart from not finding the hardware...)?

macromorgan, 2024-03-15:
I disconnected everything except the RTC battery and fan, but am still getting the error in vclog about applying the overlay. I'll troubleshoot this further (though it seems weird that power would cause this issue in the firmware). I'll report back.

macromorgan, 2024-03-15:
Root cause (could point to a firmware problem though). 

When this HAT is present it prevents the overlay for the official DSI panel from applying cleanly: https://www.waveshare.com/pcie-to-m.2-hat-plus.htm

This behavior seems to be new to the 6.6 kernel (possibly due to some changes in how the overlays are done).

pelwell, 2024-03-15:
Would you mind dumping out the HAT EEPROM for me and uploading the file somewhere?
```
$ wget https://github.com/raspberrypi/utils/blob/master/eeptools/eepflash.sh
$ sudo eepflash.sh -r -f=waveshare_m2.eep -t=24c32
```

pelwell, 2024-03-15:
Athough if it's a HAT+ then only the overlay name should be present...

pelwell, 2024-03-15:
And if it does follow the spec, the EEPROM may have one of the new addresses. This might work better:
```
$ wget https://github.com/raspberrypi/utils/blob/master/eeptools/eepflash.sh
$ sudo dtparam i2c_vc=on
$ i2cdetect -y 0
# Look for something between 50 and 53 (probably 51)
$ sudo eepflash.sh -r -f=waveshare_m2.eep -t=24c32 -a=51
# Replace 51 above with the actual EEPROM address - use hexadecimal, with no 0x prefix
```

macromorgan, 2024-03-15:
Something is there at 0x51 on i2c9, but the dump is not working.
dd: failed to open '/sys/class/i2c-adapter/i2c-0/0-0051/eeprom': No such file or directory

macromorgan, 2024-03-15:
output of i2cdetect -y 9

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- 51 -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

pelwell, 2024-03-15:
Try with `-d=9` (that should have been automatically applied):
```
$ sudo eepflash.sh -r -f=waveshare_m2.eep -t=24c32 -a=51 -d=9
```

macromorgan, 2024-03-15:
Here's the EEPROM dump, thank you for your help with this.
[waveshare_m2.zip](https://github.com/raspberrypi/linux/files/14618101/waveshare_m2.zip)

pelwell, 2024-03-15:
Thanks. Unfortunately, even with that eep file flashed into one of our M2 HATs and your config.txt I don't see the corruption. I'm now cloning my card onto an NVME drive so I can boot of that, just in case that makes a difference.

pelwell, 2024-03-15:
Nope - just the same:
```
pi@raspberrypi:~$ sudo vclog -m
006916.181: Initial voltage 800000 temp 56515
007115.849: avs_2712: AVS pred 8469 846900 temp 56515
007118.842: vpred 846 mV +0
007126.865: FB framebuffer_swap 1
007156.225: Select resolution HDMI0/2 hotplug 0 max_mode 2
007159.662: Select resolution HDMI1/2 hotplug 0 max_mode 2
007851.016: dtb_file 'bcm2712-rpi-5-b.dtb'
007958.607: SCMI:A (pos -5)
011339.555: scmi_init
011343.394: Starting OS 11343 ms
011347.175: 00000040: -> 00000480
011348.660: 00000030: -> 00100080
011352.762: 00000034: -> 00100080
011356.864: 00000038: -> 00100080
011360.966: 0000003c: -> 00100080
```

macromorgan, 2024-03-15:
Is that also with the dtoverlay=vc4-kms-dsi-7inch present? Because in order to trigger the issue I need both the hat to be present and the overlay to be loaded. The hat by itself or the overlay by itself works just fine.

macromorgan, 2024-03-15:
I've also taken the hat off and attempted to apply just the pciex1 parameter with the vc4-kms-dsi-7inch, but that also works. I'm at a really weird loss (note that I'm booting from a clean install off of the SD card now).

pelwell, 2024-03-15:
That's with your config.txt as posted above, so yes, it does include the dsi-7inch overlay.

macromorgan, 2024-03-15:
> That's with your config.txt as posted above, so yes, it does include the dsi-7inch overlay.

Let me try a completely different device and see if I can observe the same behavior...

macromorgan, 2024-03-15:
Different device. Different OS (but still fully updated). Same outcome when the Wavshare hat is present.

006233.194: dtb_file 'bcm2712-rpi-5-b.dtb'
006563.509: dterror: invalid target (phandle -1985364471)
006563.520: dterror: merge failed

macromorgan, 2024-03-15:
Even weirder... when I don't hook up the HAT to the GPIO it works... Booted from NVME and all...

So there's some weird combination of this specific HAT, the DSI screen, the 6.6 device trees that the firmware seems to not like. Removing the hat detection, removing the DSI screen overlay, or using the 6.1 kernel and everything works...

004778.381: Initial voltage 800000 temp 49370
004975.776: avs_2712: AVS pred 8662 866200 temp 49920
004975.788: vpred 866 mV +0
004980.179: FB framebuffer_swap 1
005009.126: Select resolution HDMI0/2 hotplug 0 max_mode 2
005009.142: Select resolution HDMI1/2 hotplug 1 max_mode 2
005021.573: HDMI1 edid block 0 offset 0
005021.583: 00ffffffffffff004c2dd50500000000
005021.594: 14130103801009780aee91a3544c9926
005021.606: 0f5054bfee0001010101010101010101
005021.616: 010101010101662150b051001b304070
005021.627: 3600a05a0000001e011d007251d01e20
005021.638: 6e285500a05a0000001e000000fd0018
005021.649: 4b1a5117000a202020202020000000fc
005021.660: 0053796e634d61737465720a2020016d
005034.088: HDMI1 edid block 1 offset 128
005034.099: 02031df1459004052203230907076703
005034.110: 0c001000b82de2000f83010000011d80
005034.120: 18711c1620582c2500a05a0000009e8c
005034.129: 0ad08a20e02d10103e9600a05a000000
005034.139: 1e011d007251d01e206e285500a05a00
005034.149: 00001e00000000000000000000000000
005034.158: 00000000000000000000000000000000
005034.168: 000000000000000000000000000000fb
005034.195: HDMI1: best-mode 2 (limit 2) 1920x1080 60 Hz CEA modes 38000100040000000000000000000000 extensions 1
005034.239: FB0 disp 1 max-fb 1 1920x1080 stride 3840 base 0x3f800000
005563.492: dtb_file 'bcm2712-rpi-5-b.dtb'
009189.524: Starting OS 9189 ms
009192.159: 00000040: -> 00000480
009192.175: 00000030: -> 00100080
009192.191: 00000034: -> 00100080
009192.206: 00000038: -> 00100080
009192.223: 0000003c: -> 00100080

macromorgan, 2024-03-15:
I see the same strange resistor issue. Odd that it worked despite that?
https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/595

macromorgan, 2024-03-15:
So it's not a "solution" per-se and instead points to a possible bug in the firmware, but it looks like if I do `force_eeprom_read=0` in my config.txt everything goes back to the way it was prior to the 6.6 kernel. I'll close an issue here and open a new one with the firmware.

closed 2024-03-15 by macromorgan

pelwell, 2024-03-15:
It would be great if you could replace that force_eeprom_read=0 with a "dtoverlay=" (no overlay name) at the top of config.txt

macromorgan, 2024-03-15:
That did not change anything, it still didn't work. So far the only thing I can to do get it to work is to add `force_eeprom_read=0`.