raspberrypi/linux #6036: External NTFS 8TB HDD does not mount since last update 13.03.24

ozapft, 2024-03-13:
### Describe the bug

Since the last update from March 13, 2024 my external 8TB WD Elements does not mount. It worked fine until this update. I’ve checked the drive with Windows 10 and no errors were found. Other external drives with 4TB and 2TB are working. All drives are GPT NTFS formated.
This 8TB drive is still working with the older Raspi OS with kernel 6.1 from a spare sd card. 
After updating this card I have the same issue.

### Steps to reproduce the behaviour

Connect a 8TB WD Elements drive NTFS formated before and after the update from March 13, 2024. 

### Device (s)

Raspberry Pi 5

### System

Raspberry PI reference 2024-03-12
Version 2024/01/05 version 30cc5f37
Kernel 6.6.20+rpi-2712 #1 SMP PREEMPT Debian q:6.6.20-1+rpt1 (2024-03-07) aarch64

amraz39, 2024-03-13:
I am experiencing the same issue. I also experience inability to shut down or restart the RP5 while the drive with NTFS system is inserted. When I remove the drive, all works as usual.

pelwell, 2024-03-13:
1. Starting without the drive attached, run `dmesg -C | tee ntfs_mount.txt`.
2. Plug in the drive.
3. When the output has stopped, hit Ctrl-C.
4. Post the content of ntfs_mount.txt.

ozapft, 2024-03-14:
The problem has probably already been solved. As of today the drive is mounted again.

amraz39, 2024-03-14:
I still have the issue so am hoping upcoming update will solve this issue. Thx.

ozapft, 2024-03-14:
That's really strange. The kernel version has not changed.

closed 2024-03-14 by ozapft

kungeplay, 2024-03-18:
> I am experiencing the same issue. I also experience inability to shut down or restart the RP5 while the drive with NTFS system is inserted. When I remove the drive, all works as usual.

I met the same issue after I updated 16.03.24

ozapft, 2024-03-18:
I installed a new SD card with the imager. Maybe that was the solution. I also had this instability.

LastColin, 2024-03-20:
> I installed a new SD card with the imager. Maybe that was the solution. I also had this instability.

did flashing a new sd card make you able to mount the external HDD?

ozapft, 2024-03-20:
Yes it did, but later updating the old sd card worked too. I thought the problem was solved because of a new update.

LastColin, 2024-03-20:
Wtf, everything is up to date but i still have this issue with my 4tb hdd. Been trying all kinds of stuff the last 24h

popcornmix, 2024-03-21:
I've not been able to reproduce. A number of users seem to have found it started working after doing "something".

You might want to try removing any stale mount points (e.g. See [here](https://forums.raspberrypi.com/viewtopic.php?p=2204658#p2204658)).

You may also want to try following the text of this warning you may get when trying to manually mount:
```
NTFS is inconsistent. Run chkdsk /f on Windows then reboot it TWICE!
The usage of the /f parameter is very IMPORTANT! No modification was
made to NTFS by this software.
```

shawenyao, 2024-03-25:
Here is my best attempt at summarizing what fixed my problem -
1. Plug the HDD to another Linux machine - it will be mounted. Note that it doesn't have to be another physical machine. The same Raspberry Pi with a different OS works too. Ubuntu did the trick for me, but success with LibreELEC or even Raspberry Pi OS itself have also been reported.
2. Plug back into the original system that used to have difficulty mounting the HDD - the mounting should work now.
3. (Optional) delete the unnecessary folders in /media/username for a peace of mind. If you are unsure which one to delete, you can unplug the HDD and delete everything that's stuck in there.

As for why? I have no clue - please enlighten me.

theneosloth, 2024-04-02:
> I was going to suggest putting `kernel=kernel8.img` in /boot/firmware/config.txt to select the kernel with 4KB pages, just in case the 16KB pages are a factor.
I was still running into this issue. Tried everything including wiping/swapping the sd card, connecting hdd to another device and running chkdsk on the hdd. 

The 4KB kernel is the only thing that resolved it for me

LastColin, 2024-04-02:
> Here is my best attempt at summarizing what fixed my problem -
>
> 1. Plug the HDD to another Linux machine - it will be mounted. Note that it doesn't have to be another physical machine. The same [...]

this is the only thing that has worked for me, temporarily though. after some boots it refuses to mount again and i have to swap SD cards and mount it with ubuntu again for it to mount to PiOS.

jshep321, 2024-06-18:
I am seeing this also.  Note that on bullseye on rpi4, I didn't see this.  I loaded a new pi5 + nvme ssd and when I plug the same 8TB WD elements hdd into the USB port, I get the same result.
[  150.329134] usb 2-1: new SuperSpeed USB device number 2 using xhci-hcd
[  150.353351] usb 2-1: New USB device found, idVendor=1058, idProduct=25a3, bcdDevice=10.21
[  150.353357] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  150.353360] usb 2-1: Product: Elements 25A3
[  150.353363] usb 2-1: Manufacturer: Western Digital
[  150.353365] usb 2-1: SerialNumber: xxx
[  150.354558] usb-storage 2-1:1.0: USB Mass Storage device detected
[  150.354764] scsi host0: usb-storage 2-1:1.0
[  151.369614] scsi 0:0:0:0: Direct-Access     WD       Elements 25A3    1021 PQ: 0 ANSI: 6
[  151.371536] sd 0:0:0:0: [sda] Spinning up disk...
[  151.374857] sd 0:0:0:0: Attached scsi generic sg0 type 0
[  152.393019] ...................ready
[  170.828997] sd 0:0:0:0: [sda] Very big device. Trying to use READ CAPACITY(16).
[  170.989121] sd 0:0:0:0: [sda] 15627986944 512-byte logical blocks: (8.00 TB/7.28 TiB)
[  170.989126] sd 0:0:0:0: [sda] 4096-byte physical blocks
[  170.989561] sd 0:0:0:0: [sda] Write Protect is off
[  170.989565] sd 0:0:0:0: [sda] Mode Sense: 47 00 10 08
[  170.989932] sd 0:0:0:0: [sda] No Caching mode page found
[  170.989934] sd 0:0:0:0: [sda] Assuming drive cache: write through
[  171.300032]  sda: sda1
[  171.300210] sd 0:0:0:0: [sda] Attached SCSI disk
[  171.820787] ntfs3: Max link count 4000
[  171.826610] Unable to handle kernel paging request at virtual address ffff80011a4a0ff0
[  171.834571] Mem abort info:
[  171.837400]   ESR = 0x0000000096000006
[  171.841170]   EC = 0x25: DABT (current EL), IL = 32 bits
[  171.846519]   SET = 0, FnV = 0
[  171.849583]   EA = 0, S1PTW = 0
[  171.852733]   FSC = 0x06: level 2 translation fault
[  171.857628] Data abort info:
[  171.860513]   ISV = 0, ISS = 0x00000006, ISS2 = 0x00000000
[  171.866018]   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
[  171.871087]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
[  171.876418] swapper pgtable: 16k pages, 47-bit VAs, pgdp=0000000001380000
[  171.883233] [ffff80011a4a0ff0] pgd=18000000ffff4003, p4d=18000000ffff4003, pud=18000000ffff4003, pmd=0000000000000000
[  171.893893] Internal error: Oops: 0000000096000006 [#1] PREEMPT SMP
[  171.900181] Modules linked in: ntfs3 sg rfcomm snd_seq_dummy snd_hrtimer snd_seq snd_seq_device algif_hash algif_skcipher af_alg bnep binfmt_misc brcmfmac_wcc brcmfmac brcmutil hci_uart btbcm aes_ce_blk bluetooth cfg80211 aes_ce_cipher ghash_ce gf128mul sha2_ce sha256_arm64 sha1_ce ecdh_generic ecc rpivid_hevc(C) libaes pisp_be raspberrypi_hwmon v4l2_mem2mem videobuf2_dma_contig rfkill videobuf2_memops videobuf2_v4l2 videodev videobuf2_common mc rp1_adc raspberrypi_gpiomem nvmem_rmem joydev uio_pdrv_genirq uio i2c_dev fuse dm_mod ip_tables x_tables ipv6 spidev vc4 snd_soc_hdmi_codec drm_display_helper v3d cec drm_dma_helper gpu_sched drm_kms_helper spi_bcm2835 drm_shmem_helper i2c_brcmstb snd_soc_core snd_compress snd_pcm_dmaengine drm gpio_keys snd_pcm snd_timer snd drm_panel_orientation_quirks backlight
[  171.971897] CPU: 0 PID: 2051 Comm: pool-udisksd Tainted: G         C         6.6.31+rpt-rpi-2712 #1  Debian 1:6.6.31-1+rpt1
[  171.983073] Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
[  171.988923] pstate: 80400009 (Nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  171.995909] pc : __memcpy+0x1a0/0x250
[  171.999582] lr : read_log_page+0xb8/0x188 [ntfs3]
[  172.004309] sp : ffffc000821fb940
[  172.007627] x29: ffffc000821fb940 x28: ffff800017ec5d00 x27: 0000000000000000
[  172.014789] x26: ffff8000627e8950 x25: ffff80001a4a0000 x24: ffff80001ba22000
[  172.021949] x23: 0000000000001000 x22: ffffc000821fba97 x21: 0000000000003000
[  172.029110] x20: ffffc000821fbad0 x19: 0000000000000000 x18: ffffffffffffffff
[  172.036271] x17: ffffd00085862308 x16: ffffd00084cd2c30 x15: 0000000f00000000
[  172.043432] x14: 000000000157f000 x13: 0000000f00000000 x12: 0000000200000012
[  172.050593] x11: 000311f8fffffff8 x10: aac93a37001ef7d0 x9 : ffffd0005a670624
[  172.057754] x8 : 0002686cffffffe8 x7 : 00000000fffffdd5 x6 : 0000000000000003
[  172.064915] x5 : ffff80011ba20000 x4 : ffff80011a4a1000 x3 : ffff80001a4a11fe
[  172.072075] x2 : 00000000ffffe000 x1 : ffff80001a4a3000 x0 : ffff80001ba22000
[  172.079237] Call trace:
[  172.081684]  __memcpy+0x1a0/0x250
[  172.085004]  log_replay+0x990/0x3430 [ntfs3]
[  172.089292]  ntfs_loadlog_and_replay+0x180/0x1b0 [ntfs3]
[  172.094625]  ntfs_fill_super+0x8d0/0x1570 [ntfs3]
[  172.099349]  get_tree_bdev+0x150/0x1f8
[  172.103107]  ntfs_fs_get_tree+0x20/0x38 [ntfs3]
[  172.107656]  vfs_get_tree+0x30/0xf8
[  172.111151]  path_mount+0x408/0xa88
[  172.114647]  __arm64_sys_mount+0x1dc/0x2c0
[  172.118752]  invoke_syscall+0x50/0x128
[  172.122511]  el0_svc_common.constprop.0+0x48/0xf0
[  172.127228]  do_el0_svc+0x24/0x38
[  172.130549]  el0_svc+0x40/0xe8
[  172.133609]  el0t_64_sync_handler+0x100/0x130
[  172.137977]  el0t_64_sync+0x190/0x198
[  172.141647] Code: a93e24a8 a93f2caa d65f03c0 d503201f (a97f348c)
[  172.147760] ---[ end trace 0000000000000000 ]---

popcornmix, 2024-06-18:
There is a potential fix here: https://github.com/raspberrypi/linux/pull/6218
```
sudo rpi-update pulls/6218
```
will get the fix for testing (usual caveats of rpi-update apply).

bruno-garcia, 2024-07-05:
Any idea when this patch is released as a mainline update?