raspberrypi/linux #6037: GPIO.add_event_detect no longer works with the new kernel 6.60

Blackbox-git, 2024-03-13:
### Describe the bug

Since the update to the following kernel, the "add_event_detect" function from the RPi.GPIO library no longer works. 

`Linux outdoor enclosurePi 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux`

The program crashes with the following error.

```
Traceback (most recent call last):
  File "/home/pi/scripts/test/./door_limit_switch_states.py", line 48, in <module>
    GPIO.add_event_detect(door_limit_switch_top, GPIO.BOTH, callback=all, bouncetime = 20)
RuntimeError: Failed to add edge detection
```

I use the following Raspberry

`Raspberry Pi 4 Model B Rev 1.5`

### Steps to reproduce the behaviour

Update to the actual kernel and try the RPi.GPIO library 

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, fb56ad562991cf3ae5c96ab50983e1deeaefc7b6, stage2

Feb 29 2024 12:24:53
Copyright (c) 2012 Broadcom
version f4e2138c2adc8f3a92a3a65939e458f11d7298ba (clean) (release) (start)

pelwell, 2024-03-13:
I suspect this is because the rpi-6.6.y kernel no longer overrides an upstream kernel decision to force the base number of the main GPIO controller to be global GPIO 0. Running `cat /sys/kernel/debug/gpio` will show that GPIO0 (named ID_SDA) is allocated global GPIO number 512. It is going to be painful until the 3rd party libraries are updated, but the choice to stop swimming against the tide was intentional.

Blackbox-git, 2024-03-13:
Okay, if that was the intention, then I know. Since many tutorials on the internet still use the RPi.GPIO library, this will definitely be painful for users.

I also don't think that the RPi.GPIO library will receive an update or be adapted, as it has not received an update for the Raspberry Pi 5 either.

**What sensible alternative should users use? Especially with a trigger for events?**

Old Kernel on a other RPI:

```
gpiochip0: GPIOs 0-53, parent: platform/3f200000.gpio, pinctrl-bcm2835:
 gpio-0   (ID_SDA              )
 gpio-1   (ID_SCL              )
 gpio-2   (SDA1                )
 gpio-3   (SCL1                )
 gpio-4   (GPIO_GCLK           )
 gpio-5   (GPIO5               )
 gpio-6   (GPIO6               )
 gpio-7   (SPI_CE1_N           )
 gpio-8   (SPI_CE0_N           )
 gpio-9   (SPI_MISO            )
 gpio-10  (SPI_MOSI            )
 gpio-11  (SPI_SCLK            )
 gpio-12  (GPIO12              )
 gpio-13  (GPIO13              |sysfs               ) in  lo IRQ
 gpio-14  (TXD1                )
 gpio-15  (RXD1                )
 gpio-16  (GPIO16              )
 gpio-17  (GPIO17              )
 gpio-18  (GPIO18              |sysfs               ) in  lo IRQ
 gpio-19  (GPIO19              |sysfs               ) in  lo IRQ
 gpio-20  (GPIO20              )
 gpio-21  (GPIO21              )
 gpio-22  (GPIO22              |sysfs               ) in  lo IRQ
 gpio-23  (GPIO23              )
 gpio-24  (GPIO24              )
 gpio-25  (GPIO25              |sysfs               ) in  lo IRQ
 gpio-26  (GPIO26              |sysfs               ) in  lo IRQ
 gpio-27  (GPIO27              |sysfs               ) in  lo IRQ
 gpio-28  (NC                  )
 gpio-29  (LAN_RUN_BOOT        )
 gpio-30  (CTS0                )
 gpio-31  (RTS0                )
 gpio-32  (TXD0                )
 gpio-33  (RXD0                )
 gpio-34  (SD1_CLK             )
 gpio-35  (SD1_CMD             )
 gpio-36  (SD1_DATA0           )
 gpio-37  (SD1_DATA1           )
 gpio-38  (SD1_DATA2           )
 gpio-39  (SD1_DATA3           )
 gpio-40  (PWM0_OUT            )
 gpio-41  (PWM1_OUT            )
 gpio-42  (ETH_CLK             )
 gpio-43  (WIFI_CLK            )
 gpio-44  (SDA0                )
 gpio-45  (SCL0                )
 gpio-46  (SMPS_SCL            )
 gpio-47  (SMPS_SDA            )
 gpio-48  (SD_CLK_R            )
 gpio-49  (SD_CMD_R            )
 gpio-50  (SD_DATA0_R          )
 gpio-51  (SD_DATA1_R          )
 gpio-52  (SD_DATA2_R          )
 gpio-53  (SD_DATA3_R          )

gpiochip1: GPIOs 100-101, brcmvirt-gpio, can sleep:
 gpio-100 (                    |ACT                 ) out lo

gpiochip3: GPIOs 500-503, parent: usb/1-1.2:1.0, ftdi-cbus, can sleep:

gpiochip2: GPIOs 504-511, parent: platform/soc:firmware:expgpio, raspberrypi-exp-gpio, can                                                                    sleep:
 gpio-504 (BT_ON               |shutdown            ) out hi
 gpio-505 (WL_ON               )
 gpio-506 (STATUS_LED          )
 gpio-507 (LAN_RUN             )
 gpio-508 (HDMI_HPD_N          |hpd                 ) in  lo ACTIVE LOW
 gpio-509 (CAM_GPIO0           |cam1_regulator      ) out lo
 gpio-510 (CAM_GPIO1           )
 gpio-511 (PWR_LOW_N           |PWR                 ) in  hi
```

New Kernel on the RPI 4:

```
gpiochip0: GPIOs 512-569, parent: platform/fe200000.gpio, pinctrl-bcm2711:
 gpio-512 (ID_SDA              )
 gpio-513 (ID_SCL              )
 gpio-514 (GPIO2               )
 gpio-515 (GPIO3               )
 gpio-516 (GPIO4               |onewire@4           ) out hi
 gpio-517 (GPIO5               )
 gpio-518 (GPIO6               )
 gpio-519 (GPIO7               |lg                  ) in  lo IRQ
 gpio-520 (GPIO8               )
 gpio-521 (GPIO9               )
 gpio-522 (GPIO10              )
 gpio-523 (GPIO11              )
 gpio-524 (GPIO12              )
 gpio-525 (GPIO13              )
 gpio-526 (GPIO14              )
 gpio-527 (GPIO15              )
 gpio-528 (GPIO16              )
 gpio-529 (GPIO17              )
 gpio-530 (GPIO18              |ir-receiver@12      ) in  hi IRQ ACTIVE LOW
 gpio-531 (GPIO19              )
 gpio-532 (GPIO20              )
 gpio-533 (GPIO21              )
 gpio-534 (GPIO22              )
 gpio-535 (GPIO23              )
 gpio-536 (GPIO24              |lg                  ) in  lo IRQ
 gpio-537 (GPIO25              )
 gpio-538 (GPIO26              )
 gpio-539 (GPIO27              )
 gpio-540 (RGMII_MDIO          )
 gpio-541 (RGMIO_MDC           )
 gpio-542 (CTS0                )
 gpio-543 (RTS0                )
 gpio-544 (TXD0                )
 gpio-545 (RXD0                )
 gpio-546 (SD1_CLK             )
 gpio-547 (SD1_CMD             )
 gpio-548 (SD1_DATA0           )
 gpio-549 (SD1_DATA1           )
 gpio-550 (SD1_DATA2           )
 gpio-551 (SD1_DATA3           )
 gpio-552 (PWM0_MISO           )
 gpio-553 (PWM1_MOSI           )
 gpio-554 (STATUS_LED_G_CLK    |ACT                 ) out lo
 gpio-555 (SPIFLASH_CE_N       )
 gpio-556 (SDA0                )
 gpio-557 (SCL0                )
 gpio-558 (RGMII_RXCLK         )
 gpio-559 (RGMII_RXCTL         )
 gpio-560 (RGMII_RXD0          )
 gpio-561 (RGMII_RXD1          )
 gpio-562 (RGMII_RXD2          )
 gpio-563 (RGMII_RXD3          )
 gpio-564 (RGMII_TXCLK         )
 gpio-565 (RGMII_TXCTL         )
 gpio-566 (RGMII_TXD0          )
 gpio-567 (RGMII_TXD1          )
 gpio-568 (RGMII_TXD2          )
 gpio-569 (RGMII_TXD3          )

gpiochip1: GPIOs 570-577, parent: platform/soc:firmware:gpio, raspberrypi-exp-gpio, can sl                                                                   eep:
 gpio-570 (BT_ON               )
 gpio-571 (WL_ON               )
 gpio-572 (PWR_LED_OFF         |PWR                 ) out lo ACTIVE LOW
 gpio-573 (GLOBAL_RESET        )
 gpio-574 (VDD_SD_IO_SEL       |vdd-sd-io           ) out lo
 gpio-575 (CAM_GPIO            |cam1_regulator      ) out lo
 gpio-576 (SD_PWR_ON           |regulator-sd-vcc    ) out lo
 gpio-577 (SD_OC_N             )
```

Trilife, 2024-03-14:
> I suspect this is because the rpi-6.6.y kernel no longer overrides an upstream kernel decision to force the base number of the main GPIO controller to be global GPIO 0. Running `cat [...]

Ok, what is the fix?

pelwell, 2024-03-14:
The fix is for someone to patch RPi.GPIO to locate the global GPIO number for GPIO0, and add that as an offset to the requested GPIO number. In this case it is likely to be 512.

Trilife, 2024-03-14:
Thanks Phil. I guess I'll have to wait for 'someone'. Although I'd be
willing to do such a fix, I have zero experience in RPi for the time being.

Saludos, Greetings
*Jean-Robert STRELE*
COL: +57.321.414.6328
USA: +1.925.922.4231
***@***.***

ᐧ

6by9, 2024-03-14:
Ubuntu have already been down this path (GPIO via sysfs has totally gone there), and Dave Jones has written an RPi.GPIO shim on top of lgpio
https://waldorf.waveform.org.uk/2022/the-one-where-dave-breaks-stuff.html
https://rpi-lgpio.readthedocs.io/en/release-0.4/#

I thought we'd pulled that shim into Raspberry Pi OS, but we only appear to have python3-rpi.gpio and not python3-rpi-gpio.
@XECDesign am I wrong on this one?

andi34, 2024-03-14:
On Pi 5
```
gpiochip0: GPIOs 512-543, parent: platform/107d508500.gpio, gpio-brcmstb@107d508500:

gpio-512 (-                   )

gpio-513 (2712_BOOT_CS_N      |spi10 CS0           ) out hi ACTIVE LOW

gpio-514 (2712_BOOT_MISO      )

gpio-515 (2712_BOOT_MOSI      )

gpio-516 (2712_BOOT_SCLK      )

gpio-517 (-                   )

gpio-518 (-                   )

gpio-519 (-                   )

gpio-520 (-                   )

gpio-521 (-                   )

gpio-522 (-                   )

gpio-523 (-                   )

gpio-524 (-                   )

gpio-525 (-                   )

gpio-526 (PCIE_SDA            )

gpio-527 (PCIE_SCL            )

gpio-528 (-                   )

gpio-529 (-                   )

gpio-530 (-                   )

gpio-531 (-                   )

gpio-532 (PWR_GPIO            |pwr_button          ) in  hi ACTIVE LOW

gpio-533 (2712_G21_FS         )

gpio-534 (-                   )

gpio-535 (-                   )

gpio-536 (BT_RTS              )

gpio-537 (BT_CTS              )

gpio-538 (BT_TXD              )

gpio-539 (BT_RXD              )

gpio-540 (WL_ON               |wl_on_reg           ) out hi

gpio-541 (BT_ON               |shutdown            ) out hi

gpio-542 (WIFI_SDIO_CLK       )

gpio-543 (WIFI_SDIO_CMD       )

gpiochip1: GPIOs 544-547, parent: platform/107d508500.gpio, gpio-brcmstb@107d508520:

gpio-544 (WIFI_SDIO_D0        )

gpio-545 (WIFI_SDIO_D1        )

gpio-546 (WIFI_SDIO_D2        )

gpio-547 (WIFI_SDIO_D3        )

gpiochip2: GPIOs 548-564, parent: platform/107d517c00.gpio, gpio-brcmstb@107d517c00:

gpio-548 (RP1_SDA             )

gpio-549 (RP1_SCL             )

gpio-550 (RP1_RUN             |RP1 RUN pin         ) out hi

gpio-551 (SD_IOVDD_SEL        |vdd-sd-io           ) out hi

gpio-552 (SD_PWR_ON           |sd_vcc_reg          ) out hi

gpio-553 (SD_CDET_N           |cd                  ) in  lo ACTIVE LOW

gpio-554 (SD_FLG_N            )

gpio-555 (-                   )

gpio-556 (2712_WAKE           )

gpio-557 (2712_STAT_LED       |ACT                 ) out hi ACTIVE LOW

gpio-558 (-                   )

gpio-559 (-                   )

gpio-560 (PMIC_INT            )

gpio-561 (UART_TX_FS          )

gpio-562 (UART_RX_FS          )

gpio-563 (-                   )

gpio-564 (-                   )

gpiochip3: GPIOs 565-570, parent: platform/107d517c00.gpio, gpio-brcmstb@107d517c20:

gpio-565 (HDMI0_SCL           )

gpio-566 (HDMI0_SDA           )

gpio-567 (HDMI1_SCL           )

gpio-568 (HDMI1_SDA           )

gpio-569 (PMIC_SCL            )

gpio-570 (PMIC_SDA            )

gpiochip4: GPIOs 571-624, parent: platform/1f000d0000.gpio, pinctrl-rp1:

gpio-571 (ID_SDA              )

gpio-572 (ID_SCL              )

gpio-573 (GPIO2               )

gpio-574 (GPIO3               )

gpio-575 (GPIO4               |sysfs               ) out lo

gpio-576 (GPIO5               |sysfs               ) in  hi IRQ

gpio-577 (GPIO6               |sysfs               ) in  hi IRQ

gpio-578 (GPIO7               )

gpio-579 (GPIO8               |sysfs               ) in  hi IRQ

gpio-580 (GPIO9               )

gpio-581 (GPIO10              )

gpio-582 (GPIO11              |sysfs               ) out lo

gpio-583 (GPIO12              )

gpio-584 (GPIO13              )

gpio-585 (GPIO14              )

gpio-586 (GPIO15              )

gpio-587 (GPIO16              |sysfs               ) in  hi IRQ

gpio-588 (GPIO17              )

gpio-589 (GPIO18              |sysfs               ) out lo

gpio-590 (GPIO19              )

gpio-591 (GPIO20              )

gpio-592 (GPIO21              |sysfs               ) in  hi IRQ

gpio-593 (GPIO22              )

gpio-594 (GPIO23              |sysfs               ) out lo

gpio-595 (GPIO24              )

gpio-596 (GPIO25              |sysfs               ) out lo

gpio-597 (GPIO26              )

gpio-598 (GPIO27              )

gpio-599 (PCIE_RP1_WAKE       )

gpio-600 (FAN_TACH            )

gpio-601 (HOST_SDA            )

gpio-602 (HOST_SCL            )

gpio-603 (ETH_RST_N           |phy-reset           ) out hi ACTIVE LOW

gpio-604 (-                   )

gpio-605 (CD0_IO0_MICCLK      |cam0_reg            ) out lo

gpio-606 (CD0_IO0_MICDAT0     )

gpio-607 (RP1_PCIE_CLKREQ_N   )

gpio-608 (-                   )

gpio-609 (CD0_SDA             )

gpio-610 (CD0_SCL             )

gpio-611 (CD1_SDA             )

gpio-612 (CD1_SCL             )

gpio-613 (USB_VBUS_EN         )

gpio-614 (USB_OC_N            )

gpio-615 (RP1_STAT_LED        |PWR                 ) out hi ACTIVE LOW

gpio-616 (FAN_PWM             )

gpio-617 (CD1_IO0_MICCLK      |cam1_reg            ) out lo

gpio-618 (2712_WAKE           )

gpio-619 (CD1_IO1_MICDAT1     )

gpio-620 (EN_MAX_USB_CUR      )

gpio-621 (-                   )

gpio-622 (-                   )

gpio-623 (-                   )

gpio-624 (-                   )
```

`/boot/firmware/config.txt` worked with
```
gpio=5,6,7,8,16,17,20,21,22,26,27=pu
gpio=9,10,11,12,18,19,23,24,25=op
```
In the past, now only first 5 values take effect for each line.... (Values adjusted for sure)

pelwell, 2024-03-14:
Now you are really crossing the streams. The config.txt lines are interpreted by the firmware, which should then set default pull-ups on some pins and make others outputs. These changes may persist once the kernel has booted, but they may also be overwritten.

1. Are you saying that this changed with the switch to the 6.6 kernel?

2. Can you point out some parts of the `/sys/kernepdebug/gpio` output that show the problem?

XECDesign, 2024-03-14:
> I thought we'd pulled that shim into Raspberry Pi OS, but we only appear to have python3-rpi.gpio and not python3-rpi-gpio.
> @XECDesign am I wrong on this one?

Off the top of my head, I'm not sure. Last time I looked at GPIO libraries, there was some uncertainty about what the right way forward is and what to focus on. I'll revisit when I'm back in April.

andi34, 2024-03-14:
@pelwell we can separate that, sure. Wasn't sure if it's also related here.  But yes, it's new since the switch on 6.6. not sure what's the cause, kernel or other firmware/OS related changes.

To me it looks like there's also a difference on the GPIO mapping (gpio 3 on Pi 4 on 515, Pi 5 on 574), is that true in general or depends on the kernel Version? `Linux 6.6.20+rpt-rpi-2712 aarch64` is installed. Can post more information in April, don't get hands on my Pi next 2 weeks...

andi34, 2024-03-14:
597,598 and 595 are ignored.

[image]

pelwell, 2024-03-15:
The firmware still expects the GPIOs to be numbered from 0 (the firmware and kernel are completely different worlds), so:
```
gpio=5,6,7,8,16,17,20,21,22,26,27=pu
gpio=9,10,11,12,18,19,23,24,25=op
```
was correct.

Trilife, 2024-03-15:
Hello,

If I understand the conversation correctly we are looking for a solution
sometime in April, correct?

Just so I can plan my project around this.

Have a good weekend

Saludos, Regards,

Jean-Robert STRELE
COL +57.321.414.6328
USA +1.925.922.4231
***@***.***

pelwell, 2024-03-15:
> we are looking for a solution sometime in April, correct?

You probably won't get an answer before April, but there is no timeframe for a solution.

So we can understand your situation a bit better:
1. Are you building RPi.GPIO yourself?
2. Are you shipping/going to ship it to your customers (if there are customers)?
3. In other words, would a locally modified version of RPi.GPIO be a workable solution?

Trilife, 2024-03-15:
This is a local project for my pool chemistry. One time solution.

I don't have the knowledge to build/modify my own RPi.gpio.

So, yes, a local solution would be acceptable to me as an interim solution,
until a stock solution is available...

Keep in mind, I'm a newbie at RPi...

Thanks!

Blackbox-git, 2024-03-16:
**First of all, I just wanted to thank you for your great work over the last few years.**

But now to the topic. In my humble opinion, letting die the RPi.GPIO library is one of the biggest mistakes of the Raspberry Pi Foundation. In my humble opinion, this has the following reasons:

1. there are millions of projects running with this library and probably not anymore. Backwards compatibility was always the argument in favor of the Raspberry Pi and against alternatives

2. the GPIOZERO library is not an equivalent replacement. First of all, GPIOS cannot be accessed with several programs at the same time. The new library may be suitable for beginners, but if you are an advanced beginner, this is no longer the case. Not everyone is familiar with sockets or named pipes. The leap is simply too big for many beginners. RPi.GPIO could do this without any problems.

3. if you terminate a program that uses the GPIOZERO library, the current state of the GPIO is not retained after termination. This means that no script can be programmed that changes the state of the pin and is then terminated. Beginners in the web area will find this particularly difficult.

Therefore, in my humble opinion, the library should be adapted for all models, even if the original developer no longer does this, the maintenance could be continued by the Raspberry Pi Foundation.

skaringa, 2024-03-18:
There is a related discussion/ticket at https://sourceforge.net/p/raspberry-gpio-python/tickets/210/ 
It seems that especially for event detection the RPi.GPIO library is outdated.

TimW55, 2024-03-18:
I have been struggling with this issue for a couple of days and have just come across this thread.  

I have several clients who are using RPi4s who will have to stay on the Bullseye OS until a solution is found.  To date **RPi.GPIO** has worked well 

It would be really helpful if the documentation [https://www.raspberrypi.com/documentation/computers/os.html](https://www.raspberrypi.com/documentation/computers/os.html) could be updated to cover the issues raised by @Blackbox-git .

**rpi-lgpio** seems to work well on RPi5s running Bookworm.  Perhaps this could be extended to provide at least an interim solution for RPi4s.

pelwell, 2024-03-18:
> Perhaps this could be extended to provide at least an interim solution for RPi4s.

Does it not work on all Pis?

Blackbox-git, 2024-03-18:
I use a workaround

I use gpiozero for the previous "Event_detect". The rest of the pin control still runs via RPi.GPIO. FOR EXAMPLE:

```
import gpiozero

gpio_pin_input = gpiozero.DigitalInputDevice("MY_PIN_NUMBER", pull_up = None, active_state = True)

# Function for event
def event():
    time.sleep(0.1)
    state = MY_PIN_NUMBER.value
    print("The current state of the PIN is: ", str(state))

gpio_pin_input.when_activated = event
gpio_pin_input.when_deactivated = event

while True:
    time.sleep(1)
```

This makes it possible to use the GPIO-PIN once with an event listener in program 1. In program 2 you can still read the state of the GPIO-PIN via the RPi.GPIO libary with the command GPIO.input("MY_PIN_NUMBER") without getting a "GPIO is busy" warning.

6by9, 2024-03-18:
> @pelwell , Unfortunately I don't think so. On RPi4s **rpi-lgpio** seems to mimic the behaviour of **RPi.GPIO**. Perhaps somebody has found a way to make it work.

I'm confused. It's meant to mimic the behaviour of RPi.GPIO.

Running on a Pi4 with Bookworm here, rpi-lgpio installed via pip3, and interactively doing
```
>>> import RPi.GPIO as GPIO
>>> GPIO.setmode(GPIO.BCM)
>>> GPIO.setwarnings(False) 
>>> led = 4
>>> #Initialize your pin
>>> GPIO.setup(led,GPIO.OUT) 
>>> GPIO.output(led,1) 

>>> GPIO.output(led,0) 
```
has just done the expected when each step was checked via `pinctrl get 4`.
Quit, and GPIO 4 is left as an output with the current state.

I've not looked specifically at the add_event_detect hook.

It is true that 2 different processes can not access the same GPIO as libgpiod enforces ownership, but different processes can access different GPIOs.

TimW55, 2024-03-18:
@pelwell, yes, you are right but I was specifically looking at add_event_detect which I haven't been able to get working.

pelwell, 2024-03-18:
Can you post a small sample program so we can take a look?

pelwell, 2024-03-18:
If it's more than a page you could put it in a gist (see the + Create New button at the top of the page).

Blackbox-git, 2024-03-18:
@pelwell who should post a sample program?

@6by9 this works normal. but "add_event_detect" is broken

TimW55, 2024-03-18:
@Blackbox-git , "add_event_detect works" as expected on RPi4s running Bookworm when using the **rpi-lgpio** package

TimW55, 2024-03-18:
Well, at least it doesn't crash.

Blackbox-git, 2024-03-18:
> @Blackbox-git , "add_event_detect works" as expected on RPi4s running Bookworm when using the **rpi-lgpio** package

Yes. But you cant access from another programm the gpio with this libary.

TimW55, 2024-03-18:
@Blackbox-git my use case has 2 separate processes controlling different gpio pins.  However, the documentation for **rpi-lgpio* states:

> Two processes using RPi.GPIO can happily control the same pin. This is simply not permitted by the Linux gpiochip device and will fail under rpi-lgpio. For example, if another process has reserved GPIO26, and our script also tries to allocate it:

This clearly applies to RPi-5 but I don't know what happens on an RPi-4.

Trilife, 2024-03-18:
Sorry to barge into this conversation. I'm trying to move forward with
my project, which broke under 6.6.0. on the Rpi 4B.

'

import RPi.GPIO as GPIO

GPIO.setmode(GPIO.BOARD)
GPIO.setup(self.input_pin, GPIO.IN, pull_up_down = GPIO.PUD_UP)
GPIO.add_event_detect(self.input_pin, GPIO.BOTH,
callback=self._rising_or_falling, bouncetime=200)

'
Only one process is using this as of now (or at least I can live with one
process for now).

What do I need to do to get up and running again? If this is not the right
venue, please let me know. cheers

ᐧ

TimW55, 2024-03-18:
@Trilife , yes I had the same problem.  Essentially it was solved by setting up a python virtual environment and installing **rpi-lgpio**.  The documentation on this package is very helpful [https://rpi-lgpio.readthedocs.io/en/release-0.4/](https://rpi-lgpio.readthedocs.io/en/release-0.4/).  I didn't have to make any changes to my code.

Trilife, 2024-03-18:
Thanks @TimW55.

I already had a virtual environment. So within that environment (using
PyCharm SSH) I uninstalled the Rpi.GPIO package and installed rpi-lgpio.

replaced 'import RPi.GPIO as GPIO' with 'from RPi import GPIO'. No other
changes to code.

I got an error saying lgpio wasn't found. So I installed lgpio. Now I get
this error:

'/home/Helios/.virtualenvs/PoolController/bin/python
/home/Helios/tmp/pycharm_project_840/pool_main.py
Traceback (most recent call last):
  File
"/home/Helios/.virtualenvs/PoolController/lib/python3.11/site-packages/RPi/GPIO/__init__.py",
line 18, in <module>
    lgpio.SET_PULL_NONE
AttributeError: module 'lgpio' has no attribute 'SET_PULL_NONE'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/Helios/tmp/pycharm_project_840/pool_main.py", line 6, in
<module>
    from RPi import GPIO
  File
"/home/Helios/.virtualenvs/PoolController/lib/python3.11/site-packages/RPi/GPIO/__init__.py",
line 20, in <module>
    lgpio.SET_PULL_NONE = lgpio.SET_BIAS_DISABLE
                          ^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'lgpio' has no attribute 'SET_BIAS_DISABLE'
'

I'm rather new to RPi, so forgive me if I'm making newbie errors here.

ᐧ

ziddey, 2024-03-18:
@Trilife pip install the 0.2.2.0 release from https://github.com/Gadgetoid/PY_LGPIO or create your venv with system site packages: https://forums.raspberrypi.com/viewtopic.php?t=358841#p2151984

Trilife, 2024-03-18:
Hi jn,

tried it with lgpio installed and uninstalled. Getting errors:
'
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting git+https://github.com/Gadgetoid/PY_LGPIO
  Cloning https://github.com/Gadgetoid/PY_LGPIO to
/tmp/pip-req-build-p3di5olj
  Resolved https://github.com/Gadgetoid/PY_LGPIO to commit
09844bc863476ebfca1c4826c05484ce21509f5e
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Building wheels for collected packages: lgpio
  Building wheel for lgpio (pyproject.toml): started
  Building wheel for lgpio (pyproject.toml): finished with status 'error'
Failed to build lgpio

  Running command git clone --filter=blob:none --quiet
https://github.com/Gadgetoid/PY_LGPIO /tmp/pip-req-build-p3di5olj
  Running command git submodule update --init --recursive -q
  error: subprocess-exited-with-error

  Building wheel for lgpio (pyproject.toml) did not run successfully.
  exit code: 1

  [11 lines of output]
  ['lg/lgThread.c', 'lg/lgExec.c', 'lg/lgCmd.c', 'lg/lgDbg.c',
'lg/lgI2C.c', 'lg/lgUtil.c', 'lg/lgNotify.c', 'lg/lgHdl.c', 'lg/lgCtx.c',
'lg/lgFile.c', 'lg/lgSPI.c', 'lg/lgGpio.c', 'lg/lgPthTx.c',
'lg/lgScript.c', 'lg/lgPthAlerts.c', 'lg/lgErr.c', 'lg/lgCfg.c',
'lg/lgMD5.c', 'lg/lgPthSocket.c', 'lg/lgSerial.c', 'lg/rgpiod.c', 'lgpio.i']
  running bdist_wheel
  running build
  running build_py
  file lgpio.py (for module lgpio) not found
  file lgpio.py (for module lgpio) not found
  running build_ext
  building '_lgpio' extension
  swigging lgpio.i to lgpio_wrap.c
  swig -python -o lgpio_wrap.c lgpio.i
  error: command 'swig' failed: No such file or directory
  [end of output]

  note: This error originates from a subprocess, and is likely not a
problem with pip.
  ERROR: Failed building wheel for lgpio
ERROR: Could not build wheels for lgpio, which is required to install
pyproject.toml-based projects
'
oddly, when installing lgpio, it installs version 0.0.0.2 and doesn't want
to install 0.1.0.0. Can't find it.

ᐧ

TimW55, 2024-03-18:
@trilife, I am not sure what is happening but this is what I have done:

>1) Create python virtual environment
>2) activate virtual environment
>3) install rpi-lgpio

```
tjw@RPI-TJW-DEV:~/Temp $ python3 -m venv --system-site-packages "$pyvenv"
tjw@RPI-TJW-DEV:~/Temp $ source "${pyvenv}/bin/activate"
(pyvenv) tjw@RPI-TJW-DEV:~/Temp $ pip3 install rpi-lgpio
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting rpi-lgpio
  Downloading https://www.piwheels.org/simple/rpi-lgpio/rpi_lgpio-0.4-py3-none-any.whl (11 kB)
Installing collected packages: rpi-lgpio
Successfully installed rpi-lgpio-0.4
```

Trilife, 2024-03-19:
> (pyvenv) ***@***.***:~/Temp $ pip3 install rpi-lgpio

Thanks @TimW55,

Not sure what's going on either.

I'll start a new project with it's on ven tomorrow with only rpi-lgpio
tomorrow and see, what happens.

Thanks for your help.

Saludos, Regards,

Jean-Robert STRELE
COL +57.321.414.6328
USA +1.925.922.4231
***@***.***

Trilife, 2024-03-19:
> good morning @TimW55,
>
> So, I installed a new venv and installed the  rpi-lgpio package into it.
>
> Trying to run the following program (copied from my other project. agreed
> that it might not work as is):
> '
>
>
> import RPi.GPIO as GPIO
>
> def _rising_or_falling(channel):
>     if GPIO.input(channel) == GPIO.LOW:  # Pin goes low
>         print('LOW')
>     else:  # Pin goes high
>         print('HIGH')
>
> GPIO.setmode(GPIO.BOARD)
> GPIO.setup(12, GPIO.IN, pull_up_down = GPIO.PUD_UP)
> GPIO.add_event_detect(12, GPIO.BOTH, callback=_rising_or_falling(12), bouncetime=200)
>
> '
>
> And I get the following error:
> '
> /home/Helios/.virtualenvs/TestLGPIO/bin/python
> /tmp/pycharm_project_14/testlgpio.py
> Traceback (most recent call last):
>   File "/tmp/pycharm_project_14/testlgpio.py", line 2, in <module>
>     import RPi.GPIO as GPIO
>   File
> "/home/Helios/.virtualenvs/TestLGPIO/lib/python3.11/site-packages/RPi/GPIO/__init__.py",
> line 13, in <module>
>     import lgpio
> ModuleNotFoundError: No module named 'lgpio'
> '
>
>
> ᐧ

I'm thinking that maybe installing rpi-lgpio would also install lgpio, but
for some reason cannot find it...

 Case in point:
'
***@***.***:~ $ /home/Helios/.virtualenvs/TestLGPIO/bin/python
/home/Helios/.pycharm_helpers/packaging_tool.py install lgpio==0.1.0.0
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: Could not find a version that satisfies the requirement
lgpio==0.1.0.0 (from versions: 0.0.0.2)
ERROR: No matching distribution found for lgpio==0.1.0.0
'

Saludos, Greetings
*Jean-Robert STRELE*
COL: +57.321.414.6328
USA: +1.925.922.4231
***@***.***

ᐧ

Trilife, 2024-03-19:
> @Trilife <https://github.com/Trilife>, if you list the contents of your
> virtual environment you should see something like this:
>
> ~/Temp/pyvenv $ ~/Temp/pyvenv/bin/pip3 list | grep -i gpio
> gpiozero                           2.0
> lgpio                              0.2.2.0
> pigpio                             1.78
> RPi.GPIO                           0.7.1a4
> rpi-lgpio                          0.4
>
> Notice that when you install rpi-lgpio it installs its own version of
> RPi.GPIO. In my python script I have import RPi.GPIO as GPIO. I make no
> reference to lgpio at all. Do your version numbers correspond to those
> above?

Thanks @TimW55,

I see all of those, except that lgpio is at 0.0.0.2

and when I go to https://pypi.org/project/lgpio/,

There is this message:

Installation

pypy can not handle Linux libararies properly at the moment.

You need to install the lgpio module from the lg archive.

See http://abyz.me.uk/lg/download.html

Saludos, Greetings
*Jean-Robert STRELE*
COL: +57.321.414.6328
USA: +1.925.922.4231
***@***.***

ᐧ

TimW55, 2024-03-20:
@trilife, to help understand what is going on could you please post the output from these commands:
```
cat /proc/cpuinfo | tail -n -3
cat /etc/os-release
uname -a
python -V
/usr/bin/pip3 list | grep -i gpio
<pyvenv>/bin/pip3 list | grep -i gpio # Replace <pyvenv> by the path to your virtual python environment
```

litinoveweedle, 2024-03-20:
I can confirm, that by uninstalling RPi.GPIO system package (python3-rpi.gpio) and using pip3 to install rpi.lgpio python module (rpi-lgpio) my legacy python app works, including callbacks.

For anyone suffering this huge issue, please execute this to fix it:

```
sudo apt-get remove --purge python3-rpi.gpio libraspberrypi0 rpi.gpio-common
sudo  pip3 install rpi-lgpio --break-system-packages
```

To the Raspberry maintainers. This is bad, very bad. Please consider repacking rpi.lgpio python module into python3-rpi.gpio system package as an update.

Reasoning: original RPi.GPIO python library will not function with the new kernels anyway, so replacing it with the rpi.lgpio library will fix this transparently for user. I can only add, that finding this issue and what it caused to my automation was terrible pain, which I hope others would not need to go if fixed as proposed. Thank you.

septatrix, 2024-03-20:
This is especially annoying as fixing this within (Docker) containers is even more annoying. lgpio only has a very old version up on pypi which is incompatible with gpiozero so I had to download a tar from a fork and compile that (as it uses native C bindings) inside Docker :/

Trilife, 2024-03-21:
> Hello @TimW55;
>
> Thanks for your continued support to try and get to the bottom of this.
>
> Below are the printouts you asked me to present to you. As an explainter,
> I am using Pycharm to manage my venv via SSH and I use SAMBA to gain access
> to the RPi File system. PyCharm sits on a Win11 box.
>
> Here goes, commands executed from /home/USER/:
>
> 1- ***@***.***:~ $ cat /proc/cpuinfo | tail -n -3
>
> Revision        : d03114
> Serial          : 1000000009978075
> Model           : Raspberry Pi 4 Model B Rev 1.4'
>
> 2- ***@***.***:~ $ cat /etc/os-release
> PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
> NAME="Debian GNU/Linux"
> VERSION_ID="12"
> VERSION="12 (bookworm)"
> VERSION_CODENAME=bookworm
> ID=debian
> HOME_URL="https://www.debian.org/"
> SUPPORT_URL="https://www.debian.org/support"
> [... 33 lines cut ...]
> Thanks for your help!
>
> Cheers
>
> ᐧ

Hello @TimW55 and others.

It's working! It took manually deleting all venvs and tmps that PyCharm had
created and starting from scratch. At that point I only needed to install
rpi-lgpio in the venv. Now everythings seems to be back to normal.

The current gpio.add_event_detect is running in a thread. Did I understand
correctly that with lgpio using multiple threads is not an option? as it
was with rpi.gpio?

Thanks to everyone!

ᐧ

lurch, 2024-03-21:
Just as an FYI for all the various people reading this thread... RPi.GPIO has never been officially supported by Raspberry Pi, and our officially recommended way of dealing with GPIO in Python (**especially** for new projects!) is to use [gpiozero](https://gpiozero.readthedocs.io/), see e.g. https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#gpio-in-python as this works out-of-the-box on Raspberry Pi OS on all models of Raspberry Pi, with no need to setup venvs or install additional Python modules.

pelwell, 2024-03-21:
...but rpi-lgpio is a great way forward for people invested in RPi.GPIO.

skaringa, 2024-03-21:
gpiozero hat the "feature" to automatically clean-up at the end of the script. I want to run a script that turns on some LEDs at given pins and **keep them turned on** after the script has terminated.  This is not possible with gpiozero, but with RPi.GPIO.

TimW55, 2024-03-21:
@lurch, I have to say I am somewhat confused at it seems that gpiozero is built on top of RPi.GPIO. To quote from the documentation for gpiozero:

> GPIO Zero builds on a number of underlying pin libraries, including [RPi.GPIO](https://pypi.org/project/RPi.GPIO) and [pigpio](https://pypi.org/project/pigpio), each with their own benefits.

lurch, 2024-03-21:
> This is not possible with gpiozero, but with RPi.GPIO.

In the gpiozero world, you can achieve that by using `pause()` to prevent your script from terminating - see https://gpiozero.readthedocs.io/en/latest/faq.html#how-do-i-keep-my-script-running

litinoveweedle, 2024-03-21:
OK, have to comment:

> Just as an FYI for all the various people reading this thread... RPi.GPIO has never been officially supported by Raspberry Pi, and our officially recommended way of dealing with GPIO in Python [...]

1. RPi.GPIO is standard Raspbian package (python3-rpi.gpio), you can install it using apt. Now it stopped working with recent distro kernels. So who if not Rpi maintainers do you expect to fix this state?
2. gpiozero is suboptimal for anything else than kids play or simple examples. It is very limiting and very slow. Any other lower layer GPIO library is better. It is also just translation and therefore unnecessary layer. 
3. You can say that RPi.GPIO it is not suggested or supported, but check Internet for any Raspberry Pi GPIO howto and it will 99% point to rpi.GPIO. You can dislike that, you don't have to agree to that, but RPi.GPIO is de-facto standard way of controlling GPIO on Raspberry Pi around the world. People vote in clear preference of RPi.GPIO over gpiozero long time ago.

There is nothing to be gained now by saying: we told you so. This shall be fixed and not for me, but for many poor folks, who now don't have functioning automation and have no clue, what to do. This is breaking change which happened outside of Debian/Raspbian main version release and shall be therefore treated as a bug.

6by9, 2024-03-21:
> @lurch, I have to say I am somewhat confused at it seems that gpiozero is built on top of RPi.GPIO. To quote from the documentation for gpiozero:

gpiozero can sit on top on one of a number of "pin factories" (GPIO accessor libraries), as documented in https://gpiozero.readthedocs.io/en/stable/api_pins.html#changing-the-pin-factory
Whilst rpigpio can be used as a pin factory, the default is now lgpio.

skaringa, 2024-03-21:
> > This is not possible with gpiozero, but with RPi.GPIO.
> 
> In the gpiozero world, you can achieve that by using `pause()` to prevent your script from terminating - see https://gpiozero.readthedocs.io/en/latest/faq.html#how-do-i-keep-my-script-running

No, this is what I want. As I stated in my comment, I want to keep the state of a pin **after the program has terminated**. But if the script calls pause(), it doesn't terminate at all.

aallan, 2024-03-22:
> 1. RPi.GPIO is standard Raspbian package (python3-rpi.gpio), you can install it using apt. Now it stopped working with recent distro kernels. So who if not Rpi maintainers do you expect to fix this state?

Like most of the packages that are installed via the apt package manager RPi.GPIO wasn't written, and isn't maintained by, Raspberry Pi itself. I would expect the RPi.GPIO maintainers to support and maintain their own code.

> 3. You can say that RPi.GPIO it is not suggested or supported, but check Internet for any Raspberry Pi GPIO howto and it will 99% point to rpi.GPIO. You can dislike that, you don't have to agree to [...]

I disagree with all of this. Certainly our own documentation, and the books published by Raspberry Pi Press, do no suggest using RPi.GPIO.

> There is nothing to be gained now by saying: we told you so. This shall be fixed and not for me, but for many poor folks, who now don't have functioning automation and have no clue, what to do. This [...]

If you want RPi.GPIO fixed you should report the problem to the people that can fix it. The project seems to be hosted on Sourceforge. You could potentially file a ticket there https://sourceforge.net/p/raspberry-gpio-python/tickets/ and enquire about support. 

Unfortunately it does look like the developers are pretty unresponsive, there was a ticket opened back in November asking about Pi 5 support, https://sourceforge.net/p/raspberry-gpio-python/tickets/214/, which has not yet received a response.

geekworm-com, 2024-03-25:
We also encountered this problem, but it can be solved by changing the example program to the libgpiod version, we provide an libgpiod example which can be found below:

https://raw.githubusercontent.com/geekworm-com/x728-script/main/sample/x728-v2.x-plsd-gpiod.py

Longrunner262, 2024-04-05:
I'm in the same boat.  Was running a small water flow sensor project coded in Python3 on a 3B+ with Bookwork 5.1.  Updated to Bookworm 5.2 and I get the same edge detection error at the top of the thread.  Yes, I'm using RPi.GPIO because it seems to be the best documented library, both on line and in paper books.  Since the 3B+ is being used as a dedicated controller I suppose my only work around will be to reload with 5.1 or maybe go back to Bullseye.  I agree with Blackbox-git - lots of folks use and like RPi.GPIO.  Me too.

aallan, 2024-04-06:
> I agree with Blackbox-git - lots of folks use and like RPi.GPIO. Me too.

Then you need to persuade the RPi.GPIO authors to update their library. It's not our code, so it's not something that Raspberry Pi can fix.

pelwell, 2024-04-06:
Or switch to rpi-lgpio, which has the same API: https://github.com/waveform80/rpi-lgpio

XECDesign, 2024-04-09:
I've added rpi-lgpio to the repo. You should be able to replace rpi.gpio by running `sudo apt update && sudo apt install python3-rpi-lgpio  --auto-remove --purge`

lurch, 2024-04-09:
:point_up: Is it worth mentioning that in the docs @aallan ? :shrug: 

(also ping @waveform80 as a heads-up that he might get more people using `rpi-lgpio` now :smiley: )

pelwell, 2024-04-09:
And to say thank you, obviously

waveform80, 2024-04-10:
> ☝️ Is it worth mentioning that in the docs @aallan ? 🤷
> 
> (also ping @waveform80 as a heads-up that he might get more people using `rpi-lgpio` now 😃 )

Yikes! As someone semi-connected/responsible for this state of affairs, it's probably time I waded in. I'm not going to respond to everything, but here's some important points:

> But now to the topic. In my humble opinion, letting die the RPi.GPIO library is one of the biggest mistakes of the Raspberry Pi Foundation.

It's not the Pi Foundation's library (they simply went with what the community leapt on as "the" GPIO library), and they're not "letting [it] die". This is a change that's being driven upstream from the Linux kernel. For the full story, I wrote [a long and very boring blog post](https://waldorf.waveform.org.uk/2021/the-pins-they-are-a-changin.html) about it several years ago but the long and short is that the kernel really doesn't like processes bypassing it entirely and stomping on registers it thinks it has exclusive access to, and it's a minor miracle it's worked as well as it has for as long as it did.

The abstraction also breaks the second the hardware changes fundamentally, as it's now had to with the Pi 5 (the GPIOs are no longer driven by the SoC but by the RP1 at the other end of a PCIe bus).

All in all, this state of affairs was *never* going to last.

> 2. the GPIOZERO library is not an equivalent replacement. First of all, GPIOS cannot be accessed with several programs at the same time. The new library may be suitable for beginners, but if you are [...]

> > @Blackbox-git , "add_event_detect works" as expected on RPi4s running Bookworm when using the **rpi-lgpio** package
> 
> Yes. But you cant access from another programm the gpio with this libary.

Yes, it's not an equivalent replacement. However, the complaint that GPIOs cannot be accessed simultaneously by separate processes is to do with the new kernel-based system of driving GPIOs. It's nothing to do with gpiozero (which will happily share pins between processes *if* the underlying driver, like RPi.GPIO, permits it, but when the underlying driver doesn't, like lgpio, there's nothing we can do about it).

Frankly *anything* built on the new kernel-based system of driving GPIOs [*will not support this*](https://rpi-lgpio.readthedocs.io/en/release-0.4/differences.html#simultaneous-access). If you are relying on it, you need to find a new way of doing things, like using threading instead of processes, or IPC between processes (having one process that handles your GPIOs and let your other processes talk to it).

> 3. if you terminate a program that uses the GPIOZERO library, the current state of the GPIO is not retained after termination. This means that no script can be programmed that changes the state of the pin and is then terminated. Beginners in the web area will find this particularly difficult.

Yes, and that's absolutely deliberate (there's a story behind this choice... I should write another blog post about that... *hears the audience groaning*).

It's also arguable whether this is easier or harder for "beginners in the web area". It's only harder if you're using separate processes per request (in which case you're going to get bitten by processes sharing pins anyway).

Switch to a threading or async model and you deal with both problems simultaneously. In the traditional Python [http.server module](https://docs.python.org/3/library/http.server.html), this is as simple as changing one keyword (`ForkingMixIn` to `ThreadingMixIn`) -- I'm afraid I'm rather out of date on other libraries, so I can't give recommendations there.

> Therefore, in my humble opinion, the library should be adapted for all models, even if the original developer no longer does this, the maintenance could be continued by the Raspberry Pi Foundation.

Again, it's not their library, they've never maintained it, and it's not a question of anyone maintaining it. The kernel interfaces it relies upon *are going away*, and the new ones do not permit things like sharing GPIOs between processes.

> good morning @TimW55, So, I installed a new venv and installed the rpi-lgpio package into it. Trying to run the following program (copied from my other project. agreed that it might not work as is): 
> [snip...]
> Traceback (most recent call last): File "/tmp/pycharm_project_14/testlgpio.py", line 2, in <module> import RPi.GPIO as GPIO File "/home/Helios/.virtualenvs/TestLGPIO/lib/python3.11/site-packages/RPi/GPIO/__init__.py", line 13, in <module> import lgpio ModuleNotFoundError: No module named 'lgpio' ' ᐧ

Argh, this is at least partially my fault, but *should* now be corrected.

For a long time, lgpio's had a very out of date source package up on PyPI because joan, the author of lg (the project containing lgpio) [isn't an expert on Python packaging](https://github.com/joan2937/lg/issues/24) which is a convoluted area in recent years, to say the least (ask me [how I know](https://piwheels.org/)...). I finally found a bit of time to dig into this and figure out how to build some pre-compiled wheels of lgpio for PyPI for a variety of Python versions. So, as of a couple of weeks ago there should be a [decent variety of wheels](https://pypi.org/project/lgpio/#files) for the two main architectures available from PyPI.

[... cut ...] /en/latest/differences.html) in the documentation prior to usage.

There *will* be some differences that I simply cannot paper over (like the aforementioned multiple-processes accessing one GPIO problem), and others that it's probably not worth papering over (see the [Debounce section](https://rpi-lgpio.readthedocs.io/en/release-0.4/differences.html#debounce)) because doing so would cause even more grief. The chapter should hopefully set your expectations for what's going to work and what's not.

Trilife, 2024-04-10:
> "/tmp/pycharm_project_14/testlgpio.py", line 2, in import RPi.GPIO as GPIO
> *init*.py", line 13, in import lgpio ModuleNotFoundError: No module named

Thanks @Dave Jones,

Yesterday I broke my SD Card while changing RPi cases and since I was
developing my project on PyCharm via SAMBA and SSH, and only was doing
commits instead of Pushes, I lost 8 weeks' worth of learning/coding! DUH!

But, silver lining, I get to re-evaluate my program structure and clean out
all the trial/errors that have been accumulating on the RPi and the code.

One of the findings was that when installing lgpio inside the venv, it
worked on the first try, drop-in compatible with the snippet of code I had
emailed to someone and as a result got preserved. So, thanks again @Dave
Jones for fixing that part.

On a peripherally related topic: I'm using PyCharm's AI Assistant
extensively in my coding and it is he/she that recommended using RPi.GPIO
in the first place (about 8 weeks ago). So there is some truth to an
earlier comment about RPi.GPIO being the de-facto standard. Yesterday I
told AI Assistant that RPi.GPIO in its existing form was broken and what to
recommend instead: 'gpiozero' or 'RPi.GPIO2' came back the answer. I guess
more coders on PyCharm will need to update their code to lgpio for AI to
catch up.

Cheers.
ᐧ

[6 more replies]