raspberrypi/linux #6041: Pi0 6.6 support

renewedbullet, 2024-03-14:
### Describe the bug

RPI website and RPI imager show the image for pi0 lite was update 3/12 along with all the others, and on the website the legacy os says its kernel 6.6 but the rpi 0w does not update to 6.6 it stays 6.1. Is the RPI 0w being discontinued from future kernel updates?

### Steps to reproduce the behaviour

update to latest lite image, uname -r , sudo apt-get dist-upgrade says nothing to update.

pi@raspberrypi:~ $ uname -r
6.1.21+

### Device (s)

Raspberry Pi Zero W / WH

### System

latest 32bit lite image released 3/12

pi@raspberrypi:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2024-03-12
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, b9b581e0a62afb7200d4b0b72ec2c43485fd49cc, stage2

pi@raspberrypi:~ $ vcgencmd version
Mar 17 2023 10:52:42
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux

pelwell, 2024-03-14:
We are not dropping support for Pi Zero. Did you reboot after the update?

XECDesign, 2024-03-14:
I think the issue is about the legacy release, which hasn't had any kernel or firmware updates in a while.

renewedbullet, 2024-03-14:
RPI imager only gives the option for (legacy) for the pi0
[image]

renewedbullet, 2024-03-14:
and sudo apt-get dist-upgrade does not work either. multiple reboots

delgh1, 2024-04-18:
Errh, you need to change the release from bullseye to bookworm in `/etc/apt/source.list`

XECDesign, 2024-04-18:
> Errh, you need to change the release from bullseye to bookworm in `/etc/apt/source.list`

That would cause a lot of problems. Do not do this.

andrum993, 2024-04-18:
> ### Describe the bug
> RPI website and RPI imager show the image for pi0 lite was update 3/12 along with all the others, and on the website the legacy os says its kernel 6.6 ...

No it doesn't - **the website says that Raspberry Pi OS legacy is still on kernel version 6.1**. I'm running Raspberry Pi OS legacy myself, and can confirm that it does indeed use kernel version 6.1. This is entirely expected - you would not expect the legacy OS to get a bump in kernel version - it only gets a new kernel if there are security issues that need patched.

pelwell, 2024-04-18:
> the website says that Raspberry Pi OS legacy is still on kernel version 6.1

Indeed it does.

closed 2024-04-18 by pelwell