raspberrypi/linux #6042: SO_TIMESTAMPNS Support is absent in the kernal?

SpiderUnderUrBed, 2024-03-14:
### Describe the bug

I was wondering whether or not this is the right place for this but this does seem like a bug as a kernal level feature just seemed to break? I posted questions on reddit and the forum about this issue. However I think that posting there maybe was a mistake, anyways here is the issue:
SO_TIMESTAMPNS is non-existent in the kernal or had been disabled due to some bug, there is no indication on my system that this feature exists now and maybe i need a kernal patch or this is a issue with the kernal itself, recently unbound hasnt been working and so has things like nload, the wifi monitoring tool, it graphs your wifi usage from certain interfaces and around the same time SO_TIMESTAMPNS stopped working its graphing feature stopped working:
```

[1710322190] unbound[1:0] error: setsockopt(..., SOTIMESTAMPNS, ...) failed: Protocol not available

[1710322190] unbound[1:0] warning: socket timestamping is not available

[1710322190] unbound[1:0] error: setsockopt(..., SO_TIMESTAMPNS, ...) failed: Protocol not available

[1710322190] unbound[1:0] warning: socket timestamping is not available

[1710322190] unbound[1:0] error: setsockopt(..., SO_TIMESTAMPNS, ...) failed: Protocol not available

[1710322190] unbound[1:0] warning: socket timestamping is not available

```
Dmesg (the log is uneventful):
https://pastebin.com/4ibdq4i9

While I have unbound and alot of my services in docker-compose, sending the docker compose file is probably irrelevent to this bug

### Steps to reproduce the behaviour

I dont know what kernal version this bug affects, I am on kernal version  6.1.0-rpi8-rpi-v8, my OS is Debian GNU/Linux 12 (bookworm) aarch64

While the unbound issue is new you can test SO_TIMESTAMPNS support by trying to run this in python or the equivalent is C
`
import socket`
sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
sock.setsockopt(socket.SOL_SOCKET, socket.SO_TIMESTAMPNS, 1)
`
This should return the error about SO_TIMESTAMPNS

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry Pi reference 2023-12-05
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 70cd6f2a1e34d07f5cba7047aea5b92457372e05, stage4

Oct 17 2023 15:39:16
Copyright (c) 2012 Broadcom
version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start)

Linux raspberrypi 6.1.0-rpi8-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux

### Logs

Dmesg (the log is uneventful):
https://pastebin.com/4ibdq4i9

alessandromrc, 2024-03-29:
This is entirely not a linux kernel issue, on my Windows machine I get the same issue...

I guess this has been deprecated or it probably never existed...

I've noticed there are multiple threads on StackOverflow about it and around the net.

The solution I've noticed running is this one:

```py
import socket

sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
SO_TIMESTAMPNS = 35

try:
    sock.setsockopt(socket.SOL_SOCKET, SO_TIMESTAMPNS, 1)
except socket.error as e:
    print("Error:", e)
```

(Not sure if it works by any means but it gives no execution error, also seen [here](https://stackoverflow.com/questions/46302308/python-networking-timestamp-so-timestamp-so-timestampns-so-timestamping))

lasselj, 2024-04-29:
It's very old and uncool. Can't you use SO_TIMESTAMPING instead?

SpiderUnderUrBed, 2024-05-08:
This was a issue with unbounds source code. Unbound is a DNS resolver. I figured id post this issue here in hopes that it might be added back into the kernal if the amount of people using unbound or those who have this specific issue with SO_TIMESTAMPNS made it worthwhile. 

Currently I had to remove unbound from my raspberry pi. Maybe in a future update unbound might fix this, or maybe there are others experincing the same issue with legacy applications. But given the lack of traction from this issue i doubt it. 

I might close this issue soon

alessandromrc, 2024-05-08:
> This was a issue with unbounds source code. Unbound is a DNS resolver. I figured id post this issue here in hopes that it might be added back into the kernal if the amount of people using unbound or [...]

Sure, do you have any issue open on the Unbound source? Would be nice to see what the problem is, also I noticed around kernel docs this:

https://www.kernel.org/doc/Documentation/networking/timestamping.txt

As you can see the naming convention seems to have changed to SO_TIMESTAMP_NEW or SO_TIMESTAMP_OLD.

closed 2025-03-20 by SpiderUnderUrBed