raspberrypi/linux #6047: i2c_designware 1f00074000.i2c: controller timed out with 2024-03-12 bookworm

mgrouch, 2024-03-16:
### Describe the bug

dmesg errors  with 2024-03-12 bookworm

````
[44742.269462] i2c_designware 1f00074000.i2c: controller timed out
[44743.293457] i2c_designware 1f00074000.i2c: controller timed out
[44744.317464] i2c_designware 1f00074000.i2c: controller timed out
[44745.341468] i2c_designware 1f00074000.i2c: controller timed out
[44746.365456] i2c_designware 1f00074000.i2c: controller timed out
[44747.389464] i2c_designware 1f00074000.i2c: controller timed out
[44748.413452] i2c_designware 1f00074000.i2c: controller timed out
[44749.437452] i2c_designware 1f00074000.i2c: controller timed out
[44750.461451] i2c_designware 1f00074000.i2c: controller timed out
[44751.485452] i2c_designware 1f00074000.i2c: controller timed out
[44752.509453] i2c_designware 1f00074000.i2c: controller timed out
[44753.537435] i2c_designware 1f00074000.i2c: controller timed out

````

### Steps to reproduce the behaviour

just let it run with IMU attached

### Device (s)

Raspberry Pi 5

### System

````
user@lysmarine:~ $ cat /etc/rpi-issue 
Raspberry Pi reference 2024-03-12
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 9a13a81115a8915885e2cc40fed2092e28894d41, stage2
````

pelwell, 2024-03-18:
There's a PR (#6050) that may help with this issue. Similar failures are seen when the device fails to ACK data in time, and the PR changes the mark/space ratio on SCL to give the device longer to respond at any given clock speed.

You can install a trial, beta kernel containing the fix by running `sudo rpi-update pulls/6050`. Take care to back up any important data in your image before updating in case there is a regression.

mgrouch, 2024-03-20:
Is there workaround for an existing released kernel? 

Is that due to i2c bus being faster on pi5?
Is there a way to slow down i2c bus with some options?

thanks

pelwell, 2024-03-20:
> Is there workaround for an existing released kernel?

Not really, no, other than to run it slower. You may be able to improve matters by forcing an internal pull-down on SDA (`pinctrl 2 pd`), but I would advise anyone to upgrade instead.

> Is there a way to slow down i2c bus with some options?

The default with `dtparam=i2c_arm` is 100kHz, which is not fast. Specific overlays may force higher clock speeds.