raspberrypi/linux #6048: X window not booting after recent upgrade

gqcao, 2024-03-16:
### Describe the bug

After an upgrade to the new kernel, the x window no longer boots up. Tried to reinstall a complete new system, but it does not help.. 

### Steps to reproduce the behaviour

startx
```
(EE)
(EE) Backtrace:
xinit: giving up
xinit: unable to conect to X server: Connection refused
xinit: server error
```

### Device (s)

Raspberry Pi 4 Mod. B

### System

RPi4 Model B, linux-rpi 6.6.21-2, linux-firmware-20240312

popcornmix, 2024-03-18:
> After an upgrade to the new kernel, the x window no longer boots up. Tried to reinstall a complete new system, but it does not help..

Can you explain exactly what you did with the "new system"?
e.g. what image did you flash, what was the minimum you needed to configure/install to get the failure?

gqcao, 2024-03-18:
Hi!

Thank you for the response.

The image that I flashed is `ArchLinuxARM-2023.03-rpi-armv7-rootfs.tar.gz` built on 01-Mar-2023 02:57, and the minimum setup is an essential system upgrade at the very beginning, clock time, locale, and i3 to boot up X windows in xinit. Been repeating the installation multiple times in last few years and first time encounter this..

popcornmix, 2024-03-18:
This isn't the right place for Arch Linux issues.
If you can see the same issue with RPiOS we may be able to help, but otherwise you'd be better off asking on an Arch Linux forum.

gqcao, 2024-03-18:
Good comment. I did not manage to trace down the problem to either `linux-rpi 6.6.21-2` or `linux-firmware 20240312.3b128b60-1`, as both made an update recently, or it could be the compatibility issue between the two. But I would like to raise the question here, as such a rudimentary problem may impact quite a few people, and it is likely coming from RPi image. Thanks!

popcornmix, 2024-03-18:
I'm not seeing this issue reported by RPiOS users (RPiOS recently bumped to 6.6.20 kernel),
so until I see an RPiOS report of this, it seems more likely this is an Arch issue.

gqcao, 2024-03-22:
Thank you for the help, and I confirm the issue is related to `libunwind` and downgrading it to 1.7.2-1-armv7 works.

closed 2024-03-22 by gqcao