raspberrypi/linux #6049: brcmfmac: brcmf_set_channel: set chanspec fail, reason -52

starsareintherose, 2024-03-17:
### Describe the bug

I used the rpi5 with Mcuzone MPS2280P hat for SSD, I booted the archlinux arm from nvme SSD with linux-rpi.

RPi 5 still suddenly disconnects the internet and SSH and reports `brcmfmac: brcmf_set_channel: set chanspec fail, reason -52`

### Steps to reproduce the behaviour

When I start the NetworkManger on Arch Linux ARM, I got the debug msg

```
brcmfmac: brcmf_set_channel: set chanspec ????? fail, reason -52
```started

### Device (s)

Raspberry Pi 5

### System

```
# cat /etc/issue
Arch Linux \r (\l)

```

```
# vcgencmd version
2024/02/16 15:28:41 
Copyright (c) 2012 Broadcom
version 4c845bd3 (release) (embedded)
```

```
# uname -a
Linux fruticicola 6.6.21-2-rpi #1 SMP PREEMPT Wed Mar 13 14:11:28 MDT 2024 aarch64 GNU/Linux
```

### Logs

<details>

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[    0.000000] Linux version 6.6.21-2-rpi (builduser@leming) (aarch64-unknown-linux-gnu-gcc (GCC) 12.1.0, GNU ld (GNU Binutils) 2.38) #1 SMP PREEMPT Wed Mar 13 14:11:28 MDT 2024
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 5 Model B Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000002000000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x0000000002000000..0x0000000015ffffff (327680 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x0000000000000000..0x000000000007ffff (512 KiB) nomap non-reusable atf@0
[    0.000000] OF: reserved mem: 0x000000003fd165a0..0x000000003fd165e4 (0 KiB) nomap non-reusable nvram@0
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000007ffff]
[    0.000000]   node   0: [mem 0x0000000000080000-0x000000003f7fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA: 2048 pages in unavailable ranges
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[... 678 lines cut ...]
wifi.powersave = 2
```

Before I did so, it won't use internet.

## Try2

Before installing rpi5-eeprom, the internet can be connected, however, when I tried to use the internet for any ways or ssh to the rpi5, it suddenly disconnected.

After installing rpi5-eeprom, it can be connected, however, usually, it still suddenly disconnects the internet and SSH.

## Addtional notes

I remove the `kms` in `/etc/mkinitcpio.conf`

so, the final hooks are
```
HOOKS=(base udev autodetect modconf keyboard keymap consolefont block filesystems fsck)
```

Without doing so, I'll get the report 
```
==> WARNING: errors were encountered during the build. The image may not be complete.
error: command failed to execute correctly
```

pelwell, 2024-03-17:
I don't understand why you mention the brcmfmac message and not the rather more worrying:
```
[ 313.413817] nvme nvme0: controller is down; will reset: CSTS=0xffffffff, PCI_STATUS=0x10
[ 313.422179] nvme nvme0: Does your device have a faulty power saving mode enabled?
[ 313.429908] nvme nvme0: Try "nvme_core.default_ps_max_latency_us=0 pcie_aspm=off" and report a bug
[ 313.469937] nvme0n1: I/O Cmd(0x2) @ LBA 1310068272, 32 blocks, I/O Error (sct 0x3 / sc 0x71)
[ 313.478730] I/O error, dev nvme0n1, sector 1310068272 op 0x0:(READ) flags 0x80700 phys_seg 4 prio class 2
[ 313.517822] nvme 0000:01:00.0: enabling device (0000 -> 0002)
[ 313.531362] nvme nvme0: 4/0/0 default/read/poll queues
```
which happened 4 seconds earlier.

How are you powering this Pi 5? Spontaneous errors like this that affect multiple subsystems are classic symptoms of insufficient power.

starsareintherose, 2024-03-18:
Previously I boot rpi os via tf card without hat using official rpi power… wifi is also influenced and hardly to be connected, I supposed that this is the same issue, I will check later.

Currently I use official adaptors to power rpi5, 5v3a power for hat, 3a is okay for the hat.

Before I make power_save off, usually the nvme + power_save dmesg presents before brcmfmac instead of nvme only

starsareintherose, 2024-03-18:
<details>
```
[    4.942937] Loaded X.509 cert 'benh@debian.org: 577e021cb980e0e820821ba7b54b4961b8b4fadf'
[    4.952169] rpi-gpiomem 107d508500.gpiomem: window base 0x107d508500 size 0x00000040
[    4.961276] Loaded X.509 cert 'romain.perier@gmail.com: 3abbc6ec146e09d1b6016ab9d6cf71dd233f0328'
[    4.969906] rpi-gpiomem 107d508500.gpiomem: initialised 1 regions as /dev/gpiomem1
[    4.977139] Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    4.984878] rpi-gpiomem 107d517c00.gpiomem: window base 0x107d517c00 size 0x00000040
[    4.991632] Loaded X.509 cert 'wens: 61c038651aabdcf94bd0ac7ff06c7248db18c600'
[    4.998884] rpi-gpiomem 107d517c00.gpiomem: initialised 1 regions as /dev/gpiomem2
[    5.014060] rpi-gpiomem 107d504100.gpiomem: window base 0x107d504100 size 0x00000020
[    5.022007] rpi-gpiomem 107d504100.gpiomem: initialised 1 regions as /dev/gpiomem3
[    5.033953] brcmstb-i2c 107d508200.i2c:  @97500hz registered in interrupt mode
[    5.035080] rpi-gpiomem 107d510700.gpiomem: window base 0x107d510700 size 0x00000020
[    5.041631] mc: Linux media interface: v0.10
[    5.050469] brcmstb-i2c 107d508280.i2c:  @97500hz registered in interrupt mode
[    5.061910] rpi-gpiomem 107d510700.gpiomem: initialised 1 regions as /dev/gpiomem4
[    5.071619] rpi-gpiomem 1f000d0000.gpiomem: window base 0x1f000d0000 size 0x00030000
[    5.079645] rpi-gpiomem 1f000d0000.gpiomem: initialised 1 regions as /dev/gpiomem0
[    5.088072] brcmfmac: F1 signature read @0x18000000=0x15264345
[    5.091497] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.107945] [drm] Initialized v3d 1.0.0 20180419 for 1002000000.v3d on minor 0
[    5.108443] videodev: Linux video capture interface: v2.00
[    5.144151] rpivid_hevc: module is from the staging directory, the quality is unknown, you have been warned.
[    5.155824] rpivid 1000800000.codec: bcm2712_iommu_of_xlate: MMU 1000005100.iommu
[    5.158412] Bluetooth: Core ver 2.22
[    5.167447] pispbe 1000880000.pisp_be: bcm2712_iommu_of_xlate: MMU 1000005100.iommu
[    5.175510] NET: Registered PF_BLUETOOTH protocol family
[    5.176051] rpivid 1000800000.codec: Device registered as /dev/video19
[    5.185214] Bluetooth: HCI device and connection manager initialized
[    5.195887] Bluetooth: HCI socket layer initialized
[    5.201613] Bluetooth: L2CAP socket layer initialized
[    5.201623] Bluetooth: SCO socket layer initialized
[    5.218527] pispbe 1000880000.pisp_be: pispbe_probe: HW version:  0x02252700
[    5.226259] pispbe 1000880000.pisp_be: pispbe_probe: BatchStatus: 0x00000000
[    5.233532] pispbe 1000880000.pisp_be: pispbe_probe: Status:      0x00000000
[    5.240871] Bluetooth: HCI UART driver ver 2.3
[    5.241147] pispbe 1000880000.pisp_be: Register nodes for group 0
[    5.246510] Bluetooth: HCI UART protocol H4 registered
[    5.260395] Bluetooth: HCI UART protocol Three-wire (H5) registered
[    5.260894] pispbe 1000880000.pisp_be: input device node registered as /dev/video20
[    5.267549] hci_uart_bcm serial0-0: supply vbat not found, using dummy regulator
[    5.274853] pispbe 1000880000.pisp_be: tdn_input device node registered as /dev/video21
[    5.282010] Bluetooth: HCI UART protocol Broadcom registered
[    5.290146] pispbe 1000880000.pisp_be: stitch_input device node registered as /dev/video22
[    5.290463] hci_uart_bcm serial0-0: supply vddio not found, using dummy regulator
[    5.309205] usbcore: registered new interface driver brcmfmac
[    5.311516] pispbe 1000880000.pisp_be: hog_output device node registered as /dev/video23
[    5.325448] pispbe 1000880000.pisp_be: output0 device node registered as /dev/video24
[    5.331025] macb 1f00100000.ethernet end0: PHY [1f00100000.ethernet-ffffffff:01] driver [Broadcom BCM54213PE] (irq=POLL)
[    5.333383] pispbe 1000880000.pisp_be: output1 device node registered as /dev/video25
[    5.344245] macb 1f00100000.ethernet end0: configuring for phy/rgmii-id link mode
[    5.354263] pps pps0: new PPS source ptp0
[    5.359660] pispbe 1000880000.pisp_be: tdn_output device node registered as /dev/video26
[    5.363735] macb 1f00100000.ethernet: gem-ptp-timer ptp clock registered.
[    5.371827] pispbe 1000880000.pisp_be: stitch_output device node registered as /dev/video27
[    5.387011] pispbe 1000880000.pisp_be: config device node registered as /dev/video28
[    5.395155] pispbe 1000880000.pisp_be: Register nodes for group 1
[    5.401327] pispbe 1000880000.pisp_be: input device node registered as /dev/video29
[    5.409071] pispbe 1000880000.pisp_be: tdn_input device node registered as /dev/video30
[    5.409094] pispbe 1000880000.pisp_be: stitch_input device node registered as /dev/video31
[    5.425433] pispbe 1000880000.pisp_be: hog_output device node registered as /dev/video32
[    5.435433] pispbe 1000880000.pisp_be: output0 device node registered as /dev/video33
[    5.443532] pispbe 1000880000.pisp_be: output1 device node registered as /dev/video34
[    5.451488] pispbe 1000880000.pisp_be: tdn_output device node registered as /dev/video35
[    5.459668] pispbe 1000880000.pisp_be: stitch_output device node registered as /dev/video36
[    5.468109] pispbe 1000880000.pisp_be: config device node registered as /dev/video37
[    5.535231] brcmfmac_wcc: brcmf_wcc_attach: executing
[    5.543904] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[... cut ...]
[ 1530.563118] brcmfmac: brcmf_set_channel: set chanspec 0xd02a fail, reason -52
[ 1530.671096] brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
[ 1530.779109] brcmfmac: brcmf_set_channel: set chanspec 0xd034 fail, reason -52
[ 1530.786681] brcmfmac: brcmf_set_channel: set chanspec 0xd038 fail, reason -52
[ 1530.794250] brcmfmac: brcmf_set_channel: set chanspec 0xd03c fail, reason -52
[ 1530.801810] brcmfmac: brcmf_set_channel: set chanspec 0xd040 fail, reason -52
```
</details>

pelwell, 2024-03-18:
Nice. Cause:
```
[ 1417.230236] usb 3-2: new high-speed USB device number 2 using xhci-hcd
[ 1417.384491] usb 3-2: New USB device found, idVendor=05e3, idProduct=0610, bcdDevice= 6.55
[ 1417.392709] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1417.399876] usb 3-2: Product: USB2.1 Hub
[ 1417.403813] usb 3-2: Manufacturer: GenesysLogic
[ 1417.450861] hub 3-2:1.0: USB hub found
[ 1417.454902] hub 3-2:1.0: 4 ports detected
[ 1417.770238] usb 3-2.4: new high-speed USB device number 3 using xhci-hcd
[ 1417.898768] usb 3-2.4: New USB device found, idVendor=0bda, idProduct=8152, bcdDevice=20.00
[ 1417.907160] usb 3-2.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1417.914502] usb 3-2.4: Product: USB 10/100 LAN
[ 1417.918963] usb 3-2.4: Manufacturer: Realtek
[ 1417.923249] usb 3-2.4: SerialNumber: 00E04C680B3A
[ 1418.186729] r8152-cfgselector 3-2.4: reset high-speed USB device number 3 using xhci-hcd
[ 1418.374962] r8152 3-2.4:1.0: skip request firmware
[ 1418.406751] r8152 3-2.4:1.0 eth0: v1.12.13
[ 1418.420777] usbcore: registered new interface driver cdc_ether
[ 1418.425104] r8152 3-2.4:1.0 enu2u4: renamed from eth0
```
Effect:
```
[ 1418.851124] brcmfmac: brcmf_set_channel: set chanspec 0xd022 fail, reason -52
```
It was going OK until you plugged in an external HUB with a number of devices attached.

Is that hub, or any of the devices, powered?

starsareintherose, 2024-03-18:
This is a USB Hub, however, I found I can't access it via ssh then I use USB to find what happened, I will try to see if this is true.

starsareintherose, 2024-03-21:
Without nvme or ssd issue, this network issue is still

<details>

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[    0.000000] Linux version 6.6.22-2-rpi (builduser@leming) (aarch64-unknown-linux-gnu-gcc (GCC) 12.1.0, GNU ld (GNU Binutils) 2.38) #1 SMP PREEMPT Wed Mar 20 14:31:21 MDT 2024
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 5 Model B Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000002000000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x0000000002000000..0x0000000015ffffff (327680 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x0000000000000000..0x000000000007ffff (512 KiB) nomap non-reusable atf@0
[    0.000000] OF: reserved mem: 0x000000003fd165a0..0x000000003fd165e4 (0 KiB) nomap non-reusable nvram@0
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000007ffff]
[    0.000000]   node   0: [mem 0x0000000000080000-0x000000003fbfffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA: 1024 pages in unavailable ranges
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 30 pages/cpu s85416 r8192 d29272 u122880
[    0.000000] pcpu-alloc: s85416 r8192 d29272 u122880 alloc=30*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Hardware dirty bit management
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: reboot=w coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=D8:3A:DD:E8:8E:21 vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  root=/dev/nvme0n1p2 rw rootwait console=ttyAMA10,115200 console=tty1 fsck.repair=yes
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2063360
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x00000000fbfff000-0x00000000fffff000] (64MB)
[    0.000000] Memory: 7793280K/8384512K available (14464K kernel code, 2300K rwdata, 4532K rodata, 5056K init, 1049K bss, 263552K reserved, 327680K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 45530 entries in 178 pages
[    0.000000] ftrace: allocated 178 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000000] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000150] Console: colour dummy device 80x25
[    0.000154] printk: console [tty1] enabled
[... 704 lines cut ...]
[  152.894267] brcmfmac: brcmf_set_channel: set chanspec 0xd02a fail, reason -52
[  153.002276] brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
[  153.110257] brcmfmac: brcmf_set_channel: set chanspec 0xd034 fail, reason -52
[  153.117727] brcmfmac: brcmf_set_channel: set chanspec 0xd038 fail, reason -52
[  153.125201] brcmfmac: brcmf_set_channel: set chanspec 0xd03c fail, reason -52
[  153.132665] brcmfmac: brcmf_set_channel: set chanspec 0xd040 fail, reason -52
[  160.942241] brcmfmac: brcmf_set_channel: set chanspec 0xd022 fail, reason -52
[  161.054251] brcmfmac: brcmf_set_channel: set chanspec 0xd026 fail, reason -52
[... cut ...]
[  169.410284] brcmfmac: brcmf_set_channel: set chanspec 0xd02a fail, reason -52
[  169.518263] brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
[  169.626271] brcmfmac: brcmf_set_channel: set chanspec 0xd034 fail, reason -52
[  169.633714] brcmfmac: brcmf_set_channel: set chanspec 0xd038 fail, reason -52
[  169.641217] brcmfmac: brcmf_set_channel: set chanspec 0xd03c fail, reason -52
[  169.648688] brcmfmac: brcmf_set_channel: set chanspec 0xd040 fail, reason -52
```

<\details>

starsareintherose, 2024-04-06:
I now use powerful pcie bridge and samsung 980 SSD

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[    0.000000] Linux version 6.6.23-1-rpi (builduser@leming) (aarch64-unknown-linux-gnu-gcc (GCC) 12.1.0, GNU ld (GNU Binutils) 2.38) #1 SMP PREEMPT Thu Mar 28 15:05:47 MDT 2024
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 5 Model B Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000002000000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x0000000002000000..0x0000000015ffffff (327680 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x0000000000000000..0x000000000007ffff (512 KiB) nomap non-reusable atf@0
[    0.000000] OF: reserved mem: 0x000000003fd165a0..0x000000003fd165d7 (0 KiB) nomap non-reusable nvram@0
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000007ffff]
[    0.000000]   node   0: [mem 0x0000000000080000-0x000000003f7fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA: 2048 pages in unavailable ranges
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 30 pages/cpu s85416 r8192 d29272 u122880
[    0.000000] pcpu-alloc: s85416 r8192 d29272 u122880 alloc=30*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Hardware dirty bit management
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: reboot=w coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=D8:3A:DD:E8:8E:21 vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  root=/dev/nvme0n1p2 rw rootwait console=ttyAMA10,115200 console=tty1 fsck.repair=yes
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2062336
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x00000000fbfff000-0x00000000fffff000] (64MB)
[    0.000000] Memory: 7789112K/8380416K available (14528K kernel code, 2300K rwdata, 4532K rodata, 5056K init, 1049K bss, 263624K reserved, 327680K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 45540 entries in 178 pages
[    0.000000] ftrace: allocated 178 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] workqueue: name exceeds WQ_NAME_LEN. Truncating to: events_freezable_power_efficien
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000000] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000159] Console: colour dummy device 80x25
[    0.000163] printk: console [tty1] enabled
[    0.000330] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[... 563 lines cut ...]
[  132.727293] brcmfmac: brcmf_set_channel: set chanspec 0xd026 fail, reason -52
[  132.835314] brcmfmac: brcmf_set_channel: set chanspec 0xd02a fail, reason -52
[  132.943312] brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
[  133.051310] brcmfmac: brcmf_set_channel: set chanspec 0xd034 fail, reason -52
[  133.058745] brcmfmac: brcmf_set_channel: set chanspec 0xd038 fail, reason -52
[  133.066184] brcmfmac: brcmf_set_channel: set chanspec 0xd03c fail, reason -52
[... cut ...]
[  141.951326] brcmfmac: brcmf_set_channel: set chanspec 0xd026 fail, reason -52
[  142.059316] brcmfmac: brcmf_set_channel: set chanspec 0xd02a fail, reason -52
[  142.167314] brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
[  142.275325] brcmfmac: brcmf_set_channel: set chanspec 0xd034 fail, reason -52
[  142.282818] brcmfmac: brcmf_set_channel: set chanspec 0xd038 fail, reason -52
[  142.290310] brcmfmac: brcmf_set_channel: set chanspec 0xd03c fail, reason -52
[  142.297800] brcmfmac: brcmf_set_channel: set chanspec 0xd040 fail, reason -52
```

delgh1, 2024-04-15:
I can reproduce this. Freshly flashed RPI OS bookworm arm64 image, booted using microSD card, no peripherals attached except a keyboard. I'm using wpa_supplicant + systemd-networkd instead of NetworkManager.

The syptoms are: high latency spikes (~1400ms), many packets dropped (too many). Seems to only appear on linux 6.6. If I disable wifi and use a wired connection, everything returns to normal.

I reproduced this on two RPI Zero 2W, two RPI 4B, two RPI 5. (am I hoarding too many PIs...)

bobrippling, 2024-04-18:
I see these messages on a raspberry pi 4 running nixos - oddly the messages appear two days before the symptoms and I seem to need quite a bit of uptime before the issue occurs. The symptoms are the same though - the above message, high latency and packet drops.

I can pinpoint exactly when the symptoms start and there's no logs around that time, apart from a service which is suddenly seeing the symptoms. Nothing else in `dmesg` or logs relating to networking.

The machine is left running and has no hardware (or other) changes made to it.

starsareintherose, 2024-04-18:
> I see these messages on a raspberry pi 4 running nixos - oddly the messages appear two days before the symptoms and I seem to need quite a bit of uptime before the issue occurs. The symptoms are the [...]

Could I know which country wifi/internet you used? It seems that it's relevant to country?

bobrippling, 2024-04-18:
> Could I know which country wifi/internet you used? It seems that it's relevant to country?

<details>
<summary>Sure, it's Great Britain (which matches where I am). Although perhaps the below output means it's unset for phy#0?</summary>

```
$ iw reg get
global
country GB: DFS-ETSI
  (2400 - 2483 @ 40), (N/A, 20), (N/A)
  (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
  (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
  (5470 - 5730 @ 160), (N/A, 26), (0 ms), DFS
  (5725 - 5850 @ 80), (N/A, 23), (N/A), NO-OUTDOOR
  (5925 - 6425 @ 160), (N/A, 23), (N/A), NO-OUTDOOR
  (57000 - 71000 @ 2160), (N/A, 40), (N/A)

phy#0
country 99: DFS-UNSET
  (2402 - 2482 @ 40), (6, 20), (N/A)
  (2474 - 2494 @ 20), (6, 20), (N/A)
  (5140 - 5360 @ 160), (6, 20), (N/A)
  (5460 - 5860 @ 160), (6, 20), (N/A)
```
</details>

> Could you also give the dmesg log?

Of course, I've attached a few others and I've annotated with some thoughts as well (issue happens at 2024-04-18 09:30:25). Let me know if this isn't enough and you want it from boot:

<details>
<summary>dmesg</summary>

```
$ dmesg -T
[Tue Apr 16 18:51:18 2024] net_ratelimit: 1 callbacks suppressed
[Tue Apr 16 18:51:18 2024] brcmfmac: brcmf_set_channel: set chanspec 0x100e fail, reason -52
[Tue Apr 16 18:51:18 2024] brcmfmac: brcmf_set_channel: set chanspec 0xd022 fail, reason -52
[Tue Apr 16 18:51:19 2024] brcmfmac: brcmf_set_channel: set chanspec 0xd026 fail, reason -52
[Tue Apr 16 18:51:19 2024] brcmfmac: brcmf_set_channel: set chanspec 0xd02a fail, reason -52
[Tue Apr 16 18:51:19 2024] brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
[Tue Apr 16 18:51:21 2024] brcmfmac: brcmf_set_channel: set chanspec 0xd090 fail, reason -52
[Tue Apr 16 18:51:21 2024] brcmfmac: brcmf_set_channel: set chanspec 0xd095 fail, reason -52
[Tue Apr 16 18:51:21 2024] brcmfmac: brcmf_set_channel: set chanspec 0xd099 fail, reason -52
[Tue Apr 16 18:51:21 2024] brcmfmac: brcmf_set_channel: set chanspec 0xd09d fail, reason -52
[Tue Apr 16 18:51:21 2024] brcmfmac: brcmf_set_channel: set chanspec 0xd0a1 fail, reason -52
[Wed Apr 17 04:44:37 2024] systemd-journald[271177]: Under memory pressure, flushing caches.
[Wed Apr 17 04:44:38 2024] systemd-journald[271177]: Under memory pressure, flushing caches.
[Wed Apr 17 04:44:45 2024] systemd-journald[271177]: Under memory pressure, flushing caches.
[Wed Apr 17 04:44:48 2024] systemd-journald[271177]: Under memory pressure, flushing caches.
[Wed Apr 17 04:45:33 2024] systemd-journald[271177]: Under memory pressure, flushing caches.
# issue occurs here, Thu Apr 18 09:30ish
[Thu Apr 18 12:11:29 2024] Console: switching to colour frame buffer device 240x67
[Thu Apr 18 12:11:29 2024] vc4-drm gpu: [drm] fb0: vc4drmfb frame buffer device
[Thu Apr 18 12:11:49 2024] usb 1-1.3: new low-speed USB device number 4 using xhci_hcd
[Thu Apr 18 12:11:49 2024] input: HID 04f3:0103 as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.0/0003:04F3:0103.0001/input/input0
[Thu Apr 18 12:11:49 2024] hid-generic 0003:04F3:0103.0001: input,hidraw0: USB HID v1.10 Keyboard [HID 04f3:0103] on usb-0000:01:00.0-1.3/input0
[Thu Apr 18 12:11:49 2024] input: HID 04f3:0103 Consumer Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.1/0003:04F3:0103.0002/input/input1
[Thu Apr 18 12:11:49 2024] input: HID 04f3:0103 System Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.1/0003:04F3:0103.0002/input/input2
[Thu Apr 18 12:11:49 2024] hid-generic 0003:04F3:0103.0002: input,hidraw1: USB HID v1.10 Device [HID 04f3:0103] on usb-0000:01:00.0-1.3/input1
[Thu Apr 18 14:06:43 2024] net_ratelimit: 1 callbacks suppressed
[Thu Apr 18 14:06:43 2024] brcmfmac: brcmf_set_channel: set chanspec 0x100e fail, reason -52
[Thu Apr 18 14:06:43 2024] brcmfmac: brcmf_set_channel: set chanspec 0xd022 fail, reason -52
[Thu Apr 18 14:06:43 2024] brcmfmac: brcmf_set_channel: set chanspec 0xd026 fail, reason -52
[Thu Apr 18 14:06:43 2024] brcmfmac: brcmf_set_channel: set chanspec 0xd02a fail, reason -52
[Thu Apr 18 14:06:43 2024] brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
[Thu Apr 18 14:06:45 2024] brcmfmac: brcmf_set_channel: set chanspec 0xd090 fail, reason -52
[Thu Apr 18 14:06:45 2024] brcmfmac: brcmf_set_channel: set chanspec 0xd095 fail, reason -52
[Thu Apr 18 14:06:45 2024] brcmfmac: brcmf_set_channel: set chanspec 0xd099 fail, reason -52
[Thu Apr 18 14:06:45 2024] brcmfmac: brcmf_set_channel: set chanspec 0xd09d fail, reason -52
[... 197 lines cut ...]
Apr 18 15:53:38 host wpa_supplicant[558559]: wlan0: CTRL-EVENT-CONNECTED - Connection to <bssid> completed [id=0 id_str=]
Apr 18 15:53:38 host wpa_supplicant[558559]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Apr 18 15:53:38 host wpa_supplicant[558559]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=COUNTRY_IE type=COUNTRY alpha2=GB
Apr 18 15:54:04 host wpa_supplicant[558559]: wlan0: WNM: Disassociation Imminent - Disassociation Timer 0
Apr 18 15:54:04 host wpa_supplicant[558559]: wlan0: WNM: Preferred List Available
Apr 18 15:54:04 host wpa_supplicant[558559]: wlan0: Trying to associate with SSID '<ssid>'
Apr 18 15:54:05 host wpa_supplicant[558559]: wlan0: Associated with <bssid>
Apr 18 15:54:05 host wpa_supplicant[558559]: wlan0: CTRL-EVENT-CONNECTED - Connection to <bssid> completed [id=0 id_str=]
[... cut ...]
Please let me know if I can provide anything else

#### Possibly Related

- Regulatory issues?
  - https://forums.raspberrypi.com/viewtopic.php?t=367466
  - https://askubuntu.com/questions/1466275/brcmfmac-cf880211-set-channel-set-chanspec-0x1234-fail-reason-52
- Possible fix?
  - kernel
    - https://github.com/raspberrypi/firmware/issues/1289
    - https://forums.raspberrypi.com/viewtopic.php?t=265088 (UAF)
  - power management? https://github.com/raspberrypi/linux/issues/2453
  - https://github.com/raspberrypi/linux/issues/3155

starsareintherose, 2024-04-18:
```
net_ratelimit: 1 callbacks suppressed
```
This is actually network related…

bobrippling, 2024-04-18:
Yeah fair paint, interesting that I see no issues until ~36 hours later - I wonder if there's some bug in the firmware, like the kernel UAF (which would explain why some people see success in simple fixes, while others need a reboot).

waveform80, 2024-05-08:
It doesn't appear to be network-manager specific; I can reproduce the same messages with networkd in control of the interface, but it's notable in both cases wpa_supplicant is handling the interface (I haven't tried iwd yet).

One other thing I did note, which may or may not be relevant, is that my interface happily sets its own regulatory domain implicitly these days (presumably the AP's setting the right "regdom = GB" bits in its beacon), but the `set chanspec` messages only appear when the regdom is unset. To clarify, if I fire up several terminals side-by-side:

* term 1: `watch iw reg get` (show the current regdom setting)
* term 2: `sudo dmesg -w` (tail the kernel's log)
* term 3: `nmcli conn down SSID`
* term 1: As soon as the interface goes down, the regdom changes to "00" (UNSET) and shortly afterward...
* term 2: `set chanspec` messages start spamming dmesg's output
* term 3: `nmcli conn up SSID`
* term 2: The `set chanspec` messages continue for a while, as the interface authenticates, but...
* term 1: as soon as regdom changes to "GB"...
* term 2: the `set chanspec` messages stop

Is it possible that the interface is trying channels that are permitted under regdom GB, but aren't under 00, because it "remembers" that the connection it's trying had regdom GB, but the interface isn't currently configured for that? It's wild speculation on my part, but might tie in with why the messages don't appear on initial boot -- the interface wasn't regdom GB before then -- but appear later when the interface is re-configured.

waveform80, 2024-05-08:
Aha! [This thread](https://lore.kernel.org/all/99977c876429f33d8dbab18d7c3e71590585263b.camel@sipsolutions.net/T/) looks particularly relevant. Unfortunately it doesn't have a definite resolution (at the time of writing), but from a skim through I think it's describing this exact issue and the root cause.

bobrippling, 2024-05-08:
> It doesn't appear to be network-manager specific; I can reproduce the same messages with networkd in control of the interface, but it's notable in both cases wpa_supplicant is handling the interface (I haven't tried iwd yet).

To confirm - I'm using plain `wpa_supplicant` + `dhcpcd`, no nm etc.

> Aha! [This thread](https://lore.kernel.org/all/99977c876429f33d8dbab18d7c3e71590585263b.camel@sipsolutions.net/T/) looks particularly relevant. Unfortunately it doesn't have a definite resolution [...]

Nice find, referencing [this mail](https://lore.kernel.org/all/d90cc093-e3fc-4af4-9a4a-5f4bd9a7cb1f@gmx.net/) for anyone following - seems to be the basis of a fix. Am I reading it right though, the thread seems to have gone quiet as of Feb (2024)?

tennisparty, 2024-05-17:
I am having the same issue on cm4 with WiFi on an official cm4 IO board using the latest bullseye 64bit lite image. I'm just entering the CM4 world so this stumped me for a while

WiFi county set to GB

`[    7.446309] brcmfmac: F1 signature read @0x18000000=0x15264345
[    7.463889] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    7.466002] usbcore: registered new interface driver brcmfmac
[    7.749121] brcmfmac_wcc: brcmf_wcc_attach: executing
[    7.761983] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[    7.762883] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Apr 15 2021 03:03:20 version 7.45.234 (4ca95bb CY) FWID 01-996384e2
[    8.975553] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[  105.537981] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[  109.644677] brcmfmac: brcmf_set_channel: set chanspec 0xd022 fail, reason -52
[  109.756626] brcmfmac: brcmf_set_channel: set chanspec 0xd026 fail, reason -52
[  109.864470] brcmfmac: brcmf_set_channel: set chanspec 0xd02a fail, reason -52
[  109.972425] brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
[  111.700435] brcmfmac: brcmf_set_channel: set chanspec 0xd090 fail, reason -52
[  111.700845] brcmfmac: brcmf_set_channel: set chanspec 0xd095 fail, reason -52
[  111.701231] brcmfmac: brcmf_set_channel: set chanspec 0xd099 fail, reason -52
[  111.701618] brcmfmac: brcmf_set_channel: set chanspec 0xd09d fail, reason -52
[  111.702006] brcmfmac: brcmf_set_channel: set chanspec 0xd0a1 fail, reason -52
[  111.702394] brcmfmac: brcmf_set_channel: set chanspec 0xd0a5 fail, reason -52
[  113.175791] brcmfmac: brcmf_set_channel: set chanspec 0xd022 fail, reason -52
[  113.283801] brcmfmac: brcmf_set_channel: set chanspec 0xd026 fail, reason -52
[  113.387945] brcmfmac: brcmf_set_channel: set chanspec 0xd02a fail, reason -52
[  113.491819] brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
[  115.155892] brcmfmac: brcmf_set_channel: set chanspec 0xd090 fail, reason -52
[  115.156181] brcmfmac: brcmf_set_channel: set chanspec 0xd095 fail, reason -52
[  115.156466] brcmfmac: brcmf_set_channel: set chanspec 0xd099 fail, reason -52
[  115.156751] brcmfmac: brcmf_set_channel: set chanspec 0xd09d fail, reason -52
[  115.157035] brcmfmac: brcmf_set_channel: set chanspec 0xd0a1 fail, reason -52
[  115.157320] brcmfmac: brcmf_set_channel: set chanspec 0xd0a5 fail, reason -52
[  146.689762] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[  150.659839] brcmfmac: brcmf_set_channel: set chanspec 0xd022 fail, reason -52
[  150.767804] brcmfmac: brcmf_set_channel: set chanspec 0xd026 fail, reason -52
[  150.872473] brcmfmac: brcmf_set_channel: set chanspec 0xd02a fail, reason -52
[  150.980417] brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
[  152.708504] brcmfmac: brcmf_set_channel: set chanspec 0xd090 fail, reason -52
[  152.708941] brcmfmac: brcmf_set_channel: set chanspec 0xd095 fail, reason -52
[  152.709367] brcmfmac: brcmf_set_channel: set chanspec 0xd099 fail, reason -52
[  152.709791] brcmfmac: brcmf_set_channel: set chanspec 0xd09d fail, reason -52
[  152.710213] brcmfmac: brcmf_set_channel: set chanspec 0xd0a1 fail, reason -52
[  152.710638] brcmfmac: brcmf_set_channel: set chanspec 0xd0a5 fail, reason -52`

If anyone knows a fix, an older kernel etc let me know.

spanzetta, 2024-05-22:
Same issue here on RPI3A+ with HASSIO..
Never seen it in the past...
It started with the latest OS update

tennisparty, 2024-05-22:
I checked my WiFi country and my tp link router was constantly setting the country to DE if I like with is reg get. Even if I set it back it changed back on a restart. I got a new router and it seems to be fixed.

yuklia, 2024-05-25:
device : RPi5

issue appeared after setting up `netplan` config

**steps to reproduce:**
1. Add custom config to /etc/netplan/01-iphone-ap.yaml
<details>
    <summary>Details</summary>

    network:
      version: 2
      renderer: networkd
      wifis:
        wlan0:
          dhcp4: yes
          dhcp6: yes
          access-points:
            "YourHotspotSSID":
              password: "YourHotspotPassword"
</details>

2. run `netplan generate`
3. run `netplan apply`

Write, 2024-05-28:
Same issue here

Jepson2k, 2024-05-29:
Also having this issue on a RPI 5 running Raspberry Pi OS (Debian 12.5 Bookworm). Used to run a webserver. Started occurring sometime in the last month or so. Also have a TP-Link router, but I wasn't able to get it to broadcast any beacons containing country code (I might not be doing it correctly though). Anyways I noticed something that seemed odd to me: 
1. RPI not connected. In terminal 1: dmesg showing many brcmf errors
2. Terminal 2: Running `iw event` - shows periodic scans. Whenever there is a scan more brcmf errors show up in dmesg
3. Terminal 1: Run `wpa_cli reconfigure` to reconnect to the saved WIFI network - get a FAILED error status. (This has worked in the past to reconnect to wifi, before this error started occuring.)
4. Terminal 1: Run `nmcli d connect wlan0` - dmesg shows a scan that now includes my wifi network, followed by a connection message, followed by a message noting regulatory domain change, followed by another scan including my wifi network.
5. Device successfully connects to the network. Subsequent scans don't include my wifi network, but also don't throw any errors.

<details>
<summary>iw event output</summary>

```
wlan0 (phy #0): scan finished: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 5170 5180 5190 5200 5210 5220 5230 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825, ""
wlan0 (phy #0): scan started
wlan0 (phy #0): scan finished: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 5170 5180 5190 5200 5210 5220 5230 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825, ""
wlan0 (phy #0): scan started
wlan0 (phy #0): scan finished: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 5170 5180 5190 5200 5210 5220 5230 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825, ""
wlan0 (phy #0): scan started
wlan0 (phy #0): scan finished: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 5170 5180 5190 5200 5210 5220 5230 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825, ""
wlan0 (phy #0): scan started
wlan0 (phy #0): scan finished: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 5170 5180 5190 5200 5210 5220 5230 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825, ""
wlan0 (phy #0): scan started
wlan0 (phy #0): scan finished: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 5170 5180 5190 5200 5210 5220 5230 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825, ""
wlan0 (phy #0): scan started
wlan0 (phy #0): scan finished: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 5170 5180 5190 5200 5210 5220 5230 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825, "WIFI Network" ""
wlan0 (phy #0): connected to 92:a7:b9:89:f5:07
phy #0: regulatory domain change: set to US by a country IE request on phy0
wlan0 (phy #0): scan started
wlan0 (phy #0): scan aborted: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 5170 5180 5190 5200 5210 5220 5230 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825, "WIFI Network" ""
wlan0 (phy #0): scan started
wlan0 (phy #0): scan finished: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 5170 5180 5190 5200 5210 5220 5230 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825, ""
wlan0 (phy #0): scan started
wlan0 (phy #0): scan finished: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 5170 5180 5190 5200 5210 5220 5230 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825, ""
wlan0 (phy #0): scan started
wlan0 (phy #0): scan finished: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 5170 5180 5190 5200 5210 5220 5230 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825, ""
```
</details>

Still not sure of the actual issue, but thought I would share.

stuckinthe, 2024-05-30:
I am seeing the same issue. I have tried both NetworkManager and wpa_supplicant with systemd and both of them eventually drop my wifi and I have to reboot. Sometimes its everyday or every 3 days. I have set my country code but I still get disconnects. I am running Arch Linux on a Rpi5 and I have a TP-Link router. I am like 6 feet from the router so it is close enough with no walls in between. I wish I could figure this out.

bobrippling, 2024-05-30:
There may be some value in investigating if firmware changes such as those mentioned in https://github.com/NixOS/nixpkgs/issues/82462 work, I won't have access to my machine for a while but that's my next step.

amartinr, 2024-05-31:
> Also having this issue on a RPI 5 running Raspberry Pi OS (Debian 12.5 Bookworm). Used to run a webserver. Started occurring sometime in the last month or so. Also have a TP-Link router, but I wasn't able to get it to broadcast any beacons containing country code (I might not be doing it correctly though). Anyways I noticed something that seemed odd to me:
> 
>     1. RPI not connected. In terminal 1: dmesg showing many brcmf errors
> 
>     2. Terminal 2: Running `iw event` - shows periodic scans. Whenever there is a scan more brcmf errors show up in dmesg
> 
>     3. Terminal 1: Run `wpa_cli reconfigure` to reconnect to the saved WIFI network - get a FAILED error status. (This has worked in the past to reconnect to wifi, before this error started occuring.)
> 
>     4. Terminal 1: Run `nmcli d connect wlan0` - dmesg shows a scan that now includes my wifi network, followed by a connection message, followed by a message noting regulatory domain change, followed by another scan including my wifi network.
> 
>     5. Device successfully connects to the network. Subsequent scans don't include my wifi network, but also don't throw any errors.
> 
> 
> iw event output
> 
> Still not sure of the actual issue, but thought I would share.

I'm experiencing the same issue on my Raspberry Pi 4B running Raspbian Bookworm with kernel 6.6.28+rpt-rpi-v8.

My Pi is connected to my WiFi home router using the 5.24 GHz band. Coincidentally I have several 2.4-GHz devices in my home network which use services running on the Pi. Not sure if that could be related to this particular issue, but never experienced it before.

I've enabled some debug options for the _brcmfmac_ module, but honestly, I'm rather clueless about what to look for or which flags I should enable. Maybe someone with a better understanding of the module source code can point out which bits are worth enabling to better diagnose this behaviour (options taken from [FMAC Debugging](https://community.infineon.com/t5/Knowledge-Base-Articles/FMAC-Debugging/ta-p/246042))

```
options brcmfmac debug=0xCC3C
```

It also might be worth noting that I tested both with WiFi power management enabled and disabled. I'd say that with power management enabled the percentage of lost packets is lower. It's very common for users to disable power management because it's common to find that recommendation when searching for solutions to WiFi connectivity issues.

The following are kernel logs from the `brcmfmac` kernel module. You can see a 4-second time gap in the logs which happened at the same time I stopped receiving ICMP echo replies from my Pi. For these tests I'm using  `ping -4 -n`, as I'm running an IPv4/IPv6 dual stack and don't want other factors to interfere. Power management is also enabled for my Pi's WiFi device.

```
May 31 16:20:12 totara kernel: brcmfmac: brcmf_rx_frame Enter: mmc1:0001:1: rxp=00000000e5ba4b84
May 31 16:20:12 totara kernel: brcmfmac: brcmf_netif_rx rx proto=0x800
May 31 16:20:12 totara kernel: brcmfmac: brcmf_netdev_start_xmit Enter, bsscfgidx=0
May 31 16:20:12 totara kernel: brcmfmac: brcmf_sdio_readframes processed 1 frames
May 31 16:20:12 totara kernel: brcmfmac: brcmf_sdio_bus_watchdog Enter
May 31 16:20:12 totara kernel: brcmfmac: brcmf_sdio_bus_watchdog Enter
May 31 16:20:12 totara kernel: brcmfmac: brcmf_rx_frame Enter: mmc1:0001:1: rxp=000000007f52f785
May 31 16:20:12 totara kernel: brcmfmac: brcmf_netif_rx rx proto=0x86DD
May 31 16:20:12 totara kernel: brcmfmac: brcmf_netdev_start_xmit Enter, bsscfgidx=0
May 31 16:20:12 totara kernel: brcmfmac: brcmf_sdio_readframes processed 1 frames
May 31 16:20:12 totara kernel: brcmfmac: brcmf_sdio_bus_watchdog Enter
May 31 16:20:12 totara kernel: brcmfmac: brcmf_sdio_bus_watchdog Enter
May 31 16:20:16 totara kernel: brcmfmac: brcmf_rx_frame Enter: mmc1:0001:1: rxp=00000000e5ba4b84
May 31 16:20:16 totara kernel: brcmfmac: brcmf_netif_rx rx proto=0x800
May 31 16:20:16 totara kernel: brcmfmac: brcmf_netdev_start_xmit Enter, bsscfgidx=0
May 31 16:20:16 totara kernel: brcmfmac: brcmf_sdio_readframes processed 1 frames
May 31 16:20:16 totara kernel: brcmfmac: brcmf_sdio_bus_watchdog Enter
May 31 16:20:16 totara kernel: brcmfmac: brcmf_sdio_bus_watchdog Enter
```

I'm defining the CRDA regulatory domain (ES) both for `wpa_supplicant` and as a kernel module parameter at boot with the following command line options:
```
console=serial0,115200 console=tty1 root=PARTUUID=3b0260ae-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=ES video=HDMI-A-1:1280x720@60
```

nbuchwitz, 2024-06-04:
Observed the same issue with a CM4 based device and 6.6 kernel, but they went away after the installation of the `wireless-regdb` package. Could you please check if this package is present on your setup?

stuckinthe, 2024-06-05:
I had wireless-regdb on my system but I didn't have the country code uncommented in the conf file. I have done this and I am waiting to see if I have an issue.

stuckinthe, 2024-06-05:
I am still seeing the issue. I was trying to clone chromium and I lost my internet in the middle. I couldn't just reboot as I had to kill the power. I have Kodi running and set to run in systemd so since I lose ssh and I can't drop to a shell I can't investigate the issue. I may just start kodi manually and break out the keyboard when I lose internet.

fortysix2ahead, 2024-06-06:
Is it possible that this has something to do with SSH? I run a Pi 5, which was freshly populated with lots of data via SFTP. I experienced several times that the Pi lost its network connection, data transfer was interrupted and SSH sessions were killed. Vice versa: after populating the Pi with data and after setting up backup based on Restic (again via SFTP), backup jobs failed from time to time. I have now switched to Restic with a the REST server as backup target, meaning that data transfers now run via HTTP instead of SFTP and et voilà - everything seems to work smoothly ...

I wouldn't go that far to blame SSH/SFTP, but at least my observations point into that direction.

radkokeves, 2024-06-13:
i have the same issue with rpi4 running hassio

stuckinthe, 2024-06-14:
I removed NetworkManager and installed systemd-networkd/dhcpcd and then set this up to use wpa_supplicant. I went 4 days without an issue. I had to reboot due to a kernel upgrade. I will keep testing.

DEPSTRCZ, 2024-06-14:
I am also experiencing this issue. I have an RPI 5 with an M.2 HAT from Pimoroni. This problem occurs only on Ubuntu 24.04 LTS with NetworkManager. When I boot into Raspberry Pi OS with the M.2 HAT still connected, it works without any issues.

stuckinthe, 2024-06-14:
It lasted a half day and then I lost my wifi. I can't do a reboot as I have to unplug the RpI5 and then bootup in order for it to work. I had ssh session open but I wasn't doing anything as the system was idle. I'll have to start kodi manually and then use a keyboard to see what the issue is but I don't know how long it will be.

spanzetta, 2024-06-16:
Still no fix?
It's over a month that many Raspberry users have this problem with HAOS.. that cause a lot of instability

Nirvana55, 2024-06-19:
having same issue.. everything was working smoothly even after updating the system. This issue occurred after I reboot my system. 
Specs: Rasp4 with arch arm 

now cannot connect to internet. 
Any alternative anyone ?

radkokeves, 2024-06-19:
as a workaround I downgraded hassio to last working version 12.2 using: ha os update --version 12.2

bobrippling, 2024-06-19:
> as a workaround I downgraded hassio to last working version 12.2 using: `ha os update --version 12.2`

How recent was this workaround applied? I sometimes see the driver remain stable for a few weeks before the issue occurs.

radkokeves, 2024-06-19:
It started to fail for me after upgrade to 12.3
Tried 12.4 as well, doesn't work neither

Last working version is 12.2

dzwiedziu-nkg, 2024-06-20:
I had the same problem with Raspberry Pi OS and Ubuntu Server 24.04. After setting regulatory-domain to my country, it works ok. My country is Poland (PL).

/etc/netplan/98_wifi.yaml
`
network:
  version: 2
  renderer: networkd
  wifis:
    wlan0:
      regulatory-domain: "PL"
      dhcp4: yes
      optional: true
      access-points:
        "my-wifi":
          password: "my-wifi-password"
`

stuckinthe, 2024-06-21:
I upgraded to kernel 6.6.34-1 and it made it even worse.

spanzetta, 2024-06-23:
> as a workaround I downgraded hassio to last working version 12.2 using: ha os update --version 12.2

I confirm 100%.. downgraded to 12.2 and system is now stable again without that recurring error (that was also causing a costant SD activity)

kernelport, 2024-06-25:
I had the same error on Rasp4 64 bit:
```
Linux bookworm 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux
brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
```
In my case, the message disappears after unloading and reloading the brcmfmac_wcc kernel module or activate the connection (even if it is unused)

<details>

<summary>This is my solution:</summary>

determine the connection NAME for DEVICE=wlan0 with:
`nmcli connection show --active`

stop the connection with name=NAME
`nmcli connection down "name"`

unload kernel modules regarding to "wlan0":
`modprobe -r brcmfmac_wcc brcmfmac`
load the kernel modules again:
`modprobe brcmfmac_wcc`

</details>

Udeste, 2024-06-25:
I am also facing the same problem.
Raspberry Pi 5 running Raspberry Pi OS (Debian 12.5 Bookworm) with WiFi country set to IT.

In my case I can successfully connect to 5.2GHz wifi only when I manually set the channel on the router to one between 36 (5180 Mhz) and 52 (5260 Mhz). When I set it to a channel > 52 or leave it at auto on the router, I get the brcmfmac: brcmf_set_channel: set chanspec 0xd09d fail, reason -52 error and no connection.
 
I'm using 80MHz as channel width.

Dggrr, 2024-07-08:
Same problem here with on a Pi 5 with:
Linux jjrasp01 6.6.31+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

A temporary workaround for me is to manually change the kernel links in "/" back to 6.6.28:

$ ls -l / |grep 6.6.28
lrwxrwxrwx   1 root root    33 Jul  7 23:15 initrd.img -> boot/initrd.img-6.6.28+rpt-rpi-v8
lrwxrwxrwx   1 root root    35 Jul  7 23:16 initrd.img.old -> boot/initrd.img-6.6.28+rpt-rpi-2712
lrwxrwxrwx   1 root root    30 Jul  7 23:16 vmlinuz -> boot/vmlinuz-6.6.28+rpt-rpi-v8
lrwxrwxrwx   1 root root    32 Jul  7 23:16 vmlinuz.old -> boot/vmlinuz-6.6.28+rpt-rpi-2712

Seems to be a problem with 6.6.31 to me.

Jepson2k, 2024-07-10:
As an update to my previous post, I happened to switch routers (from tp-link to ubiquiti) and to my surprise the problem completely disappeared. So there's a chance it **could** be router dependent.

Jepson2k, 2024-07-10:
I switched routers a little while ago before that was suggested so I did not get the chance to try it.

SamsterJam, 2024-07-14:
Having same issue here

bgeza, 2024-07-15:
> Did going back to 6.6.28 fix the problem with the tp-link router?
> 
> Seems everyone on here that has the problem has a tp-link router as this is the router I have as well. Not sure if it is the tp-link hw or sw as I'm not using openwrt.

Asus RT-AX58U/factory firmware has the same issue

willemprins44, 2024-07-15:
Downgraded to 12.2 and issue disappear,  thanks spanzetta

> same here

mbatesco, 2024-07-16:
I tried using

`[connection]
wifi.powersave = 2`

And on restarts, I validated the power save was still on and I was still seeing the same error messages above.

I finally create a startup service: disable-wlan0-powersave.service   to force powersave off

`[Unit]
Description=Disable wlan0 power save

[Service]
Type=oneshot
ExecStart=/sbin/iw dev wlan0 set power_save off

[Install]
WantedBy=multi-user.target`

Now after rebooting, power save is now off and I've not seen messages and everything seems fine at this point.

stuckinthe, 2024-07-18:
In a few more hours, I will have gone 14 days without losing my wifi connection. I'm using kernel 6.6.36-1

DEPSTRCZ, 2024-07-22:
Still no fix / update?

stuckinthe, 2024-07-23:
> Still no fix / update?

Which kernel are you using? I've gone almost 3 weeks without a drop.

WhySoBad, 2024-07-23:
The issue is still present for me on the Pi 4 Model B Rev 1.1 on kernel 6.6.40-v8+ on the latest raspbian version based off debian 12 (bookworm)

I have the same logs as seemingly everyone here:
```
Jul 23 09:38:11 raspberrypi systemd[1]: NetworkManager-dispatcher.service: Deactivated successfully.
Jul 23 09:38:08 raspberrypi kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd0a5 fail, reason -52
Jul 23 09:38:08 raspberrypi kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd0a1 fail, reason -52
Jul 23 09:38:08 raspberrypi kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd09d fail, reason -52
Jul 23 09:38:08 raspberrypi kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd099 fail, reason -52
Jul 23 09:38:08 raspberrypi kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd095 fail, reason -52
Jul 23 09:38:08 raspberrypi kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd090 fail, reason -52
Jul 23 09:38:06 raspberrypi kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
Jul 23 09:38:06 raspberrypi kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd02a fail, reason -52
Jul 23 09:38:06 raspberrypi kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd026 fail, reason -52
Jul 23 09:38:06 raspberrypi kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd022 fail, reason -52
```
but in addition is sometimes also get this error:
```
Jul 23 09:37:54 raspberrypi kernel: ieee80211 phy0: brcmf_set_wsec: failed to change PSK in firmware (len=0)
Jul 23 09:37:54 raspberrypi kernel: ieee80211 phy0: brcmf_vif_set_mgmt_ie: vndr ie set error : -52
Jul 23 09:37:54 raspberrypi kernel: ieee80211 phy0: brcmf_vif_set_mgmt_ie: vndr ie set error : -52
```

DEPSTRCZ, 2024-07-27:
> > Still no fix / update?
> 
> Which kernel are you using? I've gone almost 3 weeks without a drop.

Running 6.6.31+rpt-rpi-2712

bbk-al, 2024-07-28:
Downgrading wpa_supplicant from 2:2.11-2 to 2:2.10-8 worked around the issue for me.

6.6.42-1-rpi on Raspberry Pi 4 Model B Rev 1.5

i3Craig, 2024-07-28:
This isn't related to RPI, but I am adding this in case it helps anyone else.
Downgrading **wpa_supplicant** to version **2:2.10-8** as stated above also fixes Wi-Fi on the PinePhone Pro (on ArchLinux ARM).

starsareintherose, 2024-07-29:
If this is related to wpa_supplicant, could someone else create a issue for archlinux arm repo and wps_supplicant repo?

ThecaTTony, 2024-07-31:
This (https://iwd.wiki.kernel.org/offloading) work for me (rpi3 b+ / archlinuxarm / wpa_supplicant 2.11-2)

`sudo modprobe -r brcmfmac`
`sudo modprobe brcmfmac feature_disable=0x82000`
`sudo systemctl restart wpa_supplicant@wlan0.service`

If works, make a file to load the module with that option at **/etc/modprobe.d/brcmfmac.conf** with:

`options brcmfmac feature_disable=0x82000`

tsjk, 2024-07-31:
> This (https://iwd.wiki.kernel.org/offloading) work for me (rpi3 b+ / archlinuxarm / wpa_supplicant 2.11-2)
>
> `sudo modprobe -r brcmfmac` `sudo modprobe brcmfmac feature_disable=0x82000` `sudo [...]

This work on a RPI4B with ArchLinuxARM and wpa_supplicant 2.11-2 as well.

[58 more replies]