raspberrypi/linux #6055: RPI5: usb x-x: cmd cmplt err -71 Error with different USB Disks

RolandFelnhofer, 2024-03-20:
### Describe the bug

After as short time (20-60 minutes) USB-disks (a Seagate HDD and a JMicron SSD) run into errors and disconnect.
First entry in journalctl is:
`Mar 20 15:11:18 server kernel: usb 4-1: cmd cmplt err -71`
more detail later in the Log-section

### Steps to reproduce the behaviour

If I/O is relevant I don't know. So I just ran the following `fio` command in a loop:
`fio --name=random-write --ioengine=posixaio --rw=randwrite --bs=4k --size=1g --numjobs=10 --runtime=60 --time_based --end_fsync=1`
After, at most 60 minutes the error occurs. 

### Device (s)

Raspberry Pi 5

### System

```
cat /etc/rpi-issue
Raspberry Pi reference 2023-12-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 2acf7afcba7d11500313a7b93bb55a2aae20b2d6, stage2
```
```
vcgencmd version
2024/02/16 15:28:41
Copyright (c) 2012 Broadcom
version 4c845bd3 (release) (embedded)
```
```
uname -a
Linux server 6.6.22-v8-16k+ #1745 SMP PREEMPT Tue Mar 19 17:57:20 GMT 2024 aarch64 GNU/Linux
```

### Logs

```
journalctl
...
Mar 20 16:10:51 server kernel: usb 4-1: cmd cmplt err -71
Mar 20 16:10:56 server kernel: scsi host0: uas_pre_reset: timed out
Mar 20 16:10:56 server kernel: sd 0:0:0:0: [sda] tag#0 uas_zap_pending 0 uas-tag 1 inflight: CMD
Mar 20 16:10:56 server kernel: sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x28 28 00 00 00 08 00 00 00 01 00
Mar 20 16:10:56 server kernel: sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x01 driverbyte=DRIVER_OK cmd_age=5s
Mar 20 16:10:56 server kernel: sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x28 28 00 00 00 08 00 00 00 01 00
Mar 20 16:10:56 server kernel: I/O error, dev sda, sector 2048 op 0x0:(READ) flags 0x1000 phys_seg 1 prio class 2
Mar 20 16:10:56 server kernel: XFS (sda1): SB validate failed with error -5.
Mar 20 16:10:56 server kernel: device offline error, dev sda, sector 0 op 0x1:(WRITE) flags 0x800 phys_seg 0 prio class 2
Mar 20 16:10:56 server kernel: sd 0:0:0:0: [sda] Synchronizing SCSI cache
Mar 20 16:10:57 server kernel: sd 0:0:0:0: [sda] Synchronize Cache(10) failed: Result: hostbyte=0x07 driverbyte=DRIVER_OK
Mar 20 16:10:57 server kernel: usb 4-1: Disable of device-initiated U1 failed.
Mar 20 16:10:57 server kernel: usb 4-1: Disable of device-initiated U2 failed.
Mar 20 16:10:57 server kernel: usb 4-1: Enable of device-initiated U1 failed.
Mar 20 16:10:57 server kernel: usb 4-1: Enable of device-initiated U2 failed.
Mar 20 16:10:57 server kernel: usb 4-1: reset SuperSpeed USB device number 4 using xhci-hcd
Mar 20 16:10:57 server kernel: scsi host0: uas
Mar 20 16:10:57 server kernel: scsi 0:0:0:0: Direct-Access     Intenso  Portable SSD     0506 PQ: 0 ANSI: 6
Mar 20 16:10:57 server kernel: sd 0:0:0:0: Attached scsi generic sg0 type 0
Mar 20 16:10:57 server kernel: sd 0:0:0:0: [sda] 2000409264 512-byte logical blocks: (1.02 TB/954 GiB)
Mar 20 16:10:57 server kernel: sd 0:0:0:0: [sda] 4096-byte physical blocks
Mar 20 16:10:57 server kernel: sd 0:0:0:0: [sda] Write Protect is off
Mar 20 16:10:57 server kernel: sd 0:0:0:0: [sda] Mode Sense: 53 00 00 08
Mar 20 16:10:57 server kernel: sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
Mar 20 16:10:57 server kernel: sd 0:0:0:0: [sda] Preferred minimum I/O size 4096 bytes
Mar 20 16:10:57 server kernel: sd 0:0:0:0: [sda] Optimal transfer size 33553920 bytes not a multiple of preferred minimum block size (4096 bytes)
Mar 20 16:10:57 server kernel:  sda: sda1
Mar 20 16:10:57 server kernel: sd 0:0:0:0: [sda] Attached SCSI disk
...
```
[... 768 lines cut ...]
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000e
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat         512 micro seconds
Device Status:     0x000c
  (Bus Powered)
  U1 Enabled
  U2 Enabled
...

```

### Additional context

I'm using the official Pi5 27W PSU (Barcode: 5 056561 803416)
```
vcgencmd get_config usb_max_current_enable
usb_max_current_enable=1
```

ThomasKaiser, 2024-03-25:
> It seems that adding `usb-storage.quirks=152d:0579:u` to `/boot/firmware/cmdline.txt` prevent this error

Can't 'help' with your `0bc2:ab21` Seagate HDD and after blacklisting UAS when the real reason was either data integrity or power issues then error messages just will be different but symptoms pretty much the same.

matthias480, 2024-04-16:
I have similar problems since switching from the Raspberry Pi 4 to the Raspberry Pi 5.
Similar things were also discussed in #5753. Not sure if it's the same thing, but there still seems to be an issue.

I use two stacked 4-Port USB 3.1 Hubs with PPPS (Per-Port Power Switching) functionality.
The problem happens with every used drive (SSD/HDD).

Sometimes errors occurs when I want to mount the partition after turning the USB port on:
```
mount: /mnt/ssd4: /dev/sdf1 is not a valid block device.
       dmesg(1) may have more information after failed mount system call.
```

This probem already occured on my Raspberry Pi 4. I thought, maybe it happens if the drive isn't ready when the mount happens. A second mount was successful. Still so is now.

**Raspberry Pi 4:**
`Linux raspberrypi 5.10.63-v7l+ #1496 SMP Wed Dec 1 15:58:56 GMT 2021 armv7l`
```
[Fri Mar 8 19:37:35 2024] usb 2-2.3: cmd cmplt err -71
[Fri Mar 8 19:37:53 2024] usb 2-2.3: USB disconnect, device number 17
[Fri Mar 8 19:37:53 2024] sd 6:0:0:0: [sdg] tag#16 uas_zap_pending 0 uas-tag 1 inflight: CMD
[Fri Mar 8 19:37:53 2024] sd 6:0:0:0: [sdg] tag#16 CDB: opcode=0x88 88 00 00 00 00 01 d1 c0 b7 80 00 00 00 08 00 00
[Fri Mar 8 19:37:53 2024] sd 6:0:0:0: [sdg] tag#16 UNKNOWN(0x2003) Result: hostbyte=0x01 driverbyte=0x00 cmd_age=18s
[Fri Mar 8 19:37:53 2024] sd 6:0:0:0: [sdg] tag#16 CDB: opcode=0x88 88 00 00 00 00 01 d1 c0 b7 80 00 00 00 08 00 00
[Fri Mar 8 19:37:53 2024] blk_update_request: I/O error, dev sdg, sector 7814035328 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
[Fri Mar 8 19:37:53 2024] blk_update_request: I/O error, dev sdg, sector 7814035328 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[Fri Mar 8 19:37:53 2024] Buffer I/O error on dev sdg1, logical block 976754160, async page read
[Fri Mar 8 19:37:53 2024] sd 6:0:0:0: [sdg] Synchronizing SCSI cache
[Fri Mar 8 19:37:54 2024] sd 6:0:0:0: [sdg] Synchronize Cache(10) failed: Result: hostbyte=0x07 driverbyte=0x00
[Fri Mar 8 19:37:54 2024] usb 2-2.3: new SuperSpeed Gen 1 USB device number 23 using xhci_hcd
[Fri Mar 8 19:37:54 2024] usb 2-2.3: New USB device found, idVendor=174c, idProduct=55aa, bcdDevice= 1.00
[Fri Mar 8 19:37:54 2024] usb 2-2.3: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[Fri Mar 8 19:37:54 2024] usb 2-2.3: Product: ASM105x
[Fri Mar 8 19:37:54 2024] usb 2-2.3: Manufacturer: ASMT
[Fri Mar 8 19:37:54 2024] usb 2-2.3: SerialNumber: 12345678AD9C
[Fri Mar 8 19:37:54 2024] scsi host6: uas
[Fri Mar 8 19:37:54 2024] scsi 6:0:0:0: Direct-Access     Samsung  SSD 860 EVO 4TB  0    PQ: 0 ANSI: 6
[Fri Mar 8 19:37:54 2024] sd 6:0:0:0: Attached scsi generic sg9 type 0
[Fri Mar 8 19:37:54 2024] sd 6:0:0:0: [sdg] 7814037168 512-byte logical blocks: (4.00 TB/3.64 TiB)
[Fri Mar 8 19:37:54 2024] sd 6:0:0:0: [sdg] Write Protect is off
[Fri Mar 8 19:37:54 2024] sd 6:0:0:0: [sdg] Mode Sense: 43 00 00 00
[Fri Mar 8 19:37:54 2024] sd 6:0:0:0: [sdg] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[Fri Mar 8 19:37:54 2024] sd 6:0:0:0: [sdg] Optimal transfer size 33553920 bytes
[Fri Mar 8 19:37:54 2024]  sdg: sdg1
[Fri Mar 8 19:37:54 2024] sd 6:0:0:0: [sdg] Attached SCSI disk
```
 
**Raspberry Pi 5:**
`Linux raspberrypi 6.6.20+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64`
```
[Mon Apr 15 16:55:50 2024] usb 2-1.3: cmd cmplt err -71
[Mon Apr 15 16:56:21 2024] sd 4:0:0:0: [sdf] tag#1 uas_eh_abort_handler 0 uas-tag 1 inflight: CMD IN
[Mon Apr 15 16:56:21 2024] sd 4:0:0:0: [sdf] tag#1 CDB: opcode=0x88 88 00 00 00 00 03 a3 81 29 ff 00 00 00 20 00 00
[Mon Apr 15 16:56:21 2024] scsi host4: uas_eh_device_reset_handler start
[Mon Apr 15 16:56:22 2024] scsi host4: uas_eh_device_reset_handler FAILED err -19
[Mon Apr 15 16:56:22 2024] sd 4:0:0:0: Device offlined - not ready after error recovery
[Mon Apr 15 16:56:22 2024] sd 4:0:0:0: [sdf] tag#1 UNKNOWN(0x2003) Result: hostbyte=0x03 driverbyte=DRIVER_OK cmd_age=31s
[Mon Apr 15 16:56:22 2024] sd 4:0:0:0: [sdf] tag#1 CDB: opcode=0x88 88 00 00 00 00 03 a3 81 29 ff 00 00 00 20 00 00
[Mon Apr 15 16:56:22 2024] I/O error, dev sdf, sector 15628052991 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 2
[Mon Apr 15 16:56:22 2024] sd 4:0:0:0: rejecting I/O to offline device
[Mon Apr 15 16:56:22 2024] I/O error, dev sdf, sector 15628052991 op 0x0:(READ) flags 0x0 phys_seg 2 prio class 2
[Mon Apr 15 16:56:22 2024] buffer_io_error: 55 callbacks suppressed
[Mon Apr 15 16:56:22 2024] Buffer I/O error on dev sdf1, logical block 976749216, async page read
[Mon Apr 15 16:56:22 2024] Buffer I/O error on dev sdf1, logical block 976749217, async page read
[Mon Apr 15 16:56:22 2024] /dev/sdf1: Can't open blockdev
[Mon Apr 15 16:56:22 2024] usb 2-1.3: USB disconnect, device number 27
[Mon Apr 15 16:56:22 2024] sd 4:0:0:0: [sdf] Synchronizing SCSI cache
[Mon Apr 15 16:56:22 2024] sd 4:0:0:0: [sdf] Synchronize Cache(10) failed: Result: hostbyte=0x07 driverbyte=DRIVER_OK
[Mon Apr 15 16:56:47 2024] usb 2-1.3: new SuperSpeed USB device number 35 using xhci-hcd
[Mon Apr 15 16:56:47 2024] usb 2-1.3: New USB device found, idVendor=174c, idProduct=55aa, bcdDevice= 1.00
[Mon Apr 15 16:56:47 2024] usb 2-1.3: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[... 200 lines cut ...]
Apr 15 21:56:50 raspberrypi kernel: usb 2-1.2: reset SuperSpeed USB device number 36 using xhci-hcd
Apr 15 21:56:50 raspberrypi kernel: scsi host2: uas_eh_device_reset_handler success
[... cut ...]
Apr 15 21:57:22 raspberrypi kernel: sd 2:0:0:0: [sdb] tag#3 uas_eh_abort_handler 0 uas-tag 10 inflight: CMD
Apr 15 21:57:22 raspberrypi kernel: sd 2:0:0:0: [sdb] tag#3 CDB: opcode=0x28 28 00 04 b5 f8 a8 00 04 00 00
Apr 15 21:57:22 raspberrypi kernel: sd 2:0:0:0: [sdb] tag#2 uas_eh_abort_handler 0 uas-tag 9 inflight: CMD
Apr 15 21:57:22 raspberrypi kernel: sd 2:0:0:0: [sdb] tag#2 CDB: opcode=0x28 28 00 04 b5 f4 a8 00 04 00 00
...
Apr 15 22:29:09 raspberrypi kernel: usb 2-1.2: stat urb: status -71
```

float32top, 2024-05-08:
I'm getting the same issue here

`[  149.147364] sd 0:0:0:0: [sda] tag#28 uas_eh_abort_handler 0 uas-tag 12 inflight: CMD OUT
[  149.147372] sd 0:0:0:0: [sda] tag#28 CDB: opcode=0x2a 2a 00 3a 0e f9 08 00 00 08 00
[  154.239493] xhci-hcd xhci-hcd.0: xHCI host not responding to stop endpoint command
[  154.259666] xhci-hcd xhci-hcd.0: xHCI host controller not responding, assume dead
[  154.259707] xhci-hcd xhci-hcd.0: HC died; cleaning up
[  154.259792] usb 2-1: cmd cmplt err -108`

`Bus 004 Device 003: ID 0bda:9210 Realtek Semiconductor Corp. RTL9210 M.2 NVME Adapter`

Orico M.2 nvme SSD - usb 3 adapter
Tried rpiupdate, rebooted, same issue,
Tried on a second rpi5 , same issue.

Rpi5 8gb

j3ich0, 2024-05-24:
i've posted at https://forums.raspberrypi.com/viewtopic.php?p=2224397#p2224397 about this issue.

Pi5, 27w official Pi power supply, Sabrent USB to SATA adapter, 5TB Seagate HDD
HDD is formatted to NTFS (don't judge) and shared out using Samba. Pi5 is essentially a NAS box. I have also made the Pi boot from USB stick. The USB HDD is connected to the 2nd blue USB3 port, and the USB boot disk is on the 2nd (lower) USB 2 port. The USB HDD is identified as sda to the system, and the usb stick is sdb.
This replaced a previous Pi4 setup with the same HDD. Only change is the PI4 was replaced with a Pi5.

Issue:
After some time left to idle, the HDD appears to spin down. (i have not been able to time it)
when this happens, and you hit the share via SMB, the share takes a brief moment to respond. When the share will report as no items available. When you look at lsblk, the drive that was previously identified as sda, is now sdc. When this happens, dmesg says
`
[60651.191211] usb 2-1: cmd cmplt err -71
[60656.378942] scsi host1: uas_pre_reset: timed out
[60656.379169] sd 1:0:0:0: [sda] tag#1 uas_zap_pending 0 uas-tag 1 inflight: CMD
[60656.379174] sd 1:0:0:0: [sda] tag#1 CDB: opcode=0x28 28 00 00 04 3a 64 00 00 04 00
[60656.379187] sd 1:0:0:0: [sda] tag#1 UNKNOWN(0x2003) Result: hostbyte=0x01 driverbyte=DRIVER_OK cmd_age=5s
[60656.379191] sd 1:0:0:0: [sda] tag#1 CDB: opcode=0x28 28 00 00 04 3a 64 00 00 04 00
[60656.379193] I/O error, dev sda, sector 2216736 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 2
[60656.388475] Buffer I/O error on dev sda1, logical block 276836, async page read
[60656.388485] Buffer I/O error on dev sda1, logical block 276837, async page read
[60656.388487] Buffer I/O error on dev sda1, logical block 276838, async page read
[60656.388490] Buffer I/O error on dev sda1, logical block 276839, async page read
[60656.389441] Buffer I/O error on dev sda1, logical block 276836, async page read
[60656.389448] Buffer I/O error on dev sda1, logical block 276837, async page read
[60656.389451] Buffer I/O error on dev sda1, logical block 276838, async page read
[60656.389453] Buffer I/O error on dev sda1, logical block 276839, async page read
[60656.389524] Buffer I/O error on dev sda1, logical block 276920, async page read
[60656.389528] Buffer I/O error on dev sda1, logical block 276921, async page read
[60656.435027] sd 1:0:0:0: [sda] Synchronizing SCSI cache
[60656.674944] sd 1:0:0:0: [sda] Synchronize Cache(10) failed: Result: hostbyte=0x07 driverbyte=DRIVER_OK
[60656.675300] usb 2-1: Disable of device-initiated U1 failed.
[60656.675314] usb 2-1: Disable of device-initiated U2 failed.
[60656.675678] usb 2-1: Enable of device-initiated U1 failed.
[60656.675899] usb 2-1: Enable of device-initiated U2 failed.
[60656.875059] usb 2-1: reset SuperSpeed USB device number 2 using xhci-hcd
[60656.901913] scsi host2: uas
[60656.903389] scsi 2:0:0:0: Direct-Access     SABRENT                   0204 PQ: 0 ANSI: 6
[60656.907585] sd 2:0:0:0: Attached scsi generic sg0 type 0
[60661.612014] sd 2:0:0:0: [sdc] 1220942646 4096-byte logical blocks: (5.00 TB/4.55 TiB)
[60661.612197] sd 2:0:0:0: [sdc] Write Protect is off
[60661.612200] sd 2:0:0:0: [sdc] Mode Sense: 53 00 00 08
[60661.612516] sd 2:0:0:0: [sdc] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[60661.612709] sd 2:0:0:0: [sdc] Preferred minimum I/O size 4096 bytes
[60661.612711] sd 2:0:0:0: [sdc] Optimal transfer size 33550336 bytes
[60661.616025]  sdc: sdc1
[60661.616145] sd 2:0:0:0: [sdc] Attached SCSI disk
`
I've updated and rebooted several times, save for one instance where it seem to work for no reason (also documented in the thread above), it has consistently done this. The Pi4 did not do this before. Also note that if i leave it after it changed the hdd to sdc, and let the HDD idle again, and ping the share, it would do the same thing, and reconnect as sdd, e, f etc....

this is super frustrating. Some folks have recommended turning off UAS. not sure if this is the issue.

please look into this!!

fquinner, 2025-02-22:
I have this issue with an external sabrent drive too which is externally powered. I noticed it is pretty much 19 minutes on the nose that it fails, so i assumed its spindown / power saving related. So my somewhat inelegant workaround was to have a crontab simply periodically touch a file on the mounted drive to keep it alive. Has worked for me for a few hours now which is better than usual. Hoping it keeps up!

```
crontab -e
* * * * * touch /ext/keepalive
```

Primeey, 2025-03-13:
Similar issue on my end

Raspberry Pi 5
Silicon Power 4TB SSD 3D NAND A55
SABRENT 2.5” SATA HARD DRIVE/SSD TO USB TYPE-C™ ADAPTER
Official Raspberry Pi 5 Power supply.

dmesg:

```
[ 5428.342608] usb 2-1: cmd cmplt err -71
[ 5433.712433] scsi host0: uas_pre_reset: timed out
[ 5433.920499] sd 0:0:0:0: [sda] tag#16 uas_zap_pending 0 uas-tag 1 inflight: CMD
[ 5433.920514] sd 0:0:0:0: [sda] tag#16 CDB: opcode=0x2a 2a 00 1d c8 81 00 00 00 01 00
[ 5433.920534] sd 0:0:0:0: [sda] tag#16 UNKNOWN(0x2003) Result: hostbyte=0x01 driverbyte=DRIVER_OK cmd_age=5s
[ 5433.920543] sd 0:0:0:0: [sda] tag#16 CDB: opcode=0x2a 2a 00 1d c8 81 00 00 00 01 00
[ 5433.920549] I/O error, dev sda, sector 3997435904 op 0x1:(WRITE) flags 0x9800 phys_seg 1 prio class 2
[ 5433.920556] Buffer I/O error on dev sda1, logical block 499679232, lost sync page write
[ 5433.920582] JBD2: I/O error when updating journal superblock for sda1-8.
[ 5433.920585] Aborting journal on device sda1-8.
[ 5433.920614] device offline error, dev sda, sector 3997435904 op 0x1:(WRITE) flags 0x9800 phys_seg 1 prio class 2
[ 5433.920620] device offline error, dev sda, sector 3997435904 op 0x1:(WRITE) flags 0x9800 phys_seg 1 prio class 2
[ 5433.920623] Buffer I/O error on dev sda1, logical block 499679232, lost sync page write
[ 5433.920626] JBD2: I/O error when updating journal superblock for sda1-8.
[ 5433.921465] device offline error, dev sda, sector 0 op 0x1:(WRITE) flags 0x800 phys_seg 0 prio class 0
[ 5433.921785] EXT4-fs (sda1): shut down requested (2)
[ 5433.968454] sd 0:0:0:0: [sda] Synchronizing SCSI cache
[ 5434.312446] sd 0:0:0:0: [sda] Synchronize Cache(10) failed: Result: hostbyte=0x07 driverbyte=DRIVER_OK
[ 5434.312786] usb 2-1: Disable of device-initiated U1 failed.
[ 5434.312802] usb 2-1: Disable of device-initiated U2 failed.
[ 5434.313157] usb 2-1: Enable of device-initiated U1 failed.
[ 5434.313268] usb 2-1: Enable of device-initiated U2 failed.
[ 5434.496573] usb 2-1: reset SuperSpeed USB device number 2 using xhci-hcd
[ 5434.523884] scsi host1: uas
[ 5434.524865] scsi 1:0:0:0: Direct-Access SABRENT 0204 PQ: 0 ANSI: 6
[ 5434.528779] sd 1:0:0:0: Attached scsi generic sg0 type 0
[ 5434.557224] EXT4-fs (sda1): unmounting filesystem 7e582b6a-4c87-4ac4-8ef4-c74b145fc0c8.
[ 5434.565587] Buffer I/O error on dev sda1, logical block 0, lost sync page write
[ 5434.565599] EXT4-fs (sda1): I/O error while writing superblock
[ 5435.104922] sd 1:0:0:0: [sdb] 1000196694 4096-byte logical blocks: (4.10 TB/3.73 TiB)
[ 5435.105135] sd 1:0:0:0: [sdb] Write Protect is off
[ 5435.105139] sd 1:0:0:0: [sdb] Mode Sense: 53 00 00 08
[ 5435.105493] sd 1:0:0:0: [sdb] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[ 5435.152643] sd 1:0:0:0: [sdb] Preferred minimum I/O size 4096 bytes
[ 5435.152649] sd 1:0:0:0: [sdb] Optimal transfer size 33550336 bytes
[ 5435.178018] sdb: sdb1
[ 5435.178131] sd 1:0:0:0: [sdb] Attached SCSI disk
```

sigmabeta, 2025-09-20:
This is still an issue for me, almost the exact same hardware configuration as the previous commenter.

proteus0703, 2026-10-04:
## Pi 5 + JMicron (JMS578) USB-SATA bridges: random disconnects + automated recovery workaround

Adding a data point to this issue, plus a workaround that recovers automatically without a reboot or power-cycle (doesn't fix the root cause, but makes it a non-event when it happens).

### Setup
- Raspberry Pi 5, OpenMediaVault, Jellyfin for media serving
- Two separate external HDD enclosures, each with its own dedicated PSU (not bus-powered)
- Bridge chip: JMicron JMS578 (`lsusb`: `152d:0578`)
- Connected via USB 2.0 (previously tried USB 3.0, same underlying issue, switched to 2.0 for other reasons)
- `usbcore.autosuspend=-1` and `usb-storage.quirks=152d:0578:u` already applied — issue persists regardless

### Symptom
During extended playback (consistently 30–90 min in), one drive drops mid-stream. If a second drive is also active, it independently drops too, usually 30–40 min later. Both drives are otherwise completely healthy — `smartctl -a` shows zero reallocated/pending sectors, zero UDMA CRC errors, normal temps.

### Root cause (confirmed via dmesg)

usb 1-2: USB disconnect, device number 2
sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) ... I/O error
device offline error, dev sda ...
EXT4-fs (sda1): shut down requested (2)
usb usb1-port2: attempt power cycle
usb 1-2: new high-speed USB device number 5 using xhci-hcd

The USB link itself drops, the kernel auto-recovers the port ~9 seconds later, and the drive re-enumerates as a **new** device (new device/scsi host number) — same disk, same UUID, but the kernel treats it as freshly plugged in. The ext4 filesystem had already force-unmounted dirty from the I/O errors, and nothing automatically remounts it onto the old mount point just because a matching UUID reappears — hence the share showing empty until manual intervention.

`vcgencmd get_throttled` was `0x0` at the exact moment of the drop, ruling out thermal/undervoltage on the Pi's own rail as the trigger.

### Workaround — automated recovery (not a fix)
Since the drive always comes back healthy on its own, the udev "add" event can trigger an automatic remount + service restart. Two gotchas if you roll your own:
1. Don't match broadly on `ENV{ID_FS_TYPE}=="ext4"` alone — the Pi's own root partition is also ext4 and will re-fire this on every boot. Scope to the specific device nodes (`KERNEL=="sd[ab]1"` or better, by UUID).
2. Don't run `mount`/`systemctl` synchronously from `RUN+=` — it can block the udev worker and deadlock against udev's own device-settling, especially right after a dirty reconnect. Hand off to a systemd oneshot service instead.

`/usr/local/bin/usb-remount.sh`:
```bash
#!/bin/bash
sleep 3
mount -a
systemctl restart jellyfin
```

`/etc/udev/rules.d/99-usb-remount.rules`:

SUBSYSTEM=="block", ACTION=="add", KERNEL=="sd[ab]1", ENV{ID_FS_TYPE}=="ext4", TAG+="systemd", ENV{SYSTEMD_WANTS}+="usb-remount.service"

`/etc/systemd/system/usb-remount.service`:
```ini
[Unit]
Description=Remount NAS drives and restart Jellyfin after USB re-enumeration
After=local-fs.target

[Service]
Type=oneshot
ExecStart=/usr/local/bin/usb-remount.sh
```

```bash
sudo systemctl daemon-reload
sudo udevadm control --reload-rules
```

Result: drive drops, kernel recovers the port, udev fires, drive remounts and Jellyfin restarts — all within a few seconds, no manual power cycle needed.

### Still looking for
A real fix rather than a recovery workaround — if anyone's found a kernel param, EEPROM update, or firmware revision that stops the disconnect from happening in the first place on Pi 5 specifically, would love to hear it.