raspberrypi/linux #6059: Undefined symbol failures if built as module

srcshelton, 2024-03-22:
### Describe the bug

If `CONFIG_USB_DWCOTG=y` is set but `CONFIG_USB=m`, building the `6.1.73`/`stable_20240124` (and likely previous) kernel(s) fails with:

```
  LD      .tmp_vmlinux.kallsyms1
ld.lld: error: undefined symbol: usb_hcd_check_unlink_urb
>>> referenced by xarray.c
>>>               vmlinux.o:(_complete)
>>> referenced by xarray.c
>>>               vmlinux.o:(dwc_otg_urb_dequeue)

ld.lld: error: undefined symbol: usb_hcd_unlink_urb_from_ep
>>> referenced by xarray.c
>>>               vmlinux.o:(_complete)
>>> referenced by xarray.c
>>>               vmlinux.o:(_complete)
>>> referenced by xarray.c
>>>               vmlinux.o:(dwc_otg_urb_enqueue)
>>> referenced 1 more times

ld.lld: error: undefined symbol: usb_hcd_giveback_urb
>>> referenced by xarray.c
>>>               vmlinux.o:(_complete)
>>> referenced by xarray.c
>>>               vmlinux.o:(completion_tasklet_func)
>>> referenced by xarray.c
>>>               vmlinux.o:(dwc_otg_urb_dequeue)

ld.lld: error: undefined symbol: usb_create_hcd
>>> referenced by xarray.c
>>>               vmlinux.o:(hcd_init)

ld.lld: error: undefined symbol: usb_put_hcd
>>> referenced by xarray.c
>>>               vmlinux.o:(hcd_init)
>>> referenced by xarray.c
>>>               vmlinux.o:(hcd_remove)

ld.lld: error: undefined symbol: usb_add_hcd
>>> referenced by xarray.c
>>>               vmlinux.o:(hcd_init)

ld.lld: error: undefined symbol: usb_remove_hcd
>>> referenced by xarray.c
>>>               vmlinux.o:(hcd_remove)

ld.lld: error: undefined symbol: usb_hcd_resume_root_hub
>>> referenced by xarray.c
>>>               vmlinux.o:(hcd_start)
>>> referenced by xarray.c
>>>               vmlinux.o:(_start)

ld.lld: error: undefined symbol: usb_hcd_link_urb_to_ep
>>> referenced by xarray.c
>>>               vmlinux.o:(dwc_otg_urb_enqueue)
make[3]: *** [/usr/src/linux-6.1.73-raspberrypi/scripts/Makefile.vmlinux:34: vmlinux] Error 1
make[2]: *** [/usr/src/linux-6.1.73-raspberrypi/Makefile:1255: vmlinux] Error 2
make[1]: *** [/usr/src/linux-6.1.73-raspberrypi/Makefile:369: __build_one_by_one] Error 2
make: *** [Makefile:238: __sub-make] Error 2
```

### Steps to reproduce the behaviour

```
cd linux ; make USB_DWCOTG=y USB=m clean vmlinux
```

### Device (s)

Raspberry Pi CM4

### System

Kernel `6.1.73`

### Additional context

The easy fix would be to make `CONFIG_USB_DWCOTG` depend on `CONFIG_USB=y`, but might it be possible to fix the DWC driver code to be able to be built as a module itself, or integrated in such a way that it works if USB support overall is a module?

srcshelton, 2024-04-16:
Similarly, `CONFIG_FB_BCM2708` requires `CONFIG_FB_DEVICE` (for `dev` in `struct fb_info`), despite the Help text for `CONFIG_FB_DEVICE` reading:

> Say Y here if you want the legacy /dev/fb* device file and interfaces within sysfs anc [sic] procfs. It is only required if you have userspace programs that depend on fbdev for graphics output.
> This does not affect the framebuffer console.
> If unsure, say N.

pelwell, 2024-04-18:
Why are you singling out 6.1 for the USB_DWCOTG dependency problem? For me, the 6.6 kernel is the same.

pelwell, 2024-04-18:
See #6112.

srcshelton, 2024-04-18:
> Why are you singling out 6.1 for the USB_DWCOTG dependency problem? For me, the 6.6 kernel is the same.

I've not tested `USB_DWCOTG` on 6.6 myself, but I'm not meaning to infer the the issue is specific to 6.1.

(When I first filed the bug, I was trying to build a 6.1 kernel.  The 6.6 kernel sources still aren't tagged by the looks of things (see https://github.com/raspberrypi/linux/issues/6063), but I've moved on to using them regardless.)