raspberrypi/linux #6067: ADS7846 Touch controller does not work correctly after upgrade to kernel 6.6.20

hadrava, 2024-03-25:
### Describe the bug

ADS7846 Touch controller does not work anymore after upgrade to the latest kernel 6.6.20 on 64-bit Raspberry Pi OS Bookworm. It works well with the older kernel 6.1.0.

Tested on Raspberry Pi 4 Model B with Waveshare 10.1 Inch Resistive touch Screen (https://www.waveshare.com/wiki/10.1inch_HDMI_LCD https://www.waveshare.com/product/displays/lcd-oled/lcd-oled-1/10.1inch-hdmi-lcd.htm)

Touch event comes through only occasionally. If I keep finger on the display long enough, I eventually receive event with correct position. (Approximately once in a a few minutes.) So it looks a bit to me like there is something wrong with `penirq` signal and the SPI connection is fine. I could be wrong though.

### Steps to reproduce the behaviour

 - Install fresh 64-bit Bookworm image Raspberry Pi OS Lite (tested with 2024-03-15-raspios-bookworm-arm64-lite.img)
 - Connect the Waveshare 10.1 Inch display and add dtoverlay configuration for ADS7846:
 ```
echo 'dtoverlay=ads7846,cs=1,penirq=25,penirq_pull=2,speed=50000,keep_vref_on=0,swapxy=0,pmax=255,xohms=150,xmin=200,xmax=3900,ymin=200,ymax=3900' >> /boot/firmware/config.txt
 ```
 
 - Install libinput tools: `apt install libinput-tools`
 - Reboot
 - Start `libinput debug-events`
 - Touch the display: Console should be filled with the libinput events from the touch screen. But usually nothing happens with the kernel 6.6.
 - Downgrade to kernel 6.1 (Confirm kernel removal with answer No):
 ```
 apt install linux-image-6.1.0-rpi8-rpi-v8 linux-image-6.6.20+rpt-rpi-v8- linux-image-6.6.20+rpt-rpi-2712-
reboot
 ```
 - Command `libinput debug-events` will now show all the touch events as expected.

### Device (s)

Raspberry Pi 4 Mod. B

### System

```
Raspberry Pi reference 2024-03-15
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f19ee211ddafcae300827f953d143de92a5c6624, stage2
```
```
Feb 29 2024 12:24:53 
Copyright (c) 2012 Broadcom
version f4e2138c2adc8f3a92a3a65939e458f11d7298ba (clean) (release) (start)
```
Kernel with broken touch screen:
```
Linux raspberrypi 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux
```
It works well with the following kernel:
```
Linux raspberrypi 6.1.0-rpi8-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux
```

### Logs

Boot with broken touch screen kernel 6.6.20:
[dmesg-broken.txt](https://github.com/raspberrypi/linux/files/14747277/dmesg-broken.txt)

The touch screen works well with kernel 6.1.0:
[dmesg-working.txt](https://github.com/raspberrypi/linux/files/14747278/dmesg-working.txt)

6by9, 2024-03-25:
Duplicate of https://github.com/raspberrypi/bookworm-feedback/issues/88
Already fixed by #6029.

Use rpi-update to get the latest kernel with the fix.

closed 2024-03-26 by pelwell

EmEiBee, 2024-04-17:
... nope ...
I had the same problem where after a dist upgrade my Waveshare Touch (the smaller 3.5") just stopped working no matter what I did (and I tried a lot in the last 3 days).
This was the last attempt; an rpi update didn't make any change either...

`dtoverlay=ads7846,cs=0,penirq=25,penirq_pull=2,speed=50000,keep_vref_on=1,swapxy=0,pmax=255,xohms=150`

That have worked before like a smart ...

pelwell, 2024-04-17:
Your dtoverlay line is too long - there is a 100 character limit. It may not make it work, but you should split the line like this:
```
dtoverlay=ads7846,cs=0,penirq=25,penirq_pull=2,speed=50000,keep_vref_on=1
dtparam=swapxy=0,pmax=255,xohms=150
```

EmEiBee, 2024-04-17:
I have just change that without any result; same as before: no touch
BTW: The "long" line have worked before without trouble

ArcAIN6, 2024-10-24:
I'm also having issues getting the ADS7846 touchscreen from waveshare to function on a raspberry pi 2B. 
Steps to reproduce: 
-Use Raspberry Pi Imager
-Select Raspberry Pi 2 Model B
-Select Raspberry Pi OS (32-bit)
-Boot into system
-run 'sudo apt-get update && upgrade'
-run sudo reboot
-Test touch screen (NO JOY)
`libinput debug-events
(Event shows device was added, no events are happening when touching the screen)
-Google for 6 hours attempting to find a solution, after sifting through all the ancient information and calls for "just buy new things" find this issue ticket... 
-run `sudo rpi-update`
-run `sudo reboot now`
- Test touch screen... no change.. still appears in libinput, still doesn't capture any even data when touching the screen.. 

Kernel: 
`Linux pi2 6.6.58-v7+ #1809 SMP Wed Oct 23 11:51:11 BST 2024 armv7l GNU/Linux`

/boot/firmware/config.txt: 
`
#Waveshare Touch Screen:
hdmi_group=2
hdmi_mode=87
hdmi_cvt 800 480 60 6 0 0 0
hdmi_drive=1
dtoverlay=ads7846,cs=1,penirq=25,xmin=200,xmax=3900,ymin=200,ymax=3900,speed=50000
`
Not sure if something was rolled back, or steamrolled by something else, but currently unable to get the 5" waveshare touchscreen to function propery. display is perfect, touch no joy

6by9, 2024-10-24:
@ArcAIN6 Open a new issue with full details of which kernel versions have and have not worked. 

The [wiki for Waveshare's 5" touchscreen using ADS7846](https://www.waveshare.com/wiki/5inch_HDMI_LCD) says to use their `dtoverlay=waveshare-ads7846` overlay line, not the overlay provided as part of the Pi kernel. That is the guidance provided by the manufacturer, so users should follow it. If those instructions don't work, then speak to the manufacturer.

ArcAIN6, 2024-10-26:
> @ArcAIN6 Open a new issue with full details of which kernel versions have and have not worked.
>
> The [wiki for Waveshare's 5" touchscreen using ADS7846](https://www.waveshare.com/wiki/5inch_HDMI_LCD) [...]

tried theirs as well (which is almost always a year or more outdated) and still no go. To be honest,  i don't have the time, nor the inclination to reinstall the OS several times over to find out which kernel borked it. so i'll just calk it up to yet another thing broken in rasbian