raspberrypi/linux #6068: TC358743 produces BGR instead of RGB

mdevaev, 2024-03-25:
### Describe the bug

I'm trying to use RGB24 format with TC358743 instead of UYVY, and noticed that it produces BGR sequence in the raw data instead of RGB. I used the blue fullscreen image as an HDMI source and made a stream dump that confirms this, getting something like `[255, 0, 0]` for first 3 bytes.

If you have a top secret datasheet under Toshiba NDA, you can check pages 36/261 and 37/261. This confirms that UYVY is implemented correctly, but RGB has the reverse sequence. Apparently, this bug went unnoticed because everyone was using UYVY. I need RGB to output the image in DRM and encode in M2M at the same time.

### Steps to reproduce the behaviour

Use any capture software with TC358743 in RGB24 format and dump the raw image.

### Device (s)

Raspberry Pi CM4, Raspberry Pi CM4 Lite

### System

* Arch Linux ARM, but it doesn't matter
* `version c14213886ae1e31951022b74ce776ce7ee7023ee (clean) (release) (start)`
* `Linux pikvm 6.6.21-4-rpi #1 SMP Wed Mar 13 20:36:42 UTC 2024 armv7l GNU/Linux`
* Apparently, the problem is relevant for any version of kernels.

### Additional context

I found this bug when encoding JPEG in M2M and using libjpeg. It's the same with M2M H.264. Interestingly, because of the endianness, DRM_FORMAT_RGB888 is actually BGR, so the rendering of the TC* chip output in DRM was correct colors, while the encoding was not.

I don't know if it is possible to fix this with little blood without loss of performance, given that it looks like you need to use the reverse order to put bytes in memory after receiving from CSI. This is not a big problem and I can make a userspace workaround for this, but it would probably be better to fix it upstream if the fix is not complex or fatal to performance.

6by9, 2024-03-26:
And it wasn't unnoticed - there are various forum threads where GStreamer's capssetter is being used to swap bgr to rgb when using tc358743.

mdevaev, 2024-03-26:
Thanks for researching this. It looks like total hell. I believe CISCO can use byte swapping in its userspace (or a format overriding, whatever), and it looks like this is the path I should follow right now to avoid headache.

MarkusLange, 2025-06-27:
Silly question I try to pipe a signal (HDMI. RGB) through the pi, TC358743 input and PI HDMI output the tricks with capssetter and bgr and rgb swap works as long I didn't scale, for example 1080p or 1080i to 1080p or 1080i works flawless but if I scale from 720 or to 720 and use videoconvert and videoscale the colorshift with capssetter did not work anymore can someone help me with that?

MarkusLange, 2025-07-02:
Sorry for that. On the forum I hat to wait as it is moderated so I can't get the link to it instant and I wasn't near any device when it comes online so I hadn't adopt it in this issue in proper time.

MarkusLange, 2025-08-29:
> If mainline did adopt that, then for downstream I'd be tempted to put in the same mapping twice, so both BGR888_1X24 and RGB888_1X24 map to V4L2_PIX_FMT_BGR24. The handling is generally dodgy, and the CSI2 spec specifies the layout that is V4L2_PIX_FMT_BGR24. If your random source device produces RGB24 then it's non-CSI2 compliant and shouldn't be using that CSI2 data ID.

Since I have no clue, so asking cautious.

from the link to the first patch, BGR and RGB both use MIPI_CSI2_DT_RGB888 is this correct?
```
[diff](https://lore.kernel.org/linux-media/20250606-rpi-unicam-rgb-bgr-fix-v1-1-9930b963f3eb@kernel.org/#iZ31drivers:media:platform:broadcom:bcm2835-unicam.c) --git a/drivers/media/platform/broadcom/bcm2835-unicam.c b/drivers/media/platform/broadcom/bcm2835-unicam.c
index f10064107d543caf867249d0566a0f42d6d8c4c6..1f549019efd53c9aae83193e74f1a3601ebf274d 100644
--- a/drivers/media/platform/broadcom/bcm2835-unicam.c
+++ b/drivers/media/platform/broadcom/bcm2835-unicam.c
@@ -338,15 +338,10 @@ static const struct unicam_format_info unicam_image_formats[] = {
 	/* RGB Formats */
 		.fourcc		= V4L2_PIX_FMT_RGB565, /* gggbbbbb rrrrrggg */
 		.code		= MEDIA_BUS_FMT_RGB565_1X16,
 		.depth		= 16,
 		.csi_dt		= MIPI_CSI2_DT_RGB565,
-	}, {
-		.fourcc		= V4L2_PIX_FMT_RGB24, /* rgb */
-		.code		= MEDIA_BUS_FMT_RGB888_1X24,
-		.depth		= 24,
-		.csi_dt		= MIPI_CSI2_DT_RGB888,
 	}, {
 		.fourcc		= V4L2_PIX_FMT_BGR24, /* bgr */
 		.code		= MEDIA_BUS_FMT_BGR888_1X24,
 		.depth		= 24,
 		.csi_dt		= MIPI_CSI2_DT_RGB888,
```

superware, 2025-09-26:
> FWIW there are a couple of threads upstream, but nothing resolved yet.

Hello, also experiencing this color-ordering issue with TC358743, anything resolved yet?

MarkusLange, 2025-09-28:
@superware I used this:

> https://lore.kernel.org/linux-media/20250612-csi-bgr-rgb-v1-3-dc8a309118f8@kernel.org/ - changing to use that MBUS code, however it will break any existing userspace app that currently select MEDIA_BUS_FMT_RGB888_1X24 in their Media Controller setup (eg TC358743 on Pi5 I expect, so likely PiKVM). Userspace regressions are a big no-no.

Yes the big no-no @6by9 was talking about it changed the MBUS format the TC358743 used only changed the tc358743.c, I wrote a little script that puts everyting in the right order

```
#!/bin/bash

sudo apt install -y bc bison flex libssl-dev make git
git clone --depth=1 https://github.com/raspberrypi/linux

cd linux
sed -i 's#MEDIA_BUS_FMT_RGB888_1X24#MEDIA_BUS_FMT_BGR888_1X24#g' $HOME/linux/drivers/media/i2c/tc358743.c

KERNEL=kernel_2712
make bcm2712_defconfig

make -j6 Image.gz modules dtbs
make -j6 Image modules dtbs

sudo make -j6 modules_install

sudo cp /boot/firmware/$KERNEL.img /boot/firmware/$KERNEL-backup.img
sudo cp arch/arm64/boot/Image.gz /boot/firmware/$KERNEL.img
sudo cp arch/arm64/boot/dts/broadcom/*.dtb /boot/firmware/
sudo cp arch/arm64/boot/dts/overlays/*.dtb* /boot/firmware/overlays/
sudo cp arch/arm64/boot/dts/overlays/README /boot/firmware/overlays/
```
It changed RGB888 to BGR888 inside the tc358743.c so it used the  MBUS format BGR888 and then it compiles the kernel for the Pi5 and replace the existing one.