raspberrypi/linux #6077: Can't set CPU affinity on RP1 downstream interrupts

P33M, 2024-03-27:
### Describe the bug

The RP1 irqchip doesn't declare an affinity set function, so writing a sensible value to `/proc/irq/<blah>/smp_affinity` returns ` write error: Invalid argument`.

### Steps to reproduce the behaviour

Write any mask to smp_affinity in procfs.

### Device (s)

Raspberry Pi 5

### System

Any Pi 5 kernel version

### Additional context

A fix that nominally seems to work is
```
diff --git a/drivers/mfd/rp1.c b/drivers/mfd/rp1.c
index 4adfbb767136..0a498a670a81 100644
--- a/drivers/mfd/rp1.c
+++ b/drivers/mfd/rp1.c
@@ -141,11 +141,20 @@ static int rp1_irq_set_type(struct irq_data *irqd, unsigned int type)
        return ret;
 }

+static int rp1_irq_set_affinity(struct irq_data *irqd, const struct cpumask *dest, bool force)
+{
+       struct rp1_dev *rp1 = irqd->domain->host_data;
+       struct irq_data *pcie_irqd = rp1->pcie_irqds[irqd->hwirq];
+
+       return msi_domain_set_affinity(pcie_irqd, dest, force);
+}
+
 static struct irq_chip rp1_irq_chip = {
        .name            = "rp1_irq_chip",
        .irq_mask        = rp1_mask_irq,
        .irq_unmask      = rp1_unmask_irq,
        .irq_set_type    = rp1_irq_set_type,
+       .irq_set_affinity = rp1_irq_set_affinity,
 };

 static void rp1_chained_handle_irq(struct irq_desc *desc)
```
In that my eth0 interrupts now appear on a CPU of my choosing - but is it the right fix?

P33M, 2024-03-27:
If it is the right fix, it's incomplete - the pinctrl irqchip needs similar treatment - but you can only set the affinity of an entire GPIO bank (one top-level interrupt).

pelwell, 2024-03-27:
The default irq_set_affinity for MSI IRQ chips is msi_domain_set_affinity, so you're suggestion looks plausible. The pinctrl change should be equally trivial. Leave it with me.

Noltari, 2024-04-10:
Backported to OpenWrt ([v6.1 kernel](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commitdiff;h=65123335189e4600d7986e3d2fa461e6c823fb2d)), tested and working perfectly, so this issue can be closed.
Thanks @pelwell!

closed 2024-04-10 by pelwell