raspberrypi/linux #6079: USB Touchscreen kills xHCI Controller

s-dinda, 2024-03-27:
### Describe the bug

On Pi5 Bookworm an external usb touchscreens seems to kill the xHCI controller. The same device worked fine with a previous Pi4B (Bullseye).

I am trying to connect a touchscreen via USB and HDMI, unfortunately connecting the USB device seems to kill some of the other connected USB devices - until a reboot.

Thanks in advance!

### Steps to reproduce the behaviour

1. Boot with all devices connected except the touchscreen.
2. Connect the USB touchscreen (with or without an additional powered USB Hub)
3. The touchscreen doesn't work at all. Instead the xHCI Controller died.

### Device (s)

Raspberry Pi 5

### System

```
$ cat /etc/rpi-issue
Raspberry Pi reference 2024-03-15
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f19ee211ddafcae300827f953d143de92a5c6624, stage4
```
```
$ vcgencmd version
2024/02/16 15:28:41
Copyright (c) 2012 Broadcom
version 4c845bd3 (release) (embedded)
```
```
$ uname -a
Linux fotobox 6.6.20+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux
```

### Logs

dmesg
```
...
[  855.080358] usb 1-2: new full-speed USB device number 3 using xhci-hcd
[  855.229606] usb 1-2: New USB device found, idVendor=0416, idProduct=c168, bcdDevice= 0.00
[  855.229611] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  855.229614] usb 1-2: Product: MTouch
[  855.229617] usb 1-2: Manufacturer: TSTP
[  855.229619] usb 1-2: SerialNumber: CMTP_1.0
[  865.656274] xhci-hcd xhci-hcd.0: xHCI host not responding to stop endpoint command
[  865.676453] xhci-hcd xhci-hcd.0: xHCI host controller not responding, assume dead
[  865.676464] xhci-hcd xhci-hcd.0: HC died; cleaning up
[  865.677182] usb 1-1: USB disconnect, device number 2
[  865.736634] usb 1-2: USB disconnect, device number 3
```

pelwell, 2024-03-27:
1. Which power supply are you using on the Pi?
2. What devices are attached to it?
3. Are any of those devices independently powered?

s-dinda, 2024-03-27:
I'm using the official 27W Power Supply

The other devices are a printer, a DSLR and a mouse:
```
$ lsusb -tv
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 1: Dev 2, If 0, Class=Printer, Driver=, 480M
        ID 06d3:3b36 Mitsubishi Electric Corp. CP-D80DW
    |__ Port 2: Dev 3, If 0, Class=Imaging, Driver=, 480M
        ID 04a9:32b4 Canon, Inc. EOS Rebel T6
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 1: Dev 2, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        ID 046a:b091 CHERRY Mouse
```
The touchscreen and all of them (except the mouse) are independently powered.

x130844, 2024-03-28:
Same problem here, with the Canakit 45W usb-c PD power supply. 
the usb sees mtouch, then destroys the all USBs. 
Another touch driver (I have a 2nd screen) using the elecrow. ads7846. has issues as well, it sees SOME touch inputs, once every 20 seconds or so. 
If I set the exact same OS (raspbian) with same settings, **same literal displays** on the rpi4   everything works as expected. 
I have no idea what's happening.

6by9, 2024-03-28:
ads7846 is totally unrelated as it is an I2C touchscreen. Should already be fixed via #6029.

x130844, 2024-03-29:
> ads7846 is totally unrelated as it is an I2C touchscreen. Should already be fixed via #6029.

I can confirm that the ads7846 display works now, thx!    
Sadly the USB one still is not (mtouch one)

Winor, 2024-04-29:
I'm having the same problem with Mtouch on bookworm and latest stable firmware on pi5, seems like there is no solution. I was using Ubuntu before on the same pi5 and the touchscreen was working

Winor, 2024-05-01:
So I had (a lot) of free time to play with this issue and I think I just found something interesting!
### What did I do?
When I first got this rpi few good months ago I definitely installed Ubuntu 23.10 (luckily Ubuntu is changing its default background every release making it easy to know what version was installed even after wiping the sd card) and the mtouch screen was working wonderfully, until...
- I wiped the sd card and Installed latest bookworm
I updated things and installed my software over ssh just to find out the touch screen was not working. and dmesg started showing the ``xHCI host controller not responding, assume dead`` when connecteing the mtouch screen USB cable to the Pi or during boot.
- I wiped the sd card again and Installed Ubuntu 24.04LTS
the touch screen was not working. with the same ``xHCI host controller not responding, assume dead`` mesage.
- at this point I just wanted to go back, I found out I had Ubuntu 23.10 when things were working so I wiped my SD card again expecteing everything to work just as it did before I decided to mess with everything but magically the touch screen was still giving the same notorious dmesg error. 
### What else could have changed?
The eeprom!! I suspect that bookworm [automatically updated](https://github.com/raspberrypi/documentation/blob/5d3819f0fd78e484f25b811ca971fef6a4197fa5/documentation/asciidoc/computers/raspberry-pi/boot-eeprom-rpi4.adoc?plain=1#L18) my outdated 2023 bootloader to the 2024-02-16-2712 version in which something has changed. so I went back to [2023-10-30-2712](https://github.com/raspberrypi/rpi-eeprom/releases/tag/v2023.10.30-2712) and the touch screen was working again under Ubuntu 23.10! and I will soon test with newer OSs

### Was it the EEPROM all along?
What could have changed on the EEPROM side that could effect this?

timg236, 2024-05-01:
https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware-2712/release-notes.md

This is possibly a duplicate of this issue although the symptoms don't sound identical

2024-04-18: Update RP1 firmware to extend PCIe L1 entry timeout to 32 us (latest)

Extend PCIe L1 entry timeout to 32us Fix xhci soft reset on link-down Set useful xhci compatibility bits in GUCTL See https://github.com/raspberrypi/firmware/issues/1877

 If it is the RP1 firmware change then it should work in Ubuntu 24.04 LTS now since the rpi-eeprom-update service never automatically downgrades

Winor, 2024-05-01:
It appears that you are correct, I tested and the latest working release from github releases is [2024-04-17-2712](https://github.com/raspberrypi/rpi-eeprom/releases/tag/v2024.02.17-2712) and with the 2024-04-20 EEPROM release the touch screen stops to work. So I assume it's the RP1 firmware change for me. now the other folks on this thread reprted this issue on March 27 so maybe it's a different issue for them?

Winor, 2024-05-01:
For now I went back to 2024-04-17-2712, installed latest bookworm and disabled  rpi-eeprom-update service with
``sudo systemctl mask rpi-eeprom-update`` and my mtouch screen works perfectly !

Seattle-NK, 2024-11-14:
Jumping on this thread as I have the same issue, but the 2024-04-17-2712 EEPROM release did not fix the issue. I've only tried the latest Bookworm so far.

```typescript
$ cat /etc/rpi-issue
Raspberry Pi reference 2024-10-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, ecda2579ebbc125b321984c571e3128f6fd959d4, stage4
``` 

```typescript
$ vcgencmd version
2024/04/17 13:54:23
Copyright (c) 2012 Broadcom
version 0b7b6f28 (release) (embedded)
``` 

```typescript
$ uname -a
Linux domepi2 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
``` 

```typescript
$ lsusb -tv
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 2: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 1a40:0101 Terminus Technology Inc. Hub
        |__ Port 2: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
            ID 258a:6a88
        |__ Port 2: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
            ID 258a:6a88
        |__ Port 3: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
            ID 04f2:0939 Chicony Electronics Co., Ltd Amazon Basics mouse
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
``` 

```typescript
[ 1947.751928] usb 1-1.1: new full-speed USB device number 3 using xhci-hcd
[ 1947.865414] usb 1-1.1: New USB device found, idVendor=0416, idProduct=c168, bcdDevice= 0.00
[ 1947.865420] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1947.865424] usb 1-1.1: Product: MTouch
[ 1947.865427] usb 1-1.1: Manufacturer: TSTP
[ 1947.865429] usb 1-1.1: SerialNumber: CMTP_1.0
[ 1958.183849] xhci-hcd xhci-hcd.0: xHCI host not responding to stop endpoint command
[ 1958.204029] xhci-hcd xhci-hcd.0: xHCI host controller not responding, assume dead
[ 1958.204046] xhci-hcd xhci-hcd.0: HC died; cleaning up
[ 1958.205040] usb 1-1: USB disconnect, device number 2
[ 1958.205052] usb 1-1.1: USB disconnect, device number 3
[ 1958.205451] usb 2-1: USB disconnect, device number 2
``` 

^ This last dmesg output was a test with an un-powered USB hub between the Pi and the touchscreen. I actually tried three different hubs (two unpowered and one powered) and got the same results as when I plugged it directly into the Pi.

The touchscreen is powered separately, and the Pi has been powered through both a POE+ HAT and the official Raspberry 5A power supply.

janosch1337, 2025-01-01:
hi,

same issue here. rpi 4 with this mtouch usb c monitor is working fine without installing anything. just flash image and touch is working.

then rpi 5 booting up and is not recognizing the touch. mtouch with lsusb is also now showing up....

s-dinda, 2025-05-31:
Hi folks,
any update on this issue?

I just installed the latest Bookworm and the issue still persists:

```
$ cat /etc/rpi-issue
Raspberry Pi reference 2025-05-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5dabc7dc940059dfbc46af5d97b60a1e812523dd, stage4
```

```
$ vcgencmd version
2024/06/05 16:41:49
Copyright (c) 2012 Broadcom
version 6fe0b091 (release) (embedded)
```

```
$ uname -a
Linux fotobox 6.12.25+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64 GNU/Linux
```

```
$ dmesg
...
[    1.247811] usb 3-1: New USB device found, idVendor=04a9, idProduct=32b4, bcdDevice= 0.02
[    1.247817] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    1.247819] usb 3-1: Product: Canon Digital Camera
[    1.247821] usb 3-1: Manufacturer: Canon Inc.
[    1.261339] [drm] Initialized vc4 0.0.0 for axi:gpu on minor 1
[    1.337023] Console: switching to colour frame buffer device 240x67
[    1.354688] vc4-drm axi:gpu: [drm] fb0: vc4drmfb frame buffer device
[    1.428763] usb 3-2: new full-speed USB device number 3 using xhci-hcd
[    1.588215] usb 3-2: New USB device found, idVendor=0416, idProduct=c168, bcdDevice= 0.00
[    1.588220] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    1.588222] usb 3-2: Product: MTouch
[    1.588224] usb 3-2: Manufacturer: TSTP
[    1.588225] usb 3-2: SerialNumber: CMTP_1.0
[   11.489048] platform rp1_firmware: deferred probe pending: (reason unknown)
[   11.489053] platform 1f00178000.pio: deferred probe pending: rp1-pio: failed to find RP1 firmware driver
[   11.744765] xhci-hcd xhci-hcd.1: xHCI host not responding to stop endpoint command
[   11.764933] xhci-hcd xhci-hcd.1: xHCI host controller not responding, assume dead
[   11.772469] xhci-hcd xhci-hcd.1: HC died; cleaning up
[   11.777714] usb 3-1: USB disconnect, device number 2
[   11.778181] usb 3-2: USB disconnect, device number 3
```
The touchscreen is also powered separately, the Pi via official 5A power supply.

mattjb1222, 2025-08-15:
Experiencing the same issue here with a Mtouch display connected by USB (Eviciv 10.1 Touchscreen) - working fine on an old image rpi3b+ but not with latest 64-bit bookworm on rpi5. Keyboard remains operational on another usb port.
`
[    0.690285] mmc1: new ultra high speed DDR50 SDIO card at address 0001
[    0.713555] usb 1-1: new full-speed USB device number 2 using xhci-hcd
[    0.713579] usb 3-2: new low-speed USB device number 2 using xhci-hcd
[    0.872779] usb 1-1: New USB device found, idVendor=0416, idProduct=c168, bcdDevice= 0.00
[    0.872786] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    0.872788] usb 1-1: Product: MTouch
[    0.872790] usb 1-1: Manufacturer: TSTP
[    0.872791] usb 1-1: SerialNumber: CMTP_1.0
[    0.880902] usb 3-2: New USB device found, idVendor=046d, idProduct=c31c, bcdDevice=64.02
[    0.880905] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    0.880907] usb 3-2: Product: USB Keyboard
[    0.880908] usb 3-2: Manufacturer: Logitech
[    1.363913] input: Logitech USB Keyboard as /devices/platform/axi/1000120000.pcie/1f00300000.usb/xhci-hcd.1/usb3/3-2/3-2:1.0/0003:046D:C31C.0001/input/input0
[    1.513576] hid-generic 0003:046D:C31C.0001: input,hidraw0: USB HID v1.10 Keyboard [Logitech USB Keyboard] on usb-xhci-hcd.1-2/input0
[    1.619835] input: Logitech USB Keyboard Consumer Control as /devices/platform/axi/1000120000.pcie/1f00300000.usb/xhci-hcd.1/usb3/3-2/3-2:1.1/0003:046D:C31C.0002/input/input1
[    1.673551] input: Logitech USB Keyboard System Control as /devices/platform/axi/1000120000.pcie/1f00300000.usb/xhci-hcd.1/usb3/3-2/3-2:1.1/0003:046D:C31C.0002/input/input2
[    1.673631] hid-generic 0003:046D:C31C.0002: input,hiddev96,hidraw1: USB HID v1.10 Device [Logitech USB Keyboard] on usb-xhci-hcd.1-2/input1
[   11.229517] xhci-hcd xhci-hcd.0: xHCI host not responding to stop endpoint command
[   11.249685] xhci-hcd xhci-hcd.0: xHCI host controller not responding, assume dead
[   11.257212] xhci-hcd xhci-hcd.0: HC died; cleaning up
[   11.262417] usb 1-1: USB disconnect, device number 2
`

Eddiiie, 2025-09-02:
Same issue.

TSTP touchscreen does not work

```
[   75.346557] usb 3-1: new full-speed USB device number 2 using xhci-hcd
[   75.506581] usb 3-1: New USB device found, idVendor=0416, idProduct=c168, bcdDevice= 0.00
[   75.506587] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   75.506590] usb 3-1: Product: MTouch
[   75.506591] usb 3-1: Manufacturer: TSTP
[   75.506593] usb 3-1: SerialNumber: CMTP_1.0
[   85.890996] xhci-hcd xhci-hcd.1: xHCI host not responding to stop endpoint command
[   85.911171] xhci-hcd xhci-hcd.1: xHCI host controller not responding, assume dead
[   85.911182] xhci-hcd xhci-hcd.1: HC died; cleaning up
[   85.911952] usb 3-1: USB disconnect, device number 2
```

$ cat /etc/rpi-issue
```
Raspberry Pi reference 2025-05-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5dabc7dc940059dfbc46af5d97b60a1e812523dd, stage2
```

$ vcgencmd version
```
2025/05/08 15:13:17
Copyright (c) 2012 Broadcom
version 69471177 (release) (embedded)
```

$ uname -a
Linux printer3a 6.12.34+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.34-1+rpt1~bookworm (2025-06-26) aarch64 GNU/Linux

@Winor 
re-rolling back EEPROM
`sudo systemctl mask rpi-eeprom-update
`
Can this be done at any time?  Or does it need to be done on first boot, or before apt update / upgrade?

MankiniChykan, 2025-10-02:
Same problem here on a pi500 with the uperfect vertical monitor. This works perfectly fine on my pi400.

Monitor an pi500 sufficiently powered separately

dmesg
```
[    4.280785] macb 1f00100000.ethernet eth0: configuring for phy/rgmii-id link mode
[    4.284241] macb 1f00100000.ethernet: gem-ptp-timer ptp clock registered.
[    6.338518] Bluetooth: RFCOMM TTY layer initialized
[    6.338533] Bluetooth: RFCOMM socket layer initialized
[    6.338542] Bluetooth: RFCOMM ver 1.11
[    8.388032] macb 1f00100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[ 2226.329492] usb 1-1: new high-speed USB device number 3 using xhci-hcd
[ 2226.475235] usb 1-1: New USB device found, idVendor=1a40, idProduct=0101, bcdDevice= 1.00
[ 2226.475250] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[ 2226.475254] usb 1-1: Product: USB2.0 HUB
[ 2226.514759] hub 1-1:1.0: USB hub found
[ 2226.514983] hub 1-1:1.0: 4 ports detected
[ 2226.897503] usb 1-1.2: new full-speed USB device number 4 using xhci-hcd
[ 2227.092403] usb 1-1.2: New USB device found, idVendor=0416, idProduct=c168, bcdDevice= 0.00
[ 2227.092421] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2227.092424] usb 1-1.2: Product: MTouch
[ 2227.092426] usb 1-1.2: Manufacturer: TSTP
[ 2227.092428] usb 1-1.2: SerialNumber: CMTP_1.0
[ 2237.445474] xhci-hcd xhci-hcd.0: xHCI host not responding to stop endpoint command
[ 2237.465658] xhci-hcd xhci-hcd.0: xHCI host controller not responding, assume dead
[ 2237.465675] xhci-hcd xhci-hcd.0: HC died; cleaning up
[ 2237.466123] usb 1-1: USB disconnect, device number 3
[ 2237.466134] usb 1-1.2: USB disconnect, device number 4
[ 2237.468152] usb 1-2: USB disconnect, device number 2
[ 2426.877687] usb 3-1: new high-speed USB device number 3 using xhci-hcd
[ 2427.023513] usb 3-1: New USB device found, idVendor=1a40, idProduct=0101, bcdDevice= 1.00
[ 2427.023569] usb 3-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[ 2427.023571] usb 3-1: Product: USB2.0 HUB
[ 2427.070286] hub 3-1:1.0: USB hub found
[ 2427.070492] hub 3-1:1.0: 4 ports detected
[ 2427.453727] usb 3-1.2: new full-speed USB device number 4 using xhci-hcd
[ 2427.648698] usb 3-1.2: New USB device found, idVendor=0416, idProduct=c168, bcdDevice= 0.00
[ 2427.648720] usb 3-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2427.648723] usb 3-1.2: Product: MTouch
[ 2427.648726] usb 3-1.2: Manufacturer: TSTP
[ 2427.648728] usb 3-1.2: SerialNumber: CMTP_1.0
[ 2438.149751] xhci-hcd xhci-hcd.1: xHCI host not responding to stop endpoint command
[ 2438.169936] xhci-hcd xhci-hcd.1: xHCI host controller not responding, assume dead
[ 2438.169955] xhci-hcd xhci-hcd.1: HC died; cleaning up
[ 2438.171638] usb 3-1: USB disconnect, device number 3
[ 2438.171654] usb 3-1.2: USB disconnect, device number 4
[ 2438.173571] usb 3-2: USB disconnect, device number 2
```

I have also had a strange issue with the pi500 going back to bootloader after restart making me reinstall the software, multiple times. Unplugging the the USB cable to the monitor and my wireless keyboard solved the problem. It was like it was picking up my USB HID devices as a storage device and found not boot drive.

Raspberry Pi Bootcode
```
 Read File: %s, %d
 start_x.elf fixup_x.dat start_db.elf fixup_db.dat recovery.elf   autoboot.txt USB Host enumeration failed, no device detected at root
 Hub device found at addr %d, enumerating HUB
 Device found: type = %s, addr = %d
 Mass storage Ethernet adapter Device found: Unknown type
 Trying booting from MSD device addr %d
 USB device configuration set
 Initialised mass storage, block size = %d, block count = %d
 MBR read and decoded successfully
 FAT32 partition decoded successfully
 start.elf start.elf found in first partition on USB MSD, booting
 Failed initialising mass storage device
 Trying booting from Ethernet device addr %d
 Sending DHCP request
 Waiting for dhcp_reply

[... 59 lines cut ...]
```

vcgencmd bootloader_config
```
[all]
BOOT_UART=1
BOOT_ORDER=0xf461   # read right-to-left: 1(SD),6(NVMe),4(USB),f(restart)
NET_INSTALL_AT_POWER_ON=1  # forces Network Install on power-up
```

Changed to
```
[all]
BOOT_UART=1
BOOT_ORDER=0xf1
NET_INSTALL_AT_POWER_ON=0
```
BOOT_ORDER=0xf1 means “try SD (1), then restart (f)” — no USB, no NVMe, no network.

NET_INSTALL_AT_POWER_ON=0 disables the automatic network installer mode on power-up.

Unsure if this is related
```
HDMI-A-2 "DO NOT USE - RTK CX156A 0x00000001 (HDMI-A-2)" Physical size: 340x200 mm Enabled: yes Modes: 640x480 px, 59.939999 Hz 640x480 px, 59.939999 Hz 640x480 px, 60.000000 Hz 720x480 px, 59.939999 Hz 720x480 px, 59.939999 Hz 720x480 px, 60.000000 Hz 720x480 px, 60.000000 Hz 720x576 px, 50.000000 Hz 720x576 px, 50.000000 Hz 800x600 px, 60.317001 Hz 1024x768 px, 60.004002 Hz 1280x720 px, 50.000000 Hz 1280x720 px, 59.939999 Hz 1280x720 px, 60.000000 Hz 1280x720 px, 60.000000 Hz 1152x864 px, 59.966999 Hz 1280x800 px, 59.910000 Hz 1366x768 px, 59.964001 Hz 1440x900 px, 59.901001 Hz 1280x1024 px, 60.020000 Hz 1600x900 px, 60.000000 Hz 1680x1050 px, 59.882999 Hz 1920x1080 px, 50.000000 Hz 1920x1080 px, 59.939999 Hz 1920x1080 px, 60.000000 Hz 1920x1080 px, 60.000000 Hz (preferred, current) Position: 0,0 Transform: 270 Scale: 1.000000
```

MankiniChykan, 2025-10-02:
I followed this sequence to roll the Pi 500 bootloader back to the **2024-04-17** build, reboot, and then freeze it so it doesn’t silently update again.

But ended up with a black screen and a red light on my Pi500. Managed to recovery image. @Winor am I missing something here?

# 1) Download that specific EEPROM image

```bash
cd /tmp
wget -O pieeprom-2024-04-17.bin \
  https://raw.githubusercontent.com/raspberrypi/rpi-eeprom/v2024.02.17-2712/firmware-2712/default/pieeprom-2024-04-17.bin
```

(That file lives in the `v2024.02.17-2712` tag of the rpi-eeprom repo. ([[GitHub](https://github.com/raspberrypi/rpi-eeprom/tags?after=v2024.09.10-2712)][1]))

Optional: quick sanity check the file you just pulled (size should be ~1 MB):

```bash
ls -lh /tmp/pieeprom-2024-04-17.bin
```

# 2) Flash it

```bash
sudo rpi-eeprom-update -f /tmp/pieeprom-2024-04-17.bin
```

You should see it write, verify, and report **UPDATE SUCCESSFUL**. The `rpi-eeprom-update` tool is the canonical way to apply a specific .bin image. ([[GitHub](https://github.com/Tekki/raspberrypi-documentation/blob/master/hardware/raspberrypi/booteeprom.md?utm_source=chatgpt.com)][2])

# 3) Reboot to load the downgraded bootloader

```bash
sudo reboot
```

# 4) Confirm the version after reboot

```bash
vcgencmd bootloader_version
vcgencmd bootloader_config
```

You’re looking for a **date of 2024/04/17** on the version line. (Release is documented in Raspberry Pi’s bootloader release tags/release notes.) ([[GitHub](https://github.com/raspberrypi/rpi-eeprom/tags?after=v2024.09.10-2712)][1])

# 5) Lock it in (stop auto-updates)

Mask the systemd updater and put the packages on hold so a future `apt upgrade` doesn’t bump you back to a 2025 image.

```bash
# stop future automatic bootloader updates at boot
sudo systemctl mask rpi-eeprom-update

# keep apt from upgrading these packages
sudo apt-mark hold rpi-eeprom rpi-eeprom-images
```

(Masking the service prevents the periodic upgrade; holding the packages blocks apt-driven updates. ([[Raspberry Pi Stack Exchange](https://raspberrypi.stackexchange.com/questions/130723/can-i-be-sure-the-eeprom-configuration-is-the-latest-version?utm_source=chatgpt.com)][3]))

---

## Undo (if/when you want to try a fixed newer EEPROM later)

```bash
sudo systemctl unmask rpi-eeprom-update
sudo apt-mark unhold rpi-eeprom rpi-eeprom-images
sudo apt update
sudo apt install rpi-eeprom rpi-eeprom-images
sudo rpi-eeprom-update -a
sudo reboot
```

pelwell, 2025-10-02:
> I followed this sequence to roll the Pi 500 bootloader back to the 2024-04-17 

So you're trying to install a bootloader on the Pi 500 that dates to before the Pi 500 was released? That's not going to work.

MankiniChykan, 2025-10-03:
> It appears that you are correct, I tested and the latest working release from github releases is [2024-04-17-2712](https://github.com/raspberrypi/rpi-eeprom/releases/tag/v2024.02.17-2712) and with the 2024-04-20 EEPROM release the touch screen no longer works. So I assume the RP1 firmware change is what broke my mtouch display. now the other folks on this thread reported this issue on March 27 so maybe it's a different issue for them?

@pelwell this was on a Pi5 by @Winor...  I am trying anything to get this going. 

My boot order may have been a bit off yet to try again.

> vcgencmd bootloader_config
>
> ```
> [all]
> BOOT_UART=1
> BOOT_ORDER=0xf461   # read right-to-left: 1(SD),6(NVMe),4(USB),f(restart)
> NET_INSTALL_AT_POWER_ON=1  # forces Network Install on power-up
> ``` [...]

pelwell, 2025-10-03:
@P33M I'm looking at the RP1 firmware changes that are being fingered for this issue - there's a change that moves the soft reset setting to GCTL from GUCTL, and also sets SPRSCTRLTRANSEN and RESBWHSEPS on both controllers. Do have any idea which (if any) might be responsible, and why this setting persists after the kernel driver has started? Could we try undoing the changes from the Linux driver?

P33M, 2025-10-03:
High bandwidth reservation (for ISOC only) shouldn't affect FS/LS devices. The sparse control transfer bit paces each phase on a frame boundary - it'd be very odd that going _slower_ caused a data toggle error on a status stage (which is the thing that can kill the xHC on RP1).

The registers can be reverted at runtime - on latest EEPROM without the touchscreen plugged in, do:
```
sudo busybox devmem 0x1f0020c12c
sudo busybox devmem 0x1f0030c12c
```
Both commands should return `0x05030010`.
Then run:
`sudo busybox devmem 0x1f0020c12c 32 0x05020010 && sudo busybox devmem 0x1f0030c12c 32 0x05020010`
Plug the touchscreen in, and see if it still crashes (likely will).
Reboot, then instead run
`sudo busybox devmem 0x1f0020c12c 32 0x05010010 && sudo busybox devmem 0x1f0030c12c 32 0x05010010`
And plug the touchscreen in. It shouldn't crash, as the FS/LS handling will be identical to before 2024-04-17.

MankiniChykan, 2025-10-03:
> sudo busybox devmem 0x1f0020c12c 32 0x05010010 && sudo busybox devmem 0x1f0030c12c 32 0x05010010

Did all of those steps in order and xHCI still crashing. Is there any logs I can give you 

sudo dmesg -w
```
[    3.208381] brcmfmac: F1 signature read @0x18000000=0x15264345
[    3.216173] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    3.217718] usbcore: registered new interface driver brcmfmac
[    3.433618] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[    3.434176] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Aug 29 2023 01:47:08 version 7.45.265 (28bca26 CY) FWID 01-b677b91b
[    3.538143] Bluetooth: hci0: BCM: chip id 107
[    3.538401] Bluetooth: hci0: BCM: features 0x2f
[    3.539475] Bluetooth: hci0: BCM4345C0
[    3.539479] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0000
[    3.542089] Bluetooth: hci0: BCM4345C0 'brcm/BCM4345C0.raspberrypi,500.hcd' Patch
[    4.209509] Bluetooth: hci0: BCM: features 0x2f
[    4.210913] Bluetooth: hci0: BCM43455 37.4MHz Raspberry Pi 3+-0190
[    4.210923] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0382
[    4.211228] Bluetooth: hci0: BCM: Using default device address (43:45:c0:00:1f:ac)
[    4.450714] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    4.450721] Bluetooth: BNEP filters: protocol multicast
[    4.450726] Bluetooth: BNEP socket layer initialized
[    4.452272] Bluetooth: MGMT ver 1.23
[    4.457850] NET: Registered PF_ALG protocol family
[    4.493468] Bluetooth: RFCOMM TTY layer initialized
[    4.493483] Bluetooth: RFCOMM socket layer initialized
[    4.493488] Bluetooth: RFCOMM ver 1.11
[    5.470443] macb 1f00100000.ethernet eth0: PHY [1f00100000.ethernet-ffffffff:01] driver [Broadcom BCM54213PE] (irq=POLL)
[    5.470450] macb 1f00100000.ethernet eth0: configuring for phy/rgmii-id link mode
[    5.473392] macb 1f00100000.ethernet: gem-ptp-timer ptp clock registered.
[    9.571378] macb 1f00100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[  114.898149] usb 1-1: new high-speed USB device number 2 using xhci-hcd
[  115.034300] usb 1-1: New USB device found, idVendor=1a40, idProduct=0101, bcdDevice= 1.00
[  115.034319] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[  115.034323] usb 1-1: Product: USB2.0 HUB
[  115.096547] hub 1-1:1.0: USB hub found
[  115.096613] hub 1-1:1.0: 4 ports detected
[  115.474017] usb 1-1.2: new full-speed USB device number 3 using xhci-hcd
[  115.663188] usb 1-1.2: New USB device found, idVendor=0416, idProduct=c168, bcdDevice= 0.00
[  115.663201] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  115.663204] usb 1-1.2: Product: MTouch
[  115.663206] usb 1-1.2: Manufacturer: TSTP
[  115.663208] usb 1-1.2: SerialNumber: CMTP_1.0
[  125.925877] xhci-hcd xhci-hcd.0: xHCI host not responding to stop endpoint command
[  125.946063] xhci-hcd xhci-hcd.0: xHCI host controller not responding, assume dead
[  125.946077] xhci-hcd xhci-hcd.0: HC died; cleaning up
[  125.947617] usb 1-1: USB disconnect, device number 2
[  125.947632] usb 1-1.2: USB disconnect, device number 3
[  222.925344] usb 3-1: new high-speed USB device number 3 using xhci-hcd
[  223.061606] usb 3-1: New USB device found, idVendor=1a40, idProduct=0101, bcdDevice= 1.00
[  223.061625] usb 3-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[  223.061628] usb 3-1: Product: USB2.0 HUB
[  223.103227] hub 3-1:1.0: USB hub found
[  223.103287] hub 3-1:1.0: 4 ports detected
[  223.481359] usb 3-1.2: new full-speed USB device number 4 using xhci-hcd
[  223.674746] usb 3-1.2: New USB device found, idVendor=0416, idProduct=c168, bcdDevice= 0.00
[  223.674759] usb 3-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  223.674762] usb 3-1.2: Product: MTouch
[  223.674765] usb 3-1.2: Manufacturer: TSTP
[  223.674766] usb 3-1.2: SerialNumber: CMTP_1.0
[  233.957249] xhci-hcd xhci-hcd.1: xHCI host not responding to stop endpoint command
[  233.977425] xhci-hcd xhci-hcd.1: xHCI host controller not responding, assume dead
[  233.977437] xhci-hcd xhci-hcd.1: HC died; cleaning up
[  233.977811] usb 3-1: USB disconnect, device number 3
[  233.977817] usb 3-1.2: USB disconnect, device number 4
[  233.981311] usb 3-2: USB disconnect, device number 2
```

sudo dmesg -w | egrep -i 'usb|hid|touch'
[... 43 lines cut ...]
[    0.807569] usb 1-2: New USB device found, idVendor=0c45, idProduct=8101, bcdDevice= 0.01
[    0.807573] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    0.807575] usb 1-2: Product: USB Device
[    0.807577] usb 1-2: Manufacturer: USB
[    0.809636] usb 3-2: New USB device found, idVendor=2e8a, idProduct=0010, bcdDevice= 0.10
[    0.809640] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    0.809642] usb 3-2: Product: Pi 500 Keyboard
[    0.809644] usb 3-2: Manufacturer: Raspberry Pi Ltd
[    0.825810] input: Raspberry Pi Ltd Pi 500 Keyboard as /devices/platform/axi/1000120000.pcie/1f00300000.usb/xhci-hcd.1/usb3/3-2/3-2:1.0/0003:2E8A:0010.0001/input/input5
[    0.830932] input: USB USB Device Consumer Control as /devices/platform/axi/1000120000.pcie/1f00200000.usb/xhci-hcd.0/usb1/1-2/1-2:1.0/0003:0C45:8101.0002/input/input6
[    0.977810] hid-generic 0003:2E8A:0010.0001: input,hidraw0: USB HID v1.11 Keyboard [Raspberry Pi Ltd Pi 500 Keyboard] on usb-xhci-hcd.1-2/input0
[... cut ...]
[    1.281830] hid-generic 0003:2E8A:0010.0003: input,hidraw2: USB HID v1.11 Mouse [Raspberry Pi Ltd Pi 500 Keyboard] on usb-xhci-hcd.1-2/input1
[    1.281931] hid-generic 0003:0C45:8101.0004: input,hidraw3: USB HID v1.01 Mouse [USB USB Device] on usb-xhci-hcd.0-2/input1
[    1.287621] hid-generic 0003:0C45:8101.0005: hiddev96,hidraw4: USB HID v1.10 Device [USB USB Device] on usb-xhci-hcd.0-2/input2
[    3.377737] usbcore: registered new interface driver brcmfmac
[ 3477.799166] usb 1-2: USB disconnect, device number 2
```

MankiniChykan, 2025-10-04:
@pelwell how do I achieve capturing the session?

MankiniChykan, 2025-10-04:
I used this method. See next post for output

Nice — you’ve got the devmem log. There isn’t a dmesg log yet (that “No such file or directory” line confirms none was captured).

### Read the devmem log you have

```bash
sed -n '1,200p' /home/Chykan/devmem_logs/devmem.2025-10-04-103641.log
```

### Capture dmesg next time (and do the test)

Copy–paste this whole block in one go. It will:

* start a timestamped dmesg capture,
* dump the current register values,
* write the first override,
* prompt you to plug the touchscreen,
* then stop and save logs.

```bash
mkdir -p ~/devmem_logs
DMLOG=~/devmem_logs/dmesg.$(date +%F-%H%M%S).log
DEVLOG=~/devmem_logs/devmem.$(date +%F-%H%M%S).log

# 1) start kernel log capture
sudo dmesg -wT | tee -a "$DMLOG" &
DMESG_PID=$!

# 2) record the devmem session
{
  echo "=== BEFORE (expect 0x05030010) ==="
  sudo busybox devmem 0x1f0020c12c
  sudo busybox devmem 0x1f0030c12c

  echo "=== SET 0x05020010 on both ==="
  sudo busybox devmem 0x1f0020c12c 32 0x05020010
  sudo busybox devmem 0x1f0030c12c 32 0x05020010

  echo "=== AFTER writeback read ==="
  sudo busybox devmem 0x1f0020c12c
  sudo busybox devmem 0x1f0030c12c

  echo ">>> NOW plug the touchscreen and reproduce the crash. Press Enter when done..."
} | tee "$DEVLOG"

read _

# 3) stop kernel capture
sudo kill "$DMESG_PID" 2>/dev/null || true
sleep 0.2

echo "Saved:"
echo "  devmem log -> $DEVLOG"
echo "  dmesg  log -> $DMLOG"
```

### View the saved logs afterward

```bash
sed -n '1,200p' "$DEVLOG"
tail -n 200 "$DMLOG"
```

### Second boot test (different values)

After a reboot, run this to try the “0x05010010” setting and capture again:

```bash
mkdir -p ~/devmem_logs
DMLOG=~/devmem_logs/dmesg.$(date +%F-%H%M%S).log
DEVLOG=~/devmem_logs/devmem.$(date +%F-%H%M%S).log
sudo dmesg -wT | tee -a "$DMLOG" & DMESG_PID=$!
{
  echo "=== BEFORE ==="
  sudo busybox devmem 0x1f0020c12c
  sudo busybox devmem 0x1f0030c12c
  echo "=== SET 0x05010010 on both ==="
  sudo busybox devmem 0x1f0020c12c 32 0x05010010
  sudo busybox devmem 0x1f0030c12c 32 0x05010010
  echo "=== AFTER ==="
  sudo busybox devmem 0x1f0020c12c
  sudo busybox devmem 0x1f0030c12c
  echo ">>> Plug touchscreen, then press Enter..."
} | tee "$DEVLOG"
read _
sudo kill "$DMESG_PID" 2>/dev/null || true
sleep 0.2
echo "Saved: $DEVLOG and $DMLOG"
```

MankiniChykan, 2025-10-04:
Chykan@pi500:~ $ mkdir -p ~/devmem_logs
DMLOG=~/devmem_logs/dmesg.$(date +%F-%H%M%S).log
DEVLOG=~/devmem_logs/devmem.$(date +%F-%H%M%S).log
sudo dmesg -wT | tee -a "$DMLOG" & DMESG_PID=$!
{
  echo "=== BEFORE ==="
  sudo busybox devmem 0x1f0020c12c
  sudo busybox devmem 0x1f0030c12c
  echo "=== SET 0x05010010 on both ==="
  sudo busybox devmem 0x1f0020c12c 32 0x05010010
  sudo busybox devmem 0x1f0030c12c 32 0x05010010
  echo "=== AFTER ==="
  sudo busybox devmem 0x1f0020c12c
  sudo busybox devmem 0x1f0030c12c
  echo ">>> Plug touchscreen, then press Enter..."
} | tee "$DEVLOG"
read _
sudo kill "$DMESG_PID" 2>/dev/null || true
sleep 0.2
echo "Saved: $DEVLOG and $DMLOG"
[1] 1896
=== BEFORE ===
[Sat Oct  4 12:12:49 2025] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[Sat Oct  4 12:12:49 2025] Linux version 6.12.47+rpt-rpi-2712 (serge@raspberrypi.com) (aarch64-linux-gnu-gcc-14 (Debian 14.2.0-19) 14.2.0, GNU ld (GNU Binutils for Debian) 2.44) #1 SMP PREEMPT Debian 1:6.12.47-1+rpt1 (2025-09-16)
[Sat Oct  4 12:12:49 2025] KASLR enabled
[Sat Oct  4 12:12:49 2025] random: crng init done
[Sat Oct  4 12:12:49 2025] Machine model: Raspberry Pi 500 Rev 1.0
[Sat Oct  4 12:12:49 2025] efi: UEFI not found.
[Sat Oct  4 12:12:49 2025] Reserved memory: created CMA memory pool at 0x000000003a000000, size 64 MiB
[Sat Oct  4 12:12:49 2025] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[Sat Oct  4 12:12:49 2025] OF: reserved mem: 0x000000003a000000..0x000000003dffffff (65536 KiB) map reusable linux,cma
[Sat Oct  4 12:12:49 2025] OF: reserved mem: 0x0000000000000000..0x000000000007ffff (512 KiB) nomap non-reusable atf@0
[Sat Oct  4 12:12:49 2025] OF: reserved mem: 0x000000003fd165c0..0x000000003fd165fd (0 KiB) nomap non-reusable nvram@0
[Sat Oct  4 12:12:49 2025] NUMA: Faking a node at [mem 0x0000000000000000-0x00000001ffffffff]
[Sat Oct  4 12:12:49 2025] Faking node 0 at [mem 0x0000000000000000-0x000000003fffffff] (1024MB)
[Sat Oct  4 12:12:49 2025] Faking node 1 at [mem 0x0000000040000000-0x000000007fffffff] (1024MB)
[Sat Oct  4 12:12:49 2025] Faking node 2 at [mem 0x0000000080000000-0x00000000bfffffff] (1024MB)
[Sat Oct  4 12:12:49 2025] Faking node 3 at [mem 0x00000000c0000000-0x00000000ffffffff] (1024MB)
[Sat Oct  4 12:12:49 2025] Faking node 4 at [mem 0x0000000100000000-0x000000013fffffff] (1024MB)
[Sat Oct  4 12:12:49 2025] Faking node 5 at [mem 0x0000000140000000-0x000000017fffffff] (1024MB)
[Sat Oct  4 12:12:49 2025] Faking node 6 at [mem 0x0000000180000000-0x00000001bfffffff] (1024MB)
[Sat Oct  4 12:12:49 2025] Faking node 7 at [mem 0x00000001c0000000-0x00000001ffffffff] (1024MB)
[Sat Oct  4 12:12:49 2025] NUMA: Initialized distance table, cnt=8
[Sat Oct  4 12:12:49 2025] NODE_DATA(0) allocated [mem 0x3f7fd200-0x3f7fffff]
[Sat Oct  4 12:12:49 2025] NODE_DATA(1) allocated [mem 0x7fffd200-0x7fffffff]
[Sat Oct  4 12:12:49 2025] NODE_DATA(2) allocated [mem 0xbfffd200-0xbfffffff]
[Sat Oct  4 12:12:49 2025] NODE_DATA(3) allocated [mem 0xffffd200-0xffffffff]
[Sat Oct  4 12:12:49 2025] NODE_DATA(4) allocated [mem 0x13fffd200-0x13fffffff]
[Sat Oct  4 12:12:49 2025] NODE_DATA(5) allocated [mem 0x17fffd200-0x17fffffff]
[Sat Oct  4 12:12:49 2025] NODE_DATA(6) allocated [mem 0x1bfffd200-0x1bfffffff]
[Sat Oct  4 12:12:49 2025] NODE_DATA(7) allocated [mem 0x1ffb99200-0x1ffb9bfff]
[Sat Oct  4 12:12:49 2025] Zone ranges:
[Sat Oct  4 12:12:49 2025]   DMA      [mem 0x0000000000000000-0x00000000ffffffff]
[Sat Oct  4 12:12:49 2025]   DMA32    empty
[Sat Oct  4 12:12:49 2025]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[Sat Oct  4 12:12:49 2025] Movable zone start for each node
[Sat Oct  4 12:12:49 2025] Early memory node ranges
[Sat Oct  4 12:12:49 2025]   node   0: [mem 0x0000000000000000-0x000000000007ffff]
[Sat Oct  4 12:12:49 2025]   node   0: [mem 0x0000000000080000-0x000000003f7fffff]
[Sat Oct  4 12:12:49 2025]   node   1: [mem 0x0000000040000000-0x000000007fffffff]
[Sat Oct  4 12:12:49 2025]   node   2: [mem 0x0000000080000000-0x00000000bfffffff]
[Sat Oct  4 12:12:49 2025]   node   3: [mem 0x00000000c0000000-0x00000000ffffffff]
[Sat Oct  4 12:12:49 2025]   node   4: [mem 0x0000000100000000-0x000000013fffffff]
[Sat Oct  4 12:12:49 2025]   node   5: [mem 0x0000000140000000-0x000000017fffffff]
[Sat Oct  4 12:12:49 2025]   node   6: [mem 0x0000000180000000-0x00000001bfffffff]
[Sat Oct  4 12:12:49 2025]   node   7: [mem 0x00000001c0000000-0x00000001ffffffff]
[Sat Oct  4 12:12:49 2025] Initmem setup node 0 [mem 0x0000000000000000-0x000000003f7fffff]
[Sat Oct  4 12:12:49 2025] Initmem setup node 1 [mem 0x0000000040000000-0x000000007fffffff]
[Sat Oct  4 12:12:49 2025] Initmem setup node 2 [mem 0x0000000080000000-0x00000000bfffffff]
[Sat Oct  4 12:12:49 2025] Initmem setup node 3 [mem 0x00000000c0000000-0x00000000ffffffff]
[... 785 lines cut ...]
[Sat Oct  4 12:12:53 2025] Bluetooth: RFCOMM TTY layer initialized
[Sat Oct  4 12:12:53 2025] Bluetooth: RFCOMM socket layer initialized
[Sat Oct  4 12:12:53 2025] Bluetooth: RFCOMM ver 1.11
[Sat Oct  4 12:12:54 2025] macb 1f00100000.ethernet eth0: PHY [1f00100000.ethernet-ffffffff:01] driver [Broadcom BCM54213PE] (irq=POLL)
[Sat Oct  4 12:12:54 2025] macb 1f00100000.ethernet eth0: configuring for phy/rgmii-id link mode
[Sat Oct  4 12:12:54 2025] macb 1f00100000.ethernet: gem-ptp-timer ptp clock registered.
[Sat Oct  4 12:12:57 2025] macb 1f00100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[... cut ...]
[Sat Oct  4 12:15:20 2025] usb 3-1.2: SerialNumber: CMTP_1.0
[Sat Oct  4 12:15:30 2025] xhci-hcd xhci-hcd.1: xHCI host not responding to stop endpoint command
[Sat Oct  4 12:15:31 2025] xhci-hcd xhci-hcd.1: xHCI host controller not responding, assume dead
[Sat Oct  4 12:15:31 2025] xhci-hcd xhci-hcd.1: HC died; cleaning up
[Sat Oct  4 12:15:31 2025] usb 3-1: USB disconnect, device number 3
[Sat Oct  4 12:15:31 2025] usb 3-1.2: USB disconnect, device number 4
[Sat Oct  4 12:15:31 2025] usb 3-2: USB disconnect, device number 2

P33M, 2025-10-06:
Which model of monitor are you using? I see an 18.5", 21.5" and (for some reason) dual 18.5" touchscreens on Amazon.

MankiniChykan, 2025-10-07:
I am using this exact model [https://uperfect.com/products/uperfect-y-vertical-monitor-15-6](https://uperfect.com/products/uperfect-y-vertical-monitor-15-6)

MankiniChykan, 2025-11-11:
@P33M I will try hook the monitor up to my Pi400 which it works on and see if I can pull some data from that.

The monitor was finicky on the Pi400 as well. The Monitors Power Supply has to be not plugged in for it to enumerate touch, then you could plug in the monitors power supply.

I believe this is because the USB-C connection to the touch also provides power to the Pi.