raspberrypi/linux #6080: cfg80211: loaded regulatory.db is malformed or signature is missing/invalid

iam-TJ, 2024-03-29:
### Describe the bug

The default `regulatory.db` from `wireless-regdb` signed by the Debian key is failing the signature check and breaking the ability to set the regulatory domain - not great when operating as an Access Point!
```
tj@noc:~$ tail -n 1 /proc/cpuinfo
Model           : Raspberry Pi 3 Model B Rev 1.2

tj@noc:~$ uname -a
Linux noc 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux

tj@noc:~$ dpkg --print-architecture
armhf

tj@noc:~$ cat /proc/cmdline
coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=/dev/mapper/PIDRIVE01-rootfs rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=GB

tj@noc:~$ . /etc/os-release && echo $PRETTY_NAME
Raspbian GNU/Linux 12 (bookworm)

tj@noc:~$ apt list raspberrypi-kernel wireless-regdb
Listing... Done
raspberrypi-kernel/stable,now 1:1.20230405-1 armhf [installed,auto-removable]
wireless-regdb/stable,now 2022.06.06-1 all [installed,automatic]
```
My workaround is:
```
update-alternatives --config regulatory.db
```
and select the `.upstream` database so it has:
```
tj@noc:~$ readlink -e /lib/firmware/regulatory.db
/usr/lib/firmware/regulatory.db-upstream
```
This was originally reported in issue #5535 and supposed to be fixed by importing the two Debian signatures in issue #5536  but at some point has broken.

### Steps to reproduce the behaviour

Attempt to change the regulatory domain to anything other than `US` (`iw reg get` shows unset regardless)

### Device (s)

Raspberry Pi 3 Mod. B

### System

Mar 17 2023 10:52:42 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)

Linux noc 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux

pelwell, 2024-03-29:
Your kernel is nearly a year old - try with a current build.

iam-TJ, 2024-04-10:
> Your kernel is nearly a year old - try with a current build.

I checked this extensively to the point of manually inspecting the archive pool itself ... and then finally cottoned on that Bookworm release has moved over to an almost pure Debian repo with the kernels in differently named packages. I'd done a release-upgrade a long time ago but not based on any R.P. specific advice (I'm a Debian developer) and didn't realise there had been a fundamental change - sorry for the noise!

closed 2024-04-10 by iam-TJ

ValeZAA, 2025-04-12:
https://unix.stackexchange.com/a/760283
Run update-alternatives --config regulatory.db

and select 2 for upstream keys. It is expected warnings.