raspberrypi/linux #6081: No way to set Full Range RGB

boh-wyp, 2024-03-30:
### Describe the bug

No way to set Full Range RGB output except by workaround, using `kmstest -c X --property="Broadcast RGB"=1` which sets it properly until reboot

### Steps to reproduce the behaviour

Install Bookworm Raspbian Lite, plug into a display that uses limited RGB by default

### Device (s)

Raspberry Pi 5

### System

`Raspberry Pi reference 2023-12-05`
`2024/02/16 15:28:41 version 4c845bd3 (release) (embedded)`
`Linux raspberrypi 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux`

### Additional context

I have searched for quite some time and I don't see any other way to force Full Range RGB, nor an option you could use in /boot/firmware/cmdline.txt in video=.
If there's any way to do that so I don't need to launch kmstest every boot and manually close it, please let me know

6by9, 2024-04-02:
Technically your display is non-compliant to the HDMI spec if it is requiring limited range RGB for non-CEA modes.

There is no command line option to set that property - mainline are still debating the best way of handling these non-compliant devices, and ignoring them in the meantime.

The quickest route would be to use `kmstest --flip=2 -c X --property="Broadcast RGB"=1`, which will automatically quit after 2 frames.

Alternatively you can do the same thing though modetest. You need the ID for the relevant HDMI connector, which on my system comes up as 32 for HDMI-0, and 41 for HDMI-1.
`modetest -M vc4 -w 32:"Broadcast RGB":1`

boh-wyp, 2024-04-26:
Sorry for the late reply, thank you! I didn't figure I could use flip for quitting automatically. Good enough solution for now.
Cheers!

closed 2024-04-26 by boh-wyp