raspberrypi/linux #6084: writing to /sys/class/gpio/export results in write error: Invalid argumen

berndporr, 2024-04-02:
### Describe the bug

With the current kernel Linux raspberrypi 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux it's no longer possible to access gpio pins via sysfs. It works perfectly fine with libgpiod. Not sure if that's a feature or a bug? 

### Steps to reproduce the behaviour

How to reproduce:
```
echo 27 >/sys/class/gpio/export
```
this then gives an error message:
```
-bash: echo: write error: Invalid argument
```
This also affects event driven programming. For example: https://github.com/berndporr/gpio-sysfs if you compile it and then run `gpio-int-test 27` for example that complains it cannot get a file descriptor for the `poll` command.
```
$ sudo ./gpio-int-test 27
gpio/direction: No such file or directory
gpio/set-edge: No such file or directory
gpio/fd_open: No such file or directory
1: timeout
2: timeout
3: ^C
```

### Device (s)

Raspberry Pi 3 Mod. B+, Raspberry Pi 5

### System

System Information
------------------

Raspberry Pi 3 Model B Rev 1.2
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"

Raspberry Pi reference 2024-03-15
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f19ee211ddafcae300827f953d143de92a5c6624, stage2

Linux raspberrypi 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux
Revision	: a02082
Serial		: 00000000e26ad282
Model		: Raspberry Pi 3 Model B Rev 1.2
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0, libcamera interfaces=0

Videocore information
---------------------

Feb 29 2024 12:26:47 
Copyright (c) 2012 Broadcom
version f4e2138c2adc8f3a92a3a65939e458f11d7298ba (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
udev              325632       0    325632   0% /dev
tmpfs              92912    1008     91904   2% /run
/dev/mmcblk0p2  14727944 2953116  11006964  22% /
tmpfs             464552       0    464552   0% /dev/shm
tmpfs               5120      12      5108   1% /run/lock
/dev/mmcblk0p1    522230   63586    458644  13% /boot/firmware
[... 1026 lines cut ...]
[   11.210687] vc4-drm soc:gpu: [drm] Cannot find any crtc or sizes
[   11.358613] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SS
[   11.445132] Bluetooth: hci0: BCM: features 0x2e
[   11.450132] Bluetooth: hci0: BCM43438A1 37.4MHz Raspberry Pi 3-0141
[   11.450156] Bluetooth: hci0: BCM43430A1 (001.002.009) build 0508
[   12.971891] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   12.971913] Bluetooth: BNEP filters: protocol multicast
[   12.971996] Bluetooth: BNEP socket layer initialized
[   12.980158] Bluetooth: MGMT ver 1.22
[   12.999246] NET: Registered PF_ALG protocol family
[   13.210321] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   13.212911] smsc95xx 1-1.1:1.0 eth0: Link is Up - 100Mbps/Full - flow control off
[   13.267429] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   91.800707] systemd[1046]: memfd_create() called without MFD_EXEC or MFD_NOEXEC_SEAL set
[  165.206237] export_store: invalid GPIO 27
[  294.397397] export_store: invalid GPIO 27
[  449.533065] export_store: invalid GPIO 27
[  462.845137] export_store: invalid GPIO 27
[  481.076903] export_store: invalid GPIO 27
[  633.137180] export_store: invalid GPIO 27
[  902.700198] export_store: invalid GPIO 27

### Additional context

I've been using pigpio for my I/O needs and noticed that its event callback on GPIO pins is no longer firing. I know it uses internally `/sys/class/gpio/export` to get a filedescriptor for the poll() command. So I 1st thought it must be something with my ADC's data-ready pln so I soldered up a pushbutton to GPIO27. pigpio didn't fire. So I tested with libgpiod and it all worked. This morning I wrote just the pin number with shell commands and noticed it's no longer working. I know /sys is deprecated but would be very cuious if that's been a decision by the RPI team to drop support or just a bug? I haven't seen any official announcement? I would have switched earlier to pigpiod but the doc for event driven gpio has been very poor. Either way I suspect also wiringPI might be affected by this as it's also using sys as far as I know.

pelwell, 2024-04-02:
This is essentially a duplicate of #6037, in that the same answers apply.

TL;DR - the Linux maintainers don't want the global GPIO number space starting at 0. `cat /sys/kernel/debug/gpio` shows you the number assignments for each GPIO. In future, the /sys/class/gpio interface will disappear altogether - I believe Ubuntu/Canonical has already disabled it.

pelwell, 2024-04-02:
Closing this to avoid duplication - please use #6037 for further discussion.

closed 2024-04-02 by pelwell

berndporr, 2024-04-02:
Thanks for the quick reply. I see it's a feature! Well, I've switched to libgpiod yesterday and all good. I'm all for /sys/class/gpio to go away and the event processing there is so much better in libgpiod. I'm teaching realtime embedded coding and it's all about events! Well, that forced me to suss out event handling with it and already in my handout for my students!
https://berndporr.github.io/realtime_cpp_coding/node3.html#SECTION00334000000000000000

pelwell, 2024-04-02:
The only "gotcha" with libgpiod is the lack of names (or the ability to set the index values) on the gpiochips. On Pi 5, the user-facing GPIOs on the header end up being on gpiochip4, which is not ideal.

berndporr, 2024-04-02:
I've noticed. The guys from the Graz computer club who are hacking away at the wiringPI also noticed that: https://github.com/WiringPi/WiringPi/blob/master/wiringPi/wiringPi.c#L2145.
Also the accumlated numbering /sys/kernel/debug/gpio will be different between RPI1-4 and 5 isn't it? So there is no escape than detecting which rpi it is I guess. At least the gpiod library is closely developed in sync with the kernel. Anything that's not developed so closely to the kernel will or have now bombed. Or is there an attempt close to RPI to have c API which hides it all nicely away? wiringPI tried that 10 years ago and then created a lot of confusion.

pelwell, 2024-04-02:
There are a number of strategies one could use:
* On current Pi DTBs (by which I mean not old - we will try to keep this valid in the future), look for the pin called "GPIO2" and use that gpiochip.
* We might in future find a way to make the Pi 5 GPIOs appear on gpiochip0, but until then you could use gpiochip4 if it exists, otherwise gpiochip0.

Dave Jones at Canonical/Ubuntu has produced rpi-lgpio (https://github.com/waveform80/rpi-lgpio), an RPi.GPIO clone that uses libgpiod for the backend, but that's Python only.