raspberrypi/linux #6085: offical 7" touchscreen does not rotate on the Raspberry Pi 5

spockfish, 2024-04-02:
### Describe the bug

Using the KMS driver, the touchscreen does not rotate when given the proper setup (adding` invx,invy` and `rotate`) in config.txt.

So, this works on a Raspberry Pi 4:

config.txt:
```
gpu_mem=128
max_framebuffers=2
disable_fw_kms_setup=1
disable_overscan=1
ignore_lcd=1
dtoverlay=vc4-kms-v3d-pi5,noaudio,nohdmi
dtoverlay=vc4-kms-dsi-7inch,invx,invy
```

cmdline:
`video=DSI-1:800x480M,rotate=180
`
However, on the Pi 5 it results in a working touchscreen, but not a rotated one.

### Steps to reproduce the behaviour

Configure the screen for rotation.

### Device (s)

Raspberry Pi 5

### System

custom built OS, Linux 6.6 (latest build from this repo), latest firmware

### Logs

The logs show no difference between a startup with 'normal' orientation vs 'rotated'.

`journalctl -b | grep vc4` returns:

```
Apr 02 21:17:46 ropieee5 kernel: vc4-drm axi:gpu: bcm2712_iommu_of_xlate: MMU 1000005200.iommu
Apr 02 21:17:46 ropieee5 kernel: vc4-drm axi:gpu: bound 107c580000.hvs (ops vc4_hvs_ops [vc4])
Apr 02 21:17:46 ropieee5 kernel: vc4-drm axi:gpu: bound 107c500000.mop (ops vc4_txp_ops [vc4])
Apr 02 21:17:46 ropieee5 kernel: vc4-drm axi:gpu: bound 107c501000.moplet (ops vc4_txp_ops [vc4])
Apr 02 21:17:46 ropieee5 kernel: [drm] Initialized vc4 0.0.0 20140616 for axi:gpu on minor 2
Apr 02 21:17:46 ropieee5 kernel: vc4-drm axi:gpu: [drm] Cannot find any crtc or sizes

```

### Additional context

I use a Qt app that directly accesses DRM. So no X or Wayland involved.

pelwell, 2024-04-02:
This is possibly a stupid question - this is not really my field, and I'm just guessing - but is the DSI-1 designator still correct on Pi 5? With no video= in the cmdline do you still get working video and a touchscreen?

pelwell, 2024-04-02:
Yes, I think it was a stupid question - the screen is probably being rotated correctly as a result of that addition.

spockfish, 2024-04-02:
Well, I tried without the 'screen' remark in the cmdline.txt, but it does not make a difference. So yes, it works, but no rotate. 

So you might be on to something, because one thing I just discovered: the touch part *is* rotated. So it is only the video part that isn't.

spockfish, 2024-04-02:
BINGO!

This works:

video=DSI-2:800x480M,rotate=180

I really don't get why DSI-2 works. I've connected to DSI1 on the board (as this is default).

6by9, 2024-04-02:
As per [README](https://github.com/raspberrypi/linux/blob/rpi-6.6.y/arch/arm/boot/dts/overlays/README#L5015-L5016) `invx` and `invy` are `Touchscreen inverted x axis` and `Touchscreen inverted y axis`. Touch overlay only. They will not affect the rendering of the image.

CAM/DISP0 should always be known as DSI-1, and CAM/DISP1 as DSI-2.

`video=...rotate=X` asks fbcon (framebuffer console) to rotate the image. If the primary plane on the device supports hardware rotation then it will use it, otherwise it will do it in software when rendering the console text to the framebuffer. Any application which writes directly to the framebuffer will not be rotated.
vc4 (Pi0-4) supports H & V flips on planes, but DSI on RP1 (Pi5) doesn't.

spockfish, 2024-04-03:
Ok... So the DSI-2 is conform what to be expected. Right?

It still does not explain why on the Pi 4 the application also rotates 'automagically', but not on the Pi 5. Or is this the 'vc4 (Pi0-4) supports H & V flips on planes, but DSI on RP1 (Pi5) doesn't.' part?

spockfish, 2024-04-03:
Thanks @6by9 .

So to summarize:

- on the Pi 5 I need to look for a software solution to rotate my app. And I can 'forget' about the video parameter.
- this issue is not a bug; expect maybe for a hint in the documentation.

So that means we can close this issue. Correct?

spockfish, 2024-04-20:
As this is expected behaviour I'm closing this issue.

closed 2024-04-20 by spockfish

colesnicov, 2024-05-05:
Hello here.  I'm glad I found you here. I am currently **dealing with something similar with a Raspberry PI 2B**, I have listed my searches so far here https://github.com/raysan5/raylib/issues/3958.

But I'm starting to understand that I'll probably have to handle everything myself, but I'll ask you here just to be sure. Does this behavior _(not a bug)_ apply to all current Raspberry (1,2,3,4 and 5)? 

Thank you for the answer, I'm at peace now (: